platformio/platform-nordicnrf52 #3: Arduino Core for Nordic Semiconductor nRF5 based boards

ivankravets, 2017-06-24:
# Moved from https://github.com/platformio/platformio-core/issues/868

Arduino Core:

* https://github.com/sandeepmistry/arduino-nRF5

/cc @amq, @jrobeson, @colinblake, @boneskull, @profra, @PierreKircher, @sabas1080, @danielpgross, @cujomalainey

mtiutiu, 2017-07-28:
Is it possible to make the **sandeepmistry arduino-nRF5 core** available for **platform-nordicnrf51** too? It seems that this core is more up to date and working well so far as I'm using it on some projects. And to be honest I want to get rid of Arduino IDE - this nrf5 core is all that keeps me using it for now. Thanks a lot.

ivankravets, 2017-07-31:
@mtiutiu We are working on that. Please follow us on Twitter to stay tuned https://twitter.com/PlatformIO_Org

cujomalainey, 2017-07-31:
@ivankravets the package by sandeepmistry only supports BLE soft devices, how hard would it be to integrate a variant (that I am working on) that also supports the ANT stack.

ivankravets, 2017-07-31:
@cujomalainey could you give some links?

ivankravets, 2017-08-22:
Resolved in https://github.com/platformio/platform-nordicnrf52/releases/tag/v1.2.0

closed 2017-08-22 by ivankravets