platformio/platform-nordicnrf52 #8: Support for the Adafruit Feather nRF52 Bluefruit LE board

slugh23, 2017-08-26:
Please add support for the Adafruit Feather nRF52 Bluefruit LE board:

https://www.adafruit.com/product/3406

https://learn.adafruit.com/bluefruit-nrf52-feather-learning-guide/introduction

cujomalainey, 2017-08-27:
This would likely be an entirely different platform since adafruit has their own toolchain for that board.

stonehippo, 2017-10-31:
Any detail on the differences @cujomalainey? Beyond supporting the CP2104 USB to UART bridge (which is also used by the ES32 Feather, a board supported by PIO), are there issues with the fireware? Trying to get why the PIO nrF52 toolchain wouldn't work.

cujomalainey, 2017-10-31:
The reason is the boards are built upon two very different Arduino implementations (bootloaders included.) Here is the [Feather implementation](https://github.com/sandeepmistry/arduino-nRF5), an original by Adafruit. This repo is based on this [repo](https://github.com/sandeepmistry/arduino-nRF5) by Sandeep Mistry. Might be able to get something to load, but I have no idea what is in the Adafruit bootloader and they also import some stuff from [FreeRTOS](https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/cores/nRF5/freertos) so you would probably lose support for the Adafruit Android BLE library as well by replacing this code.

jeremypoulter, 2017-12-16:
FYI I have started work on this in #12

jeremypoulter, 2018-03-18:
Managed to get back to this. I have the firmware building and I can manually upload it using the `nrfutil` package:

```
nrfutil dfu genpkg --dev-type 0x0052 --sd-req 0x00A5 --application .pioenvs/bluefruit_nrf52/firmware.hex .pioenvs/bluefruit_nrf52/firmware.zip
nrfutil dfu serial -pkg .pioenvs/bluefruit_nrf52/firmware.zip -p /dev/ttyUSB0 -b 115200
```

So basically the remaining tasks are:
- [ ] Change to HTTPS Git links
- [ ] Integrate `nrfutil` and upload over serial - Any pointer?
- [ ] Upload via programmer, J-Link/nrfprog
- [ ] flash bootloader

FYI see https://github.com/jeremypoulter/platformio-examples for an example

jeremypoulter, 2018-08-25:
#12 basically works, at least for my use case. There are a few manual steps that I need some guidance on, plus some more general info on wet else needs to be done to allow the PR to be merged.

Navis-Raven, 2018-09-02:
There is also this board which can be added: "Adafruit Feather nRF52 Pro with myNewt Bootloader"

https://www.adafruit.com/product/3574

enny1170, 2018-12-13:
I'm also interessted, to use this board with platform IO. How is the state?

jeremypoulter, 2018-12-13:
As above it basically works, been using it for a while. I however need some input as to the best way to have the two Arduino Cores running in parallel. As an FYI I am currently working on updating to the latest Adafruit core with nRF52840 support.

jeremypoulter, 2018-12-31:
> I updated @jeremypoulter <https://github.com/jeremypoulter>'s codebase to
> use the latest version of Adafruit's Board Support Package and got it
> working :)
>
> To use the Adafruit nRF52832, use the following settings for your
> platform.ini:
>
> [env:nrf52]
> platform = https://github.com/nsamala/platform-nordicnrf52.git#feather_nrf52832
> board = adafruit_feather_nrf52832
> framework = arduino-adafruit
> upload_protocol = nrfutil
> build_flags =
>   -DARDUINO_BSP_VERSION=\"0.9.2\"
>
> I haven't added support for the nRF52840, but it shouldn't be too crazy to
> add board settings for it using my branch.

Fantastic, can you create a PR for me please?

Jeremy

enny1170, 2018-12-31:
sound's great, i will try it.

nsamala, 2019-01-19:
@jeremypoulter sorry for the delay, yeah I'll go ahead and create a PR

jeremypoulter, 2019-01-20:
No worries, is ok, I pulled all your changes into my PR, or rather I
created a new branch from your stuff and added back all the J-Link stuff
and a few other things.

nsamala, 2019-01-20:
@jeremypoulter be sure to pull in the latest changes from yesterday. I've updated it to work alongside the existing PIO BSP for the nRF

jeremypoulter, 2019-01-28:
Try doing a 'pio update' I think I fixed a lot of those errors over the weekend.

ivankravets, 2019-02-01:
Resolved in https://github.com/platformio/platform-nordicnrf52/pull/23

Thanks, @jeremypoulter! 🍺 

Guys, please re-test with the latest upstream version https://docs.platformio.org/en/latest/platforms/nordicnrf52.html#upstream

closed 2019-02-01 by ivankravets

jwanga, 2019-02-01:
thanks @jeremypoulter and @ivankravets! What board should i use for the Adafruit Feather nRF52?

stonehippo, 2019-02-01:
@jwanga it looks like it's `adafruit_feather_nrf52832`.

jwanga, 2019-02-02:
@stonehippo Yup. Here is my .ini'

`[env:adafruit_feather_nrf52]
platform = https://github.com/platformio/platform-nordicnrf52.git
board = adafruit_feather_nrf52832
framework = arduino
upload_protocol = nrfutil
dependencies = ArduinoSTL
build_flags = -DNRF52_S132
lib_deps = BLEPeripheral`

stonehippo, 2019-02-02:
@jwanga according to `pio boards --installed`:

```
Platform: nordicnrf52
-------------------------------------------------------------------------------------------------------------------------------------------
ID                    MCU            Frequency  Flash   RAM    Name
-------------------------------------------------------------------------------------------------------------------------------------------
nrf52840_dk           NRF52840       64MHz     1MB     256KB  Nordic nRF52840-DK
stct_nrf52_minidev    NRF52832       64MHz     512KB   64KB   Taida Century nRF52 mini board
redbear_blenano2      NRF52832       64MHz     512KB   64KB   RedBearLab BLE Nano 2
redbear_blend2        NRF52832       64MHz     512KB   64KB   RedBearLab Blend 2
bluey                 NRF52832       64MHz     512KB   64KB   Bluey nRF52832 IoT
delta_dfbm_nq620      NRF52832       64MHz     512KB   64KB   Delta DFBM-NQ620
nrf52_dk              NRF52832       64MHz     512KB   64KB   Nordic nRF52-DK
adafruit_feather_nrf52832 NRF52832       64MHz     512KB   64KB   Adafruit Bluefruit nRF52832 Feather
hackaBLE              NRF52832       64MHz     512KB   64KB   hackaBLE
ublox_evk_nina_b1     NRF52832       64MHz     512KB   64KB   u-blox EVK-NINA-B1
```

The very first time I ran it, I did get an error, but I had entered it by hand, so I cut and pasted it from the boards list, and it was fine, so might have been a typo. I  just`pio run` on a little demo with the following `platformio.ini` and it get the platform, board, and framework installed.

```
env:adafruit_feather_nrf52832]
platform = https://github.com/platformio/platform-nordicnrf52.git
board = adafruit_feather_nrf52832
framework = arduino
```

jwanga, 2019-02-02:
@stonehippo this is so strange. The board is showing as installed and when I copy the name it still fails. Here is my exact output.

```
> Executing task: platformio run --target upload <

Processing adafruit_feather_nrf52832 (platform: https://github.com/platformio/platform-nordicnrf52.git; board: adafruit_feather_nrf52832; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Error: Unknown board ID 'adafruit_feather_nrf52832'
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

stonehippo, 2019-02-02:
@jwanga perhaps try removing the BSP with `pio platform uninstall nordicnrf52`, then give it a go again.

I’m getting a different issue when trying to build, with an error being thrown while compiling the Adafruit Bluefruit libs. Might be because I’m running this on a Raspberry Pi, though. Will switch to my Mac, perhaps.

ericlangel, 2019-02-25:
Processing adafruit_feather_nrf52832 (platform: https://github.com/platformio/platform-nordicnrf52.git; board: adafruit_feather_nrf52832; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/adafruit_feather_nrf52832.html
PLATFORM: Nordic nRF52 > Adafruit Bluefruit nRF52832 Feather
HARDWARE: NRF52832 64MHz 64KB RAM (512KB Flash)
DEBUG: CURRENT(jlink) ON-BOARD(jlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 9 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pioenvs\adafruit_feather_nrf52832\src\main.cpp.o
Linking .pioenvs\adafruit_feather_nrf52832\firmware.elf
Checking size .pioenvs\adafruit_feather_nrf52832\firmware.elf
Building .pioenvs\adafruit_feather_nrf52832\firmware.hex
arm-none-eabi-objcopy: .pioenvs\adafruit_feather_nrf52832\firmware.hex 64-bit address 0x4b4fa300026000 out of range for Intel Hex file
arm-none-eabi-objcopy:.pioenvs\adafruit_feather_nrf52832\firmware.hex: bad value
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   3.4% (used 2212 bytes from 65536 bytes)
PROGRAM: [          ]   1.9% (used 9732 bytes from 524288 bytes)
*** [.pioenvs\adafruit_feather_nrf52832\firmware.hex] Error 1
================================================= [ERROR] Took 1.75 seconds =================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

This is the Error with a new project for Adafruit NRF52 Feather.
No additional code.

stonehippo, 2019-02-25:
@jwanga My RPi issue was down to a toolchain problem, which I resolved in #28 by supplying a new gcc-arm-none-eabi for armv7.

@ericlangel Are you on Windows? Can you send post up the following:

- output of `pio platform list`
- the `nordicnrf52` section of the output from `pio boards --installed`
- whatever shows up for the Nordic nrf52 platform when you do `pio update`
- contents of the `platfromio.ini` for the test project you're trying to build

I assume that you're running on Windows. I can give it a try there tonight. What did you throw in for test code?

stonehippo, 2019-02-25:
@ericlangel also, if you are on Windows, it looks like the package for the toolchain-gccarmnoneeabi was updated a couple of days ago. I'm running that version on MacOS without any issues on a simple test app, but there may be an issue with the Windows version. Again, I'll test later.

ericlangel, 2019-02-26:
@stonehippo 
yes, iam on Windows.

PS C:\Users\eric\Documents\PlatformIO\NRF52_Test> pio platform list
nordicnrf52 ~ Nordic nRF52
==========================
The nRF52 Series are built for speed to carry out increasingly complex tasks in the shortest possible time and return to sleep, conserving precious battery power. They have a Cortex-M4F processor and are the most capable Bluetooth Smart SoCs on the market.

Home: http://platformio.org/platforms/nordicnrf52
Frameworks: arduino, mbed
Packages: tool-jlink, tool-nrfjprog, toolchain-gccarmnoneeabi, framework-arduinonordicnrf5, tool-openocd, framework-mbed, tool-sreccat, framework-arduinoadafruitnrf52
Version: 8434d98

PS C:\Users\eric\Documents\PlatformIO\NRF52_Test> pio boards --installed

Platform: nordicnrf52
------------------------------------------------------------------------------------------------------------------------------
ID                    MCU            Frequency  Flash   RAM    Name
------------------------------------------------------------------------------------------------------------------------------
nrf52840_dk           NRF52840       64MHz     1MB     256KB  Nordic nRF52840-DK
adafruit_feather_nrf52840 NRF52840       64MHz     796KB   243KB  Adafruit Feather nRF52840 Express
stct_nrf52_minidev    NRF52832       64MHz     512KB   64KB   Taida Century nRF52 mini board
redbear_blenano2      NRF52832       64MHz     512KB   64KB   RedBearLab BLE Nano 2
redbear_blend2        NRF52832       64MHz     512KB   64KB   RedBearLab Blend 2
bluey                 NRF52832       64MHz     512KB   64KB   Bluey nRF52832 IoT
delta_dfbm_nq620      NRF52832       64MHz     512KB   64KB   Delta DFBM-NQ620
nrf52_dk              NRF52832       64MHz     512KB   64KB   Nordic nRF52-DK
adafruit_feather_nrf52832 NRF52832       64MHz     512KB   64KB   Adafruit Bluefruit nRF52832 Feather
nrf52840_dk_adafruit  NRF52840       64MHz     796KB   243KB  Nordic nRF52840-DK (Adafruit BSP)
hackaBLE              NRF52832       64MHz     512KB   64KB   hackaBLE
ublox_evk_nina_b1     NRF52832       64MHz     512KB   64KB   u-blox EVK-NINA-B1

Platform Manager
================
Platform Nordic nRF52
--------
Updating nordicnrf52                     @ 8434d98        [Up-to-date]
Updating tool-sreccat                    @ 1.164.0        [Up-to-date]
Updating toolchain-gccarmnoneeabi        @ 1.80201.181220 [Up-to-date]
Updating framework-arduinoadafruitnrf52  @ 1.903.0        [Up-to-date]

[env:adafruit_feather_nrf52832]
platform = https://github.com/platformio/platform-nordicnrf52.git
board = adafruit_feather_nrf52832
framework = arduino

#include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}

stonehippo, 2019-02-26:
@ericlangel ok, I've confirmed that I can replicate your issue on Windows 10.

The issue appears to be the latest x86_64 gcc-arm-none-eabi toolchain available (1.80201.181220). I modified the Feather board support to force using the previous version available in the PIO Bintray repo (1.70201) and everything compiled, no memory error in the hex file. So I'm pretty sure it's a toolchain bug. I can't look too closely right now, but I'll dig into it a little more later (might be a known issue with that version of the ARM cross-compiler on Windows). As I mentioned earlier, compiling against the same version on MacOS produced no errors. 

/cc @ivankravets

ericlangel, 2019-02-26:
@stonehippo 
how can i force using the old toolchain?

stonehippo, 2019-02-26:
@ericlangel I deleted `.platformio\packages\toolchain-gccarmnoneeabi`, then modified the version field for the toolchain in `.platformio\platforms\nordicnrf52\platform.json as shown in the excerpt below:

```json
"packages": {
    "toolchain-gccarmnoneeabi": {
      "type": "toolchain",
      "version": "=1.70201.0"
    },
```

This change won't be permanent (it'll get overridden if you update the platform), but it should allow you to temporarily downgrade the gcc-arm-none-eabi toolchain.

@ivankravets I've done a quick check and found that the error @ericlangel reported has shown up on the Nordic forums, too. It seems that this is a known bug with the 1.8 toolchain, but I don't have any details on that yet.

stonehippo, 2019-02-26:
Ah, found the upstream bug: https://bugs.launchpad.net/gcc-arm-embedded/+bug/1810274. @ivankravets does it make sense to pull the Windows gcc-arm-none-eabi 1.80201 toolchain, since it seems to be broken?

ivankravets, 2019-02-26:
We've just updated toolchain to https://github.com/gnu-mcu-eclipse/arm-none-eabi-gcc/releases/tag/v8.2.1-1.2

Could you try `$ pio update` and re-run project building? Does it work?

stonehippo, 2019-02-27:
@ivankravets the Windows updated toolchain works. It claims to be the amd64 version in the package.json (the earlier versions claimed x86_64), but that doesn't seem to make a difference.

ericlangel, 2019-02-27:
@stonehippo 
It's working ;)
Now i can try to program my nrf52 feather.

Thank you