platformio/platform-nordicnrf52 #10: Softdevice flashing

jeremypoulter, 2017-10-09:
I am probably missing something really obvious, but how doe you flash a softdevice under Platform IO?

valeros, 2017-10-12:
Hi @jeremypoulter ! The resulting binary is merged from the user application and selected softdevice.
https://github.com/platformio/platform-nordicnrf52/blob/develop/builder/main.py#L130

jeremypoulter, 2017-10-12:
I am not so sure that is what this line is doing. Certainly the end result does not work until I use Arduino IDE to flash the Soft Device. Then everything is ok.

valeros, 2017-10-12:
We use a special tool called `SRecord` to combine main application and the softdevice binary.
Could you please describe in greater detail what exactly doesn't work? A simple blink, BLE functionality? What is your board? Did you use `build_flags` to specify the softdevice version?
Thanks!

jeremypoulter, 2017-10-12:
It is our own custom board with a [Liard BL652 module](https://www.lairdtech.com/products/bl652-ble-module). I can see the soft device being merged into the firmware and the result being flashed:

```
C:\Users\jeremy\.platformio\packages\tool-sreccat\srec_cat C:\Users\jeremy\.platformio\packages\framework-arduinonordicnrf5@src-d2a7ef78dfbec2ba5b0b645c94ed87d3\cores\nRF5\SDK\components\softdevice\s132\hex\s132_
nrf52_2.0.1_softdevice.hex -intel .pioenvs\nudge_laser\userfirmware.hex -intel -o .pioenvs\nudge_laser\firmware.hex -intel --line-length=44
"openocd" -s C:\Users\jeremy\.platformio\packages\tool-openocd -f scripts/interface/jlink.cfg -c "transport select swd" -f scripts/target/nrf52_arduino.cfg -c "program {{.pioenvs\nudge_laser\firmware.hex}} verify
 reset; shutdown;"
```

but the  code fails to execute until the device is flashed with the softdevice using Arduino IDE and the nRF5FlashSoftDevice.jar tool. These modules do come pre-flashed with 'smartBASIC' so maybe there is some state that is erased by nRF5FlashSoftDevice.jar?

valeros, 2017-10-12:
I haven't dealt with BL652, but probably you are right and `nRF5FlashSoftDevice.jar` erases the entire MCU flash memory before flashing the softdevice.

closed 2018-03-24 by jeremypoulter

jzumaeta, 2018-08-17:
Hi, I have the same problem with nRF51-DK board. Anyone knows how to solve this issue?