platformio/platform-nordicnrf52 #11: Error uploading to RedBear BLE Nano v2

neu-rah, 2017-11-23:
- Compiles ok
- interface recognized
- arduino ide upload to board with no problem
- i can drag/drop or copy platformio generated .hex file to board, with no problem

however when uploading from platformio ide (atom), even using interface or terminal `pio run -t upload` I have the following error:

```
Info : CMSIS-DAP: Interface ready
Info : reduce speed request: 10000kHz to 5000kHz maximum
Info : clock speed 10000 kHz
Info : SWD DPIDR 0x2ba01477
Info : nrf52.cpu: hardware has 6 breakpoints, 4 watchpoints
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x00000824 msp: 0x20010000
** Programming Started **
embedded:startup.tcl:476: Error: ** Programming Failed **
in procedure 'program'
```

**using:**
- linux ubuntu 16.04 (64 bits)
- atom 1.22.1 x64
- PlatformIO, version 3.5.0b3

**platformio.ini**
```
[env:redbear_blenano2]
platform = nordicnrf52
board = redbear_blenano2
framework = arduino
```

what am i missing?

neu-rah, 2017-12-10:
from platformio terminal i'm able to upload the sketch by copy to the device, it works fine.

`cp .pioenvs/redbear_blenano2/firmware.hex /media/azevedo/DAPLINK`

however if trying to copy to /dev/sde it fails with timeout.

this is the mount info:
/dev/sde on /media/azevedo/DAPLINK type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,showexec,

ivankravets, 2018-08-07:
Could you reproduce the issue with the latest release?

neu-rah, 2018-08-10:
seems fixed :+1: 
thank you

closed 2018-08-10 by neu-rah