platformio/platform-nordicnrf52 #19: Update break the Build

lhcavalcanti, 2018-09-11:
After updating for the new version of the platform for Nordic NRF52 my old codes don't build anymore. 
I'm using VSCode on OSX, but also tried on ubuntu and had the same problem.

Here's the error:
`#include "objects_cryptocell.h"
^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/nrf52840_dk/FrameworkMbedCore/drivers/BusInOut.o] Error 1
In file included from /Users/cavalcanti/.platformio/packages/framework-mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/TARGET_NRF52840_DK/device.h:36:0,
from /Users/cavalcanti/.platformio/packages/framework-mbed/platform/platform.h:29,
from /Users/cavalcanti/.platformio/packages/framework-mbed/drivers/DigitalOut.h:19,
from /Users/cavalcanti/.platformio/packages/framework-mbed/drivers/BusOut.h:19,
from /Users/cavalcanti/.platformio/packages/framework-mbed/drivers/BusOut.cpp:16:
/Users/cavalcanti/.platformio/packages/framework-mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/objects.h:58:10: fatal error: objects_cryptocell.h: No such file or directory`

lhcavalcanti, 2018-09-11:
I downgraded for the version 2.3.2 and the same code builds =0

iceananas, 2018-10-12:
I can confirm this. Something is broken in 2.4.1.

RogerHeil, 2019-09-05:
Hi, got still this problem. :-(

#####################

[env:nrf52840_dk]
platform = https://github.com/platformio/platform-nordicnrf52.git
board = nrf52840_dk
framework = mbed

board_build.mcu = nrf52840
board_build.f_cpu = 64000000L

upload_protocol = jlink

#####################
main.cpp:
#include "mbed.h"

DigitalOut myled(LED1);

int main() {
    while(1) {
        myled = 1;
        wait(1);
        myled = 0;
        wait(1);
    }
}

#####################
Terminal:

> Executing task in folder nRF52840_DK_Test: C:\Users\rheil\.platformio\penv\Scripts\platformio.exe run <

Processing nrf52840_dk (platform: https://github.com/platformio/platform-nordicnrf52.git; board: nrf52840_dk; framework: mbed)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/nrf52840_dk.html
PLATFORM: Nordic nRF52 3.7.0 #3332faf > Nordic nRF52840-DK
HARDWARE: NRF52840 64MHz, 256KB RAM, 1MB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap, jlink) External (blackmagic, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-mbed 5.51304.190826 (5.13.4), tool-sreccat 1.164.0 (1.64)
Collecting mbed sources...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 4 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pio\build\nrf52840_dk\FrameworkMbed\drivers\AnalogIn.o
Compiling .pio\build\nrf52840_dk\FrameworkMbed\drivers\BusIn.o
Compiling .pio\build\nrf52840_dk\FrameworkMbed\drivers\BusInOut.o
Compiling .pio\build\nrf52840_dk\FrameworkMbed\drivers\BusOut.o
In file included from C:/Users/rheil/.platformio/packages/framework-mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/TARGET_NRF52840_DK/device.h:36:0,
                 from C:/Users/rheil/.platformio/packages/framework-mbed/platform/platform.h:30,
                 from C:/Users/rheil/.platformio/packages/framework-mbed/drivers/AnalogIn.h:20,
                 from C:\Users\rheil\.platformio\packages\framework-mbed\drivers\AnalogIn.cpp:18:
C:/Users/rheil/.platformio/packages/framework-mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/objects.h:58:10: fatal error: objects_cryptocell.h: No such file or directory

****************************************************************************
* Looking for objects_cryptocell.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:objects_cryptocell.h"
* Web  > https://platformio.org/lib/search?query=header:objects_cryptocell.h
*
****************************************************************************

 #include "objects_cryptocell.h"
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from C:/Users/rheil/.platformio/packages/framework-mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/TARGET_NRF52840_DK/device.h:36:0,
[... 33 lines cut ...]
 #include "objects_cryptocell.h"
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\nrf52840_dk\FrameworkMbed\drivers\BusInOut.o] Error 1
In file included from C:/Users/rheil/.platformio/packages/framework-mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/TARGET_NRF52840_DK/device.h:36:0,
                 from C:/Users/rheil/.platformio/packages/framework-mbed/platform/platform.h:30,
                 from C:/Users/rheil/.platformio/packages/framework-mbed/drivers/DigitalOut.h:20,
                 from C:/Users/rheil/.platformio/packages/framework-mbed/drivers/BusOut.h:20,
                 from C:\Users\rheil\.platformio\packages\framework-mbed\drivers\BusOut.cpp:17:
C:/Users/rheil/.platformio/packages/framework-mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/objects.h:58:10: fatal error: objects_cryptocell.h: No such file or directory

****************************************************************************
* Looking for objects_cryptocell.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:objects_cryptocell.h"
* Web  > https://platformio.org/lib/search?query=header:objects_cryptocell.h
*
****************************************************************************

 #include "objects_cryptocell.h"
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\nrf52840_dk\FrameworkMbed\drivers\BusOut.o] Error 1
============================================================================================ [FAILED] Took 5.61 seconds ============================================================================================
Der Terminalprozess wurde mit folgendem Exitcode beendet: 1

valeros, 2019-09-05:
So, have you tried adding `build_flags = -DPIO_FRAMEWORK_MBED_RTOS_PRESENT`?

RogerHeil, 2019-09-05:
I add the build flag.... ->

> Executing task in folder nRF52840_DK_Test: C:\Users\rheil\.platformio\penv\Scripts\platformio.exe run <

Processing nrf52840_dk (platform: https://github.com/platformio/platform-nordicnrf52.git; board: nrf52840_dk; framework: mbed)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/nrf52840_dk.html
PLATFORM: Nordic nRF52 3.7.0 #3332faf > Nordic nRF52840-DK
HARDWARE: NRF52840 64MHz, 256KB RAM, 1MB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap, jlink) External (blackmagic, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-mbed 5.51304.190826 (5.13.4), tool-sreccat 1.164.0 (1.64)
Collecting mbed sources...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pio\build\nrf52840_dk\FrameworkMbed\cmsis\TARGET_CORTEX_M\mbed_tz_context.o
arm-none-eabi-gcc: error: CreateProcess: No such file or directory
Compiling .pio\build\nrf52840_dk\FrameworkMbed\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAtmel.o
arm-none-eabi-g++: error: CreateProcess: No such file or directory
Compiling .pio\build\nrf52840_dk\FrameworkMbed\components\802.15.4_RF\atmel-rf-driver\source\at24mac.o
arm-none-eabi-g++: error: CreateProcess: No such file or directory
*** [.pio\build\nrf52840_dk\FrameworkMbed\cmsis\TARGET_CORTEX_M\mbed_tz_context.o] Error 1
*** [.pio\build\nrf52840_dk\FrameworkMbed\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAtmel.o] Error 1
Compiling .pio\build\nrf52840_dk\FrameworkMbed\components\802.15.4_RF\mcr20a-rf-driver\source\MCR20Drv.o
*** [.pio\build\nrf52840_dk\FrameworkMbed\components\802.15.4_RF\atmel-rf-driver\source\at24mac.o] Error 1
arm-none-eabi-gcc: error: CreateProcess: No such file or directory
*** [.pio\build\nrf52840_dk\FrameworkMbed\components\802.15.4_RF\mcr20a-rf-driver\source\MCR20Drv.o] Error 1
================================================================================================ [FAILED] Took 13.58 seconds ================================================================================================
Der Terminalprozess wurde mit folgendem Exitcode beendet: 1

valeros, 2019-09-05:
That's issue probably caused by too long paths in the compiler command line. Could you try to shorten it by installing packages somewhere in root of your logical disk? 
```
[platformio]
packages_dir = C:\\pio-packages

[env:nrf52840_dk]
platform = https://github.com/platformio/platform-nordicnrf52.git
board = nrf52840_dk
framework = mbed
board_build.mcu = nrf52840
board_build.f_cpu = 64000000L
upload_protocol = jlink
```

RogerHeil, 2019-09-05:
[platformio]
packages_dir = E:\\pio\packages ;C:\\pio-packages

[env:nrf52840_dk]
platform = https://github.com/platformio/platform-nordicnrf52.git
board = nrf52840_dk
framework = mbed

board_build.mcu = nrf52840
board_build.f_cpu = 64000000L

upload_protocol = jlink

build_flags = -DPIO_FRAMEWORK_MBED_RTOS_PRESENT

#######################

> Executing task in folder nRF52840_DK_Test: C:\Users\rheil\.platformio\penv\Scripts\platformio.exe run <

Processing nrf52840_dk (platform: https://github.com/platformio/platform-nordicnrf52.git; board: nrf52840_dk; framework: mbed)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/nrf52840_dk.html
PLATFORM: Nordic nRF52 3.7.0 #3332faf > Nordic nRF52840-DK
HARDWARE: NRF52840 64MHz, 256KB RAM, 1MB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap, jlink) External (blackmagic, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-mbed 5.51304.190826 (5.13.4), tool-sreccat 1.164.0 (1.64)
Collecting mbed sources...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pio\build\nrf52840_dk\FrameworkMbed\cmsis\TARGET_CORTEX_M\mbed_tz_context.o
arm-none-eabi-gcc: error: CreateProcess: No such file or directory
Compiling .pio\build\nrf52840_dk\FrameworkMbed\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAtmel.o
Compiling .pio\build\nrf52840_dk\FrameworkMbed\components\802.15.4_RF\atmel-rf-driver\source\at24mac.o
arm-none-eabi-g++: error: CreateProcess: No such file or directory
arm-none-eabi-g++: error: CreateProcess: No such file or directory
*** [.pio\build\nrf52840_dk\FrameworkMbed\cmsis\TARGET_CORTEX_M\mbed_tz_context.o] Error 1
Compiling .pio\build\nrf52840_dk\FrameworkMbed\components\802.15.4_RF\mcr20a-rf-driver\source\MCR20Drv.o
*** [.pio\build\nrf52840_dk\FrameworkMbed\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAtmel.o] Error 1
*** [.pio\build\nrf52840_dk\FrameworkMbed\components\802.15.4_RF\atmel-rf-driver\source\at24mac.o] Error 1
arm-none-eabi-gcc: error: CreateProcess: No such file or directory
*** [.pio\build\nrf52840_dk\FrameworkMbed\components\802.15.4_RF\mcr20a-rf-driver\source\MCR20Drv.o] Error 1
================================================================================================ [FAILED] Took 24.43 seconds ================================================================================================
Der Terminalprozess wurde mit folgendem Exitcode beendet: 1

valeros, 2019-09-05:
Try to build your project in verbose mode (`pio run -v` in the IDE terminal) and attach the build log here.

RogerHeil, 2019-11-18:
Hi, I'm back and got this.....

[platformio]
packages_dir = E:/pio-packages

[env:nrf52840_dk]
platform = https://github.com/platformio/platform-nordicnrf52.git
board = nrf52840_dk
framework = mbed
board_build.mcu = nrf52840
board_build.f_cpu = 64000000L
upload_protocol = jlink

lib_ignore = mbed-mbedtls
build_unflags = -DFEATURE_CRYPTOCELL310=1
build_flags = -DPIO_FRAMEWORK_MBED_RTOS_PRESENT

>>>>>
C:\Users\rheil\Documents\PlatformIO\Projects\IPS_BS2_Probe>pio run -v
Processing nrf52840_dk (platform: https://github.com/platformio/platform-nordicnrf52.git; board: nrf52840_dk; framework: mbed; board_build.mcu: nrf52840; board_build.f_cpu: 64000000L; upload_protocol: jlink; lib_ignore: mbed-mbedtls; build_unflags: -DFEATURE_CRYPTOCELL310=1; build_flags: -DPIO_FRAMEWORK_MBED_RTOS_PRESENT)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/nrf52840_dk.html
PLATFORM: Nordic nRF52 3.8.0 #ef37e8e (git+https://github.com/platformio/platform-nordicnrf52.git) > Nordic nRF52840-DK
HARDWARE: NRF52840 64MHz, 256KB RAM, 1MB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap, jlink) External (blackmagic, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-mbed 5.51401.191023 (5.14.1), tool-sreccat 1.164.0 (1.64)
Collecting mbed sources...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Framework incompatible library C:\Users\rheil\.platformio\lib\Adafruit BMP280 Library_ID528
Framework incompatible library C:\Users\rheil\.platformio\lib\Adafruit GFX Library_ID13
Framework incompatible library C:\Users\rheil\.platformio\lib\Adafruit SSD1306_ID135
Platform incompatible library C:\Users\rheil\.platformio\lib\AsyncTCP_ID1826
Framework incompatible library C:\Users\rheil\.platformio\lib\ESP Async WebServer_ID306
Framework incompatible library C:\Users\rheil\.platformio\lib\ESP32-DHT11_ID5817
Framework incompatible library C:\Users\rheil\.platformio\lib\ESP32_APA102_ID5577
Framework incompatible library C:\Users\rheil\.platformio\lib\ESP8266_SSD1306_ID562
Platform incompatible library C:\Users\rheil\.platformio\lib\ESPAsyncTCP_ID305
Framework incompatible library C:\Users\rheil\.platformio\lib\ESPAsyncUDP_ID359
Framework incompatible library C:\Users\rheil\.platformio\lib\EasyOTA_ID5731
Framework incompatible library C:\Users\rheil\.platformio\lib\FatFs_ID1805
Framework incompatible library C:\Users\rheil\.platformio\lib\LWMQTT library for ESP32 _ESP-IDF__ID5671
Framework incompatible library C:\Users\rheil\.platformio\lib\MQTT library for ESP32 _ESP-IDF__ID5670
Framework incompatible library C:\Users\rheil\.platformio\lib\SparkFun LSM9DS1 IMU_ID1825
Found 0 compatible libraries
Framework incompatible library C:\Users\rheil\.platformio\lib\Time_ID44
Scanning dependencies...
More details about "Library Compatibility Mode": https://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode
No dependencies
Building in release mode
arm-none-eabi-gcc -o .pio\build\nrf52840_dk\FrameworkMbed\cmsis\TARGET_CORTEX_M\mbed_tz_context.o -c -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -g -DMBED_TRAP_ERRORS_ENABLED=1 -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -DMBED_ROM_START=0x0 -DMBED_ROM_SIZE=0x100000 -DMBED_RAM_START=0x20000000 -DMBED_RAM_SIZE=0x40000 -include mbed_config.h -DPLATFORMIO=40100 -DARDUINO_NRF52_DK -DPIO_FRAMEWORK_MBED_RTOS_PRESENT -DARM_MATH_CM4 -DBOARD_PCA10056 -DCMSIS_VECTAB_VIRTUAL -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE=\"cmsis_nvic.h\" -DCOMPONENT_NSPE=1 -DCOMPONENT_PSA_SRV_EMUL=1 -DCOMPONENT_PSA_SRV_IMPL=1 -DCOMPONENT_QSPIF=1 -DCONFIG_GPIO_AS_PINRESET -DDEVICE_ANALOGIN=1 -DDEVICE_FLASH=1 -DDEVICE_I2C=1 -DDEVICE_I2C_ASYNCH=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_ITM=1 -DDEVICE_LPTICKER=1 -DDEVICE_PORTIN=1 -DDEVICE_PORTINOUT=1 -DDEVICE_PORTOUT=1 -DDEVICE_PWMOUT=1 -DDEVICE_QSPI=1 -DDEVICE_SERIAL=1 -DDEVICE_SERIAL_ASYNCH=1 arm-none-eabi-gcc: error: CreateProcess: No such file or directory
-DDEVICE_SERIAL_FC=1 -DDEVICE_SLEEP=1 -DDEVICE_SPI=1 -DDEVICE_SPI_ASYNCH=1 -DDEVICE_SYSTICK_CLK_OFF_DURING_SLEEP=1 -DDEVICE_TRNG=1 -DDEVICE_USBDEVICE=1 -DDEVICE_USTICKER=1 -DFEATURE_BLE=1 -DFEATURE_CRYPTOCELL310=1 -DMBEDTLS_CONFIG_HW_SUPPORT -DMBED_MPU_CUSTOM -DMBED_TICKLESS -DNRF52840_XXAA -DNRF52_ERRATA_20 -DSWI_DISABLE0 -DTARGET_CORDIO -DTARGET_CORDIO_LL -DTARGET_CORTEX -DTARGET_CORTEX_M -DTARGET_FF_ARDUINO -DTARGET_LIKE_CORTEX_M4 -DTARGET_LIKE_MBED -DTARGET_M4 -DTARGET_MCU_NRF52840 -DTARGET_NAME=NRF52840_DK -DTARGET_NORDIC -DTARGET_NORDIC_CORDIO -DTARGET_NRF52 -DTARGET_NRF52840 -DTARGET_NRF52840_DK -DTARGET_NRF5x -DTARGET_RELEASE -DTARGET_RTOS_M4_M7 -DTARGET_SDK_15_0 -DTARGET_SOFTDEVICE_NONE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -DWSF_MAX_HANDLERS=10 -D__CMSIS_RTOS -D__CORTEX_M4 -D__FPU_PRESENT=1 -D__MBED_CMSIS_RTOS_CM -D__MBED__=1 @"C:\Users\rheil\Documents\PlatformIO\Projects\IPS_BS2_Probe\.pio\build\nrf52840_dk\longcmd-96f957bfa42364dda59150145d56b182" E:\pio-packages\framework-mbed\cmsis\TARGET_CORTEX_M\mbed_tz_context.c
[... cut ...]
NE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -DWSF_MAX_HANDLERS=10 -D__CMSIS_RTOS -D__CORTEX_M4 -D__FPU_PRESENT=1 -D__MBED_CMSIS_RTOS_CM -D__MBED__=1 @"C:\Users\rheil\Documents\PlatformIO\Projects\IPS_BS2_Probe\.pio\build\nrf52840_dk\longcmd-96f957bfa42364dda59150145d56b182" E:\pio-packages\framework-mbed\components\802.15.4_RF\mcr20a-rf-driver\source\MCR20Drv.c
=================================================================================================================== [FAILED] Took 9.68 seconds ===================================================================================================================