platformio/platform-nordicnrf52 #28: Adafruit Feather nrf52: WString.h errors on Raspberry Pi

stonehippo, 2019-02-03:
I’ve run into an issue compiling Adafruit Feather nRF52 code when using a Raspberry Pi as a host. The errors occur when compiling `WString.cpp`, and it comes down to not having an implementation of `itoa` or `utoa` available. Here’s an excerpt from trying to run `pio run` on a essentially no-op project:

```
~/.platformio/packages/framework-arduinoadafruitnordicnrf5/cores/nRF5/WString.cpp:307:19: error: 'utoa' was not declared in this scope
```

A few of these crop up, as `WString.cpp` attempts calls to `itoa` and `utoa` a few times.

I ran the same code on a x86 system running Ubuntu, and things work out fine. I did some checking and it seems that on that platform, the GCC arm-none-eabi platform includes the following in `~/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/stdlib.h` which seems to be how the non-standard `itoa` and `utoa` end up getting included:

```
char *  _EXFUN(__itoa,(int, char *, int));
char *  _EXFUN(__utoa,(unsigned, char *, int));
char *  _EXFUN(itoa,(int, char *, int));
char *  _EXFUN(utoa,(unsigned, char *, int));
```

A quick check against the top-level `stdlib.h` in the arm-none-eabi toolchain on the RPi reveals that it’s pretty different from its counterpart on the x86, including no references like the above. So it looks like that leaves things hanging.

I *did* get past this issue, but in a hacky way. I noticed that `WString.cpp` includes `itoa.h` (also in `framework-arduinoadafruitnordicnrf5/cores/NRF5`), and calls what seem to be implementations of methods that look a lot like they might be parallel implementations to `itoa` and `utoa` for `long`, called `ltoa` and `ulota`. To get things to compile, I swapped out the calls to the `int` versions for the `long` versions, and got the build working. 

I know this is most likely an upstream issue, so I’m wondering if this needs to get fixed in the Adafruit BSP (using local implementations of `(u|i)toa`) or if the problem needs to get resolved in the arm-none-eabi, since it’s clearly got something in the build scripts that’s branching (I’m guessing based on the CPU architecture). Thoughts?

stonehippo, 2019-02-03:
I meant to cc @jeremypoulter, since he put in the hard work to get the Feather nRF52 support going in the first place.

stonehippo, 2019-02-03:
@ivankravets the root of this issue seems to be in how `stdlib.h` is getting built for the different CPU arch versions of the toolchain. Based on some looking into Newlib, it seems that `itoa` and `utoa` were added for non-strict ANSI C back in 2014: 
https://sourceware.org/git/gitweb.cgi?p=newlib-cygwin.git;a=commit;h=32c96ddd14acadc82e7fccf110367ec3c8320c85

So I’m guessing that the armv6 version of the toolchain is being built with a config that elides their inclusion. @ivankravets can you point me to the build scripts for the pre-built toolchain for `toolchain-gccarmnoneeabi` in the packages repository on Bintray?

stonehippo, 2019-02-04:
Thanks @jeremypoulter. It looks like the `arm-none-eabi-gcc` included in the 1.60301.0 package is pretty old, version 4.8.3. On the Linux x86_64 arch, the newer 1.70201.0 `toolchain-gccarmnoneeabi` package is getting installed. It looks like that version didn’t get built/deployed to Bintray for armv6/v7 arch. So this is a broader issue the just the nrf52 port.

Should asking for an newer toolchain package go against the core repo as a new issue, @ivankravets? Seems like I should close this one, since it’s not really a platform level issue.

stonehippo, 2019-02-04:
@ivankravets probably. I don’t know enough about how the packages for PlatformIO get built for Bintray, so I’d want to how that works (for example, I assume that the packages are built from the source tarballs at 
https://developer.arm.com/open-source/gnu-toolchain/gnu-rm, but it’s not clear to me where the package semver comes from). I’d be happy to take it on, if you can point me to the docs on how it’s managed.

stonehippo, 2019-02-05:
@ivankravets does Bintray have a history feature? I think I see how the package semver is supposed to work, where the package minor version is actually the semver of the gcc package. If that’s correct, then looking at the package list, it looks like perhaps the 1.60301.0 toolchain-gccarmnoneeabi package, which I would expect to contain arm-none-gcc-eabi 6.3.1 got overridden with the 1.40803.0 package. Looking at the file list in Bintray, the packages are the same size and seem to have the same upload date. See 
https://bintray.com/platformio/dl-packages/toolchain-gccarmnoneeabi#files

If I’ve got this right, I think I how to proceed. I’m going to build the arm-none-eabi-gcc toolchain using the source tarballs for 6.3.1 and 7.2.1, to match what’s available for other platforms. The new packages should contain the toolchain + `package.json` metadata. Does that sound right? And if I do get those built, how should I hand them off to deploy to Bintray?

ivankravets, 2019-02-05:
Please point me to the precompiled toolchain and I'll deploy it. I don't see here binaries for ARMv6 ISA and up.

stonehippo, 2019-02-10:
@ivankravets quick update: I’m in the process of compiling arm-none-eabi-gcc toolchains for the RPis now. It’s taking some time, since I’ve got to run it on RPi3 and that’s not the fastest host. :-)

I’m going to do both the last major 6 and the 7 major updates. I think these will match the versions deployed on Bintray for other platforms.

stonehippo, 2019-02-14:
@ivankravets I finally finished building version 7.2.1 of the arm-none-eabi-gcc toolchain for armv6/Raspberry Pi. I’ve given it a try and it resolved the `itoa`/`utoa` issues I got with current toolchain. I checked the .hex produced against the .hex for the same project on an x86_64 CPU and there’s no diff, so I think it’s ready to go.

I’ve added a package.json to the whole thing and tarred it up with a name format that follows the other packages on Bintray. You can grab it from here:

http://x.stonehippo.com/toolchains/toolchain-gccarmnoneeabi-linux_armv6l-1.70201.0.tar.gz

I’m going to build the 6.3.1 toolchain next, so you can replace the broken version on Bintray, too.

ivankravets, 2019-02-15:
@stonehippo thank you so much for your contribution! I've just deployed a package to our CDN network. Could you re-test it?

P.S: I think we don't need `6.3.1 toolchain`

Please re-open if it does not work.

closed 2019-02-15 by ivankravets

stonehippo, 2019-02-15:
@iankravets things look pretty good. 1.7020.1 downloaded and code compiled with warnings only, no errors.

I'll skip he 1.6030.1 build. Thanks!