platformio/platform-nordicnrf52 #29: Adafruit nRF52 Feather: create target to update bootloader/SoftDevice for nrf52832

stonehippo, 2019-03-03:
The Adafruit nRF52 Feather with the nrf52832 chip (but _not_ the nrf52840) requires a bootloader update to be compatible with the Adafruit BSP included here. The bootloader includes an update for the nrf52832 SoftDevice. Without this update, attempts to update a feather via `pio run -t upload` will fail with the following message:

```
Timed out waiting for acknowledgement from device.

Failed to upgrade target. Error is: No data received on serial port. Not able to proceed.
Traceback (most recent call last):
  File "nordicsemi\__main__.py", line 294, in serial
  File "nordicsemi\dfu\dfu.py", line 235, in dfu_send_images
  File "nordicsemi\dfu\dfu.py", line 203, in _dfu_send_image
  File "nordicsemi\dfu\dfu_transport_serial.py", line 155, in send_init_packet
  File "nordicsemi\dfu\dfu_transport_serial.py", line 243, in send_packet
  File "nordicsemi\dfu\dfu_transport_serial.py", line 282, in get_ack_nr
nordicsemi.exceptions.NordicSemiException: No data received on serial port. Not able to proceed.
```

Adafruit describes this issue here: 
https://learn.adafruit.com/bluefruit-nrf52-feather-learning-guide/arduino-board-setup#4-run-a-test-sketch-6-12

And provides instructions for updating the bootloader via the Arduino IDE here. They also nicely describe the `adafruit-nrfutil` script used to push the new bootloader here: 
https://learn.adafruit.com/bluefruit-nrf52-feather-learning-guide/updating-the-bootloader

The BSP package include here does have an updated bootloader that can be applied, which can be done in the following way:

```sh
$ cd ~/.plartformio/packages/framework-arduinoadafruitnrf52/bootloader/feather_nrf52832/
$ adafruit-nrfutil --verbose dfu serial --package feather_nrf52832_bootloader-0.2.6_s132_6.1.1.zip -p /dev/ttyUSB0 -b 115200 --singlebank --touch 1200
```
Once applied, the upload target will work as expected.

I propose adding a new target for updating the bootloader, `pio run -t bootloader` that will apply this update on request. Thoughts?

/cc @jeremypoulter 

P.S. It looks like Adafruit has updated the BSP, and included a newer bootloader image than the one here. I’ve tried them both and either one will work with this package in PIO. I’m wondering about the process for keeping the BSP here synced with the upstream, though.

jeremypoulter, 2019-03-03:
Already done: https://github.com/platformio/platform-nordicnrf52/blob/develop/builder/main.py#L189

stonehippo, 2019-03-03:
@jeremypoulter Great!

Unfortunately, I don’t think it’s universal, since `nrfprog` doesn’t have a binary available on all platforms. For example, on Raspberry Pi, the current `pio run -t bootloader` leads to the following error:

```
PackageManager: Installing tool-nrfjprog @ ~1.90702.0
Error: Could not find a version that satisfies the requirement '~1.90702.0' for your system 'linux_armv7l'
```

Might be able to find a version of the tool that runs on armv6/7. I’ll do some digging. If not, how do you feel about either swapping out for `adafruit-nrfutil`, or perhaps providing a way to select the bootloader protocol, as is done with the `upload` target?

jeremypoulter, 2019-03-03:
Ah right, that is true. There is a degree of danger in using nrfutil as it can 'brick' the board if the OTA fails (and you would need to update via a J-Link), so it does need to be configurable. 

Probably should just use whatever the configured uploader is. Think that should be possible, but should also support upload via the J-Link tools in addition to nrfutil and nrfjprog

stonehippo, 2019-03-03:
I think binding the bootloader upload protocol to the app uploaded protocol selection makes sense.

Adafruit defaults to nrfutil, so they probably consider it stable enough. At the same time, it makes sense to support all three methods.

nsamala, 2019-03-15:
@jeremypoulter @stonehippo the original code when I added the Adafruit BSP originally to PIO used the `adafruit-nrfutil` and worked perfectly fine. It shouldn't be too crazy to add that back in

stonehippo, 2019-05-01:
Thanks @nsamala. I’ve been away from this for a bit, but I’ll dig up your code and do a PR for this some point soon.

mauriezh, 2019-07-24:
Hello @stonehippo , thanks for setting this help page up. 
I'm having a problem updating the bootloader though. 

Having a problem running my original code that I tested last week, which worked on multiple days last week.

I tried to do a DFU reset and update the BootLoader, but it didn't work. I checked the baudrate and board selections in the Tools tab, but those were fine.

I didn't change anything in the code, I checked the wires on the hardware setup (nothing looks loose).

Thought it could be an error with the USB cable, changed the USB cable to a different one, still same error.

Also - using a Mac. But the same issue arose on the Windows PC. Did not try to reset using the Windows computer, but reset issue has more to do with the other hardware?
[image]

stonehippo, 2019-07-24:
@mauriezh I’m not sure what you’re asking. Looking at your screencap, I’m going to guess that you’re using the upstream version of the package (based on the what I can see of the firmware file, it seems to be `feather_nrf52832_bootloader-0.2.11_s132_6.1.1.zip`). I’ve found that I’m getting some compilation errors with upstream (which may to be due to the tinyusb upgrade in the latest Adafruit board support update: https://github.com/adafruit/Adafruit_nRF52_Arduino/releases/tag/0.11.1).

That said, I stepped back to the 3.5.0 release of this package, and tried an OTA update using steps I described before and the `feather_nrf52832_bootloader-0.2.9_s132_6.1.1.zip` firmware provided in that version. My feather nrf52832 updated no issues. I then jumped back to upstream, and tried it with the newer firmware. Again, no issues:

```
package feather_nrf52832_bootloader-0.2.11_s132_6.1.1.zip -p /dev/ttyUSB0 -b 115200 --singlebank --touch 1200
Upgrading target on /dev/ttyUSB0 with DFU package /home/george_w/.platformio/packages/framework-arduinoadafruitnrf52/bootloader/feather_nrf52832/feather_nrf52832_bootloader-0.2.11_s132_6.1.1.zip. Flow control is disabled, Single bank, Touch 1200
Touched serial port /dev/ttyUSB0
Opened serial port /dev/ttyUSB0
Starting DFU upgrade of type 3, SoftDevice size: 147792, bootloader size: 20284, application size: 0
Sending DFU start packet
Sending DFU init packet
Sending firmware file
########################################
########################################
########################################
########################################
########################################
########################################
########################################
########################################
#########
Activating new firmware

DFU upgrade took 37.63793992996216s
Device programmed.
```

I can give this a try on a MacBook later, but I can’t think of any reason it should fail there, given the basic tools being used.

You might want to try the Arduino IDE with the latest Adafruit BSP for nrf52 installed and see if you can run the update that way. If that doesn’t work, you might have a bad board.

stonehippo, 2019-07-24:
@mauriezh one other thought: are you using a Feather nrf52 (with an nrf52832 onboard) or the Feather nrf52840? The firmware procedure here is only meant for the former board (unless Adafruit has changed something recently).

mauriezh, 2019-07-24:
Actually realized what the issue was. Had the wrong port selected. Step Zero. 

Thanks , appreciate it.

mauriezh, 2019-07-24:
Actually realized what the issue was. Had the wrong port selected. Step
Zero.

Thanks though, appreciate it.

*Maurie Zhang*
T:  857 - 283-5030
E: m <mmchrist@bu.edu>auriezh@bu.edu