platformio/platform-nordicnrf52 #32: Missing objects_cryptocell.h

antoniotorres, 2019-03-10:
Blink example cannot compile with the following configurations. Compilation shows missing objects_cryptocell.h, and looks related to [`Issue #19`](https://github.com/platformio/platform-nordicnrf52/issues/19). Code works with version 2.5.0. 

Using board NRF52480.

platformio.ini
```
[env:nrf52_dk]
platform = https://github.com/platformio/platform-nordicnrf52.git
board = nrf52840_dk
framework = mbed
```

PlatformIO -> Build
```bash
Processing nrf52_dk (platform: https://github.com/platformio/platform-nordicnrf52.git; board: nrf52840_dk; framework: mbed)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/nrf52840_dk.html
PLATFORM: Nordic nRF52 > Nordic nRF52840-DK
HARDWARE: NRF52840 64MHz 256KB RAM (1MB Flash)
DEBUG: CURRENT(cmsis-dap) ON-BOARD(cmsis-dap, jlink) EXTERNAL(blackmagic, stlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 4 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pioenvs/nrf52_dk/FrameworkMbed/cmsis/TARGET_CORTEX_M/mbed_fault_handler.o
Compiling .pioenvs/nrf52_dk/FrameworkMbed/drivers/AnalogIn.o
Compiling .pioenvs/nrf52_dk/FrameworkMbed/drivers/BusIn.o
Compiling .pioenvs/nrf52_dk/FrameworkMbed/drivers/BusInOut.o
Compiling .pioenvs/nrf52_dk/FrameworkMbed/drivers/BusOut.o
Compiling .pioenvs/nrf52_dk/FrameworkMbed/drivers/CAN.o
In file included from /Users/antonio/.platformio/packages/framework-mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/TARGET_NRF52840_DK/device.h:36:0,
from /Users/antonio/.platformio/packages/framework-mbed/cmsis/TARGET_CORTEX_M/mbed_fault_handler.c:23:
/Users/antonio/.platformio/packages/framework-mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/objects.h:58:10: fatal error: objects_cryptocell.h: No such file or directory

****************************************************************************
* Looking for objects_cryptocell.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:objects_cryptocell.h"
* Web  > https://platformio.org/lib/search?query=header:objects_cryptocell.h
*
****************************************************************************

#include "objects_cryptocell.h"
^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pioenvs/nrf52_dk/FrameworkMbed/drivers/Ethernet.o
In file included from /Users/antonio/.platformio/packages/framework-mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/TARGET_NRF52840_DK/device.h:36:0,
from /Users/antonio/.platformio/packages/framework-mbed/platform/platform.h:30,
from /Users/antonio/.platformio/packages/framework-mbed/drivers/DigitalInOut.h:20,
from /Users/antonio/.platformio/packages/framework-mbed/drivers/BusInOut.h:20,
from /Users/antonio/.platformio/packages/framework-mbed/drivers/BusInOut.cpp:17:
/Users/antonio/.platformio/packages/framework-mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/objects.h:58:10: fatal error: objects_cryptocell.h: No such file or directory

****************************************************************************
* Looking for objects_cryptocell.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:objects_cryptocell.h"
* Web  > https://platformio.org/lib/search?query=header:objects_cryptocell.h
*
****************************************************************************

#include "objects_cryptocell.h"
^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /Users/antonio/.platformio/packages/framework-mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/TARGET_NRF52840_DK/device.h:36:0,
from /Users/antonio/.platformio/packages/framework-mbed/platform/platform.h:30,
from /Users/antonio/.platformio/packages/framework-mbed/drivers/AnalogIn.h:20,
from /Users/antonio/.platformio/packages/framework-mbed/drivers/AnalogIn.cpp:18:
[... 144 lines cut ...]
#include "objects_cryptocell.h"
^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/nrf52_dk/FrameworkMbed/drivers/I2CSlave.o] Error 1
In file included from /Users/antonio/.platformio/packages/framework-mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/TARGET_NRF52840_DK/device.h:36:0,
from /Users/antonio/.platformio/packages/framework-mbed/platform/platform.h:30,
from /Users/antonio/.platformio/packages/framework-mbed/drivers/InterruptIn.h:20,
from /Users/antonio/.platformio/packages/framework-mbed/drivers/InterruptIn.cpp:17:
/Users/antonio/.platformio/packages/framework-mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/objects.h:58:10: fatal error: objects_cryptocell.h: No such file or directory

****************************************************************************
* Looking for objects_cryptocell.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:objects_cryptocell.h"
* Web  > https://platformio.org/lib/search?query=header:objects_cryptocell.h
*
****************************************************************************

#include "objects_cryptocell.h"
^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/nrf52_dk/FrameworkMbed/drivers/InterruptIn.o] Error 1
=================================================================================== [ERROR] Took 2.74 seconds ===================================================================================
The terminal process terminated with exit code: 1
```

valeros, 2019-03-11:
Hi @antoniotorres ! Thanks for reporting. Could you please try to compile your project with enabled OS?
```
[env:nrf52_dk]
platform = https://github.com/platformio/platform-nordicnrf52.git
board = nrf52840_dk
framework = mbed
build_flags = -DPIO_FRAMEWORK_MBED_RTOS_PRESENT
```

antoniotorres, 2019-03-13:
@valeros I confirm Build and Upload work with that build_flag.

hansipete, 2019-08-12:
same problem here but fixed by adding `build_flags = -DPIO_FRAMEWORK_MBED_RTOS_PRESENT
` to .ini.

thank you!

valeros, 2019-09-02:
Feel free to reopen the issue if you still have this problem.

closed 2019-09-02 by valeros

FLXCompanydesings, 2020-05-13:
I am trying to compile a simple led flash. I've added the flags as recommended: @valeros
but i get following errors. any possible solution?

```
> Compiling .pio\build\nrf52_dk\FrameworkMbed\cmsis\TARGET_CORTEX_M\mbed_tz_context.o
> arm-none-eabi-gcc: error: CreateProcess: No such file or directory
> Compiling .pio\build\nrf52_dk\FrameworkMbed\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAtmel.o
> arm-none-eabi-g++: error: CreateProcess: No such file or directory
> Compiling .pio\build\nrf52_dk\FrameworkMbed\components\802.15.4_RF\atmel-rf-driver\source\at24mac.o
> arm-none-eabi-g++: error: CreateProcess: No such file or directory
> Compiling .pio\build\nrf52_dk\FrameworkMbed\components\802.15.4_RF\mcr20a-rf-driver\source\MCR20Drv.o
> arm-none-eabi-gcc: error: CreateProcess: No such file or directory
> Compiling .pio\build\nrf52_dk\FrameworkMbed\components\802.15.4_RF\mcr20a-rf-driver\source\NanostackRfPhyMcr20a.o
> arm-none-eabi-g++: error: CreateProcess: No such file or directory
> Compiling .pio\build\nrf52_dk\FrameworkMbed\components\802.15.4_RF\stm-s2lp-rf-driver\source\NanostackRfPhys2lp.o
> arm-none-eabi-g++: error: CreateProcess: No such file or directory
> Compiling .pio\build\nrf52_dk\FrameworkMbed\components\802.15.4_RF\stm-s2lp-rf-driver\source\at24mac_s2lp.o
> Compiling .pio\build\nrf52_dk\FrameworkMbed\components\802.15.4_RF\stm-s2lp-rf-driver\source\rf_configuration.o
> *** [.pio\build\nrf52_dk\FrameworkMbed\cmsis\TARGET_CORTEX_M\mbed_tz_context.o] Error 1
> *** [.pio\build\nrf52_dk\FrameworkMbed\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAtmel.o] Error 1
> arm-none-eabi-g++: error: CreateProcess: No such file or directory
> *** [.pio\build\nrf52_dk\FrameworkMbed\components\802.15.4_RF\atmel-rf-driver\source\at24mac.o] Error 1
> *** [.pio\build\nrf52_dk\FrameworkMbed\components\802.15.4_RF\mcr20a-rf-driver\source\MCR20Drv.o] Error 1
> *** [.pio\build\nrf52_dk\FrameworkMbed\components\802.15.4_RF\mcr20a-rf-driver\source\NanostackRfPhyMcr20a.o] Error 1
> *** [.pio\build\nrf52_dk\FrameworkMbed\components\802.15.4_RF\stm-s2lp-rf-driver\source\NanostackRfPhys2lp.o] Error 1
> *** [.pio\build\nrf52_dk\FrameworkMbed\components\802.15.4_RF\stm-s2lp-rf-driver\source\at24mac_s2lp.o] Error 1
> arm-none-eabi-gcc: error: CreateProcess: No such file or directory
> *** [.pio\build\nrf52_dk\FrameworkMbed\components\802.15.4_RF\stm-s2lp-rf-driver\source\rf_configuration.o] Error 1
> ================================================================ [FAILED] Took 12.15 seconds =================================================================
```

```

valeros, 2020-05-14:
@FLXCompanydesings ! Try to add the following lines at the top your `platformio.ini`:
```
[platformio]
core_dir = C:\\.pio
```