platformio/platform-nordicnrf52 #35: Adafruit Feather NRF52832 function delay() causes crash.

Splitframe, 2019-03-20:
The following code crashes with the Adafruit Feather NRF52832 and "Test B" is not shown.
A possible workaround is to use delayMicroseconds().

```
#include <main.h>

int main() {
  Serial.begin(9600);
  Serial.println(F("Test A"));
  delay(100);
  Serial.println(F("Test B"));
}
```

jeremypoulter, 2019-03-20:
I am supprised that even compiles ;-) the Adafruit BSP is an Arduino core and as such you should use `setup` and `loop` not `main`.

The Adafruit BSP uses FreeRTOS under the hood the `main` in the BSP launches a task and `setup` and `loop` are call in that task. `delay` then puts that task to sleep.

Calling it like you have there the RTOS is not setup nor are you calling `delay` in task.

Splitframe, 2019-03-20:
Oh! I am sorry,
I wasn't aware of this. I used to just pump normal cpp into arduinos.
Thank you.

closed 2019-03-20 by Splitframe