platformio/platform-nordicnrf52 #41: Adafruit Feather NRF52 BSP update to 0.10.1?

wb8wka, 2019-04-23:
When will the PlatformIO BSP for the Adafruit NRF52 be updated from 0.9.3 to 0.10.1?

jeremypoulter, 2019-04-24:
@ivankravets What is the best process for updating? I assume you need to create a new package of the Adafruit BSP then a PR can be created with the required updates?

ivankravets, 2019-04-24:
@jeremypoulter done! https://bintray.com/beta/#/platformio/dl-packages/framework-arduinoadafruitnrf52/1.1001.0?tab=files

Please note that new boards were added in the new release. It would be good to have them in PlatformIO too.

wb8wka, 2019-04-24:
@ivankravets wow thank you.   Excuse the newbie question, as day before yesterday was my first use of PlatformIO, but I'd just replace this directory:

C:\Users\kingr\.platformio\packages\framework-arduinoadafruitnrf52

with this: framework-arduinoadafruitnrf52-1.1001.0.tar.gz\  **framework-arduinoadafruitnrf52** 
 -1.1001.0.tar\      (keeping the same directory structure and name)

ivankravets, 2019-04-24:
> PlatformIO, but I'd just replace this directory:

Does it work?

wb8wka, 2019-04-24:
I'll be trying it later this evening, I will let you know.

wb8wka, 2019-04-26:
@ivankravets 

Sorry for the delay. I deleted everything in:

C:\Users\kingr.platformio\packages\framework-arduinoadafruitnrf52

and then replaced it with the contents of the tar.gz file you attached.

then did a build and all seemed fine but I noticed it created a director that contained the 0.9.3 BSP:

C:\Users\kingr\.platformio\packages\framework-arduinoadafruitnrf52@1.903.0

and on closer examination was building with 0.9.3.  I deleted that directory and did a clean and it ended up downloading the 0.9.3 files again and recreating that directory.

wb8wka, 2019-04-26:
Oh, and this was with core 3.6.6

wb8wka, 2019-04-27:
Any thoughts?

jeremypoulter, 2019-04-27:
Was looking at this yesterday, there are a few changes needed in the Platform IO side of things. Update to get the new package and currently the version numbers are hard coded, doing an update to read from the framework. Plus adding the additional boards ;-)

wb8wka, 2019-04-30:
So perhaps in the next release of PlatformIO?  I don't need the features of 0.10.1 at this point, was more curious if it could be incorporated sooner.

ivankravets, 2019-05-04:
Please re-test with upstream version https://docs.platformio.org/en/latest/platforms/nordicnrf52.html#upstream

closed 2019-05-04 by ivankravets

wb8wka, 2019-05-04:
@ivankravets yes, that works. Had to reinstall GitCLient and restart platformIO but worked fine after that.

Thanks!

PR #43 (2019-05-04): Update to the 0.10.1 BSP
Fixes #41 

(Sorry I haven't had a chance to add the new boards)