platformio/platform-nordicnrf52 #45: Adafruit Feather nRF52: 3.4.0 ignores upload_protocol in platformio.ini

stonehippo, 2019-05-13:
It looks like the 3.4.0 release includes a change that breaks the use the `upload_protocol` setting in `platform.ini`. For example, with 3.3.0, this ini:

```
[env:adafruit_feather_nrf52832]
platform = nordicnrf52
board = adafruit_feather_nrf52832
framework = arduino
upload_protocol = nrfutil
```

Works fine and uses `nrfutil` as expected. However, with 3.4.0, that same config attempts to download `nrfjprog`:

```
PackageManager: Installing tool-nrfjprog @ ~1.90702.0
Error: Could not find a version that satisfies the requirement '~1.90702.0' for your system 'linux_armv7l'
```

I’m typically running on a Raspberry Pi, where `nrfjprog` isn’t available, which is why I use `nrfutil`. I took a look through the commits since 3.3.0 and it seems like the culprit is probably in a93b267, with was to address #34. That commit removed setting `upload_protocol` in `platform.py`, which seems like an unexpected side effect of the erase implementation.

/cc @ivankravets

stonehippo, 2019-05-13:
@ivankravets I have confirmed that the issue was the change in a93b267. I reversed the changes in `platform.py` and nrfutil is used as expected.

I’m not sure what that change was supposed to accomplish, so I won’t attempt a fix and PR. If I know what the intent was, I can try to make a fix for it.

ivankravets, 2019-05-14:
Sorry, fixed! Please re-test with upstream version http://docs.platformio.org/en/latest/platforms/nordicnrf52.html#upstream

closed 2019-05-14 by ivankravets

stonehippo, 2019-05-27:
@ivankravets I’ve tested with upstream and things look good. Thanks!

ivankravets, 2019-05-27:
Thanks! There is also final release https://github.com/platformio/platform-nordicnrf52/releases

commit 828b8eff (2019-05-14): Do not require `tool-nrfjprog` if it will not be used // Resolve #45