platformio/platform-nordicnrf52 #47: Adafruit Feather nrF52 build error: adafruit-nrfutil not found on ubuntu 18.04

neu-rah, 2019-05-22:
https://github.com/platformio/platform-nordicnrf52/issues/26

Have same problem on linux (ubuntu 18.04), on the script it seems that other OS rely on the tool being on path...

from this PR: https://github.com/platformio/platform-nordicnrf52/pull/27

on `~/.platformio/platforms/nordicnrf52/builder/main.py`
nrfutil_path = join(FRAMEWORK_DIR, "tools", "adafruit-nrfutil", os_platform, "adafruit-nrfutil")

would serve both systems "macos" and "linux2"

os_platform="linux2"
created `linux2` folder on `~.platformio/packages/framework-arduinoadafruitnrf52/tools/adafruit-nrfutil`

linked `adafruit-nrfutil` to `~/.local/bin/adafruit-nrfutil` 

`ln ~/.local/bin/adafruit-nrfutil`

that is the correct path on my system when following install procedures from https://github.com/adafruit/Adafruit_nRF52_nrfutil
`pip3 install --user adafruit-nrfutil`

I could patch this (not sure if it worth, because it can vary from system to system?), anyway the `linux2` folder would be missing along with the link, and i have no idea where in the toolchain install process this can be done

int3ll3ct, 2019-09-10:
I think clearer instructions in the Linux section at https://learn.adafruit.com/bluefruit-nrf52-feather-learning-guide/arduino-bsp-setup would suffice here, particularly the line about "Add pip3 installation dir to your PATH if it is not added already. Make sure adafruit-nrfutil can be executed in terminal by running..." 
For the neophyte, going from "Add pip3 installation dir to your PATH" to the real work required - for example, in Ubuntu 18.04.3 I edited my ~/.profile file and added:

`# add pip3 site packages folder to PATH for adafruit-nrfutil
PATH="$PATH:$HOME/.local/bin"`

Which did the trick. Posting for posterity.

closed 2019-09-10 by neu-rah

ametedinov, 2019-09-19:
This fixes the issue on Windows! But on Linux, I am getting a different kind of error: 

Multiple libraries were found for "Adafruit_LittleFS.h"
 Used: /home/my/.arduino15/packages/adafruit/hardware/nrf52/0.11.0/libraries/Adafruit_LittleFS
Multiple libraries were found for "InternalFileSystem.h"
 Used: /home/my/.arduino15/packages/adafruit/hardware/nrf52/0.11.0/libraries/InternalFileSytem
Multiple libraries were found for "bluefruit.h"
 Used: /home/my/.arduino15/packages/adafruit/hardware/nrf52/0.11.0/libraries/Bluefruit52Lib
Multiple libraries were found for "Servo.h"
 Used: /home/my/.arduino15/packages/adafruit/hardware/nrf52/0.11.0/libraries/Servo
 Not used: /home/my/Desktop/arduino-1.8.10/libraries/Servo
Multiple libraries were found for "Wire.h"
 Used: /home/my/.arduino15/packages/adafruit/hardware/nrf52/0.11.0/libraries/Wire
Multiple libraries were found for "Firmata.h"
 Used: /home/my/Desktop/arduino-1.8.10/libraries/Firmata
exec: "adafruit-nrfutil": executable file not found in $PATH
Error compiling for board Adafruit Bluefruit Feather nRF52832.