platformio/platform-nordicnrf52 #48: Argument list too long

janhaa, 2019-05-24:
I consistently get the following when trying to use mbed through platform.io:

```
Processing nrf52_dk (platform: nordicnrf52; board: nrf52_dk; framework: mbed)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/nrf52_dk.html
PLATFORM: Nordic nRF52 > Nordic nRF52-DK
HARDWARE: NRF52832 64MHz 64KB RAM (512KB Flash)
DEBUG: CURRENT(cmsis-dap) ON-BOARD(cmsis-dap, jlink) EXTERNAL(blackmagic, stlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 5 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <mbed-os>
Compiling .pioenvs/nrf52_dk/src/main.o
Compiling .pioenvs/nrf52_dk/libd6b/mbed-os_ID2499/TESTS/events/queue/main.o
Compiling .pioenvs/nrf52_dk/libd6b/mbed-os_ID2499/TESTS/integration/basic/main.o
*** [.pioenvs/nrf52_dk/src/main.o] Argument list too long
*** [.pioenvs/nrf52_dk/libd6b/mbed-os_ID2499/TESTS/events/queue/main.o] Argument list too long
Compiling .pioenvs/nrf52_dk/libd6b/mbed-os_ID2499/TESTS/mbed_drivers/c_strings/main.o
*** [.pioenvs/nrf52_dk/libd6b/mbed-os_ID2499/TESTS/integration/basic/main.o] Argument list too long
*** [.pioenvs/nrf52_dk/libd6b/mbed-os_ID2499/TESTS/mbed_drivers/c_strings/main.o] Argument list too long
```

gcc -v:
```
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 7.4.0-1ubuntu1~18.04' --with-bugurl=file:///usr/share/doc/gcc-7/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-7 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 7.4.0 (Ubuntu 7.4.0-1ubuntu1~18.04) 
```

valeros, 2019-05-24:
Hi @janhaa! Thanks for the report. Can you please clarify why you need a separate `mbed-os` library if you are already using the mbed framework?

janhaa, 2019-06-03:
Hi @valeros, I haven't further persueded the mbed approach (but would maybe like to at later stages again). So I can't exactly remember right now, but I think this was just a basic example I tried to run, no changes to it.

neu-rah, 2019-09-07:
if it helps... i got "Argument list too long" when https://platformio.org/lib/show/2499/mbed-os?utm_source=platformio&utm_medium=piohome is installed on global (message goes away if removing).

my ini:
```ini
[platformio]

[env:AquagrowNRF51]
platform = nordicnrf51
framework = mbed
...
```

valeros, 2019-10-03:
Should be fixed in the latest `platformio-core`

closed 2019-10-03 by valeros

rudydevolder, 2019-10-25:
same problem here with the latest pio-core

the-real-blackh, 2019-12-09:
A patch to fix this link error when building with PlatformIO on Linux:
*** [.pio/build/nrf52840_dk/firmware.elf] sh: Argument list too long

--- .platformio/packages/tool-scons/engine/SCons/Action.py.orig	2019-12-10 10:37:28.620842014 +1300
+++ .platformio/packages/tool-scons/engine/SCons/Action.py	2019-12-10 11:07:02.572349238 +1300
@@ -107,6 +107,7 @@
 import subprocess
 import itertools
 import inspect
+import tempfile
 from collections import OrderedDict
 
 import SCons.Debug
@@ -949,7 +950,15 @@
         for cmd_line in filter(len, cmd_list):
             # Escape the command line for the interpreter we are using.
             cmd_line = escape_list(cmd_line, escape)
-            result = spawn(shell, escape, cmd_line[0], cmd_line, ENV)
+            tempFn = tempfile.mkstemp()
+            try:
+                with open(tempFn[0], "w") as text_file:
+                    text_file.write(' '.join(cmd_line))
+                proc = subprocess.Popen([shell, tempFn[1]], env = ENV, close_fds = True)
+                result = proc.wait()
+            finally:
+                if os.path.isfile(tempFn[1]):
+                    os.remove(tempFn[1])
             if not ignore and result:
                 msg = "Error %s" % result
                 return SCons.Errors.BuildError(errstr=msg,