platformio/platform-nordicnrf52 #50: nRF52810 support

vladkozlov69, 2019-06-21:
Any plans to add support for nRF52810-based boards? 
For example, http://www.ebyte.com/en/product-view-news.aspx?id=587

silardg, 2019-11-13:
Any update on this?

valeros, 2019-11-13:
What framework do you plan to use with this boards?

vladkozlov69, 2019-11-13:
Arduino or mbed, what is available first.

valeros, 2019-11-13:
If I'm not mistaken none of these frameworks support nRF52810 MCU.

silardg, 2019-11-17:
Is there a plan to have it supported? I mean it is quite a common used MCU. Mostly for budget things, because ebyte implements it.

valeros, 2019-11-18:
If I'm not mistaken only Zephyr framework supports this board, so probably we'll add it sooner or later. How do you upload code to this board?

silardg, 2019-11-27:
Just a simple ST-LINK V2.

carminevalentino, 2020-10-03:
According to the nrf52 datasheet, nrf52810 is architecturally identical to the nrf52832 with the exception of the flash and ram size. As a result, I have been able to successfully flash the board using J-LINK with the following platform.ini settings:

`; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:nrf52_dk]
platform = nordicnrf52
board = nrf52_dk
framework = arduino
board_build.mcu = nrf52810
board_upload.maximum_ram_size = 24575
board_upload.maximum_size = 196607
board_debug.jlink_device = NRF52810_XXAA`

maxgerhardt, 2020-11-19:
@valeros  this was asked in https://community.platformio.org/t/does-platformio-support-nordics-nrf52810/17264 again

maxgerhardt, 2021-09-20:
@valeros this was asked again in https://community.platformio.org/t/nrf52810-support/23529

bojankoce, 2021-09-20:
Hello, guys! Anything new concerning nRF52810 support?

maxgerhardt, 2021-09-21:
@bojankoce What exact dev board are you using? Zephyr only knows https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/nrf52dk_nrf52810/doc/index.rst and https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/ubx_bmd330eval_nrf52810/doc/index.rst. 

Note that native SDK support from Nordic is not integrated in PlatformIO for any nRF board, and since no other framework supports a nRF52810 *but* Zephyr, that's the only sensible integration as of now.

ErikHougaard, 2021-09-23:
I too would certainly be interested in NRF52810 support. https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/nrf52dk_nrf52810/doc/index.rst looks like a good choice for me, and I'd happily switch to Zephyr.