platformio/platform-nordicnrf52 #53: Support for Arduino Nano 33 BLE (Sense)

manuelbl, 2019-08-12:
Now that the [Arduino Nano 33 BLE](https://store.arduino.cc/nano-33-ble) and [Arduino Nano 33 BLE Sense](https://store.arduino.cc/nano-33-ble-sense) have been released and introduce:

- First genuine Arduino board based on Nordic nRF52840
- [Arduino core based on Mbed OS framework](https://blog.arduino.cc/2019/07/31/why-we-chose-to-build-the-arduino-nano-33-ble-core-on-mbed-os/)

Are there any plans to bring them to PlatformIO? What will be needed? How can it be supported?

clwillingham, 2019-09-14:
What would be needed to add support and close this issue? We're designing a nano 33 shield, and I would love to add PlatformIO support to the library. 
Is it a matter of adding a configuration file to the boards folder? or are there other changes that need to be made?
If there is any way we can help add the nano 33 BLE boards to the PlatformIO system we'd be happy to help.

charly37, 2019-09-30:
Should it rely on the ArduinoCore running on MBEDos or directly on MBEDos ?

manuelbl, 2019-09-30:
I would prefer if priority was given to the *Arduino core based on Mbed OS framework*. This is likely the better solution for most users and in particular for users coming from the Arduino IDE.

manuelbl, 2019-09-30:
In case anyone starts to work on this: the Arduino guys managed to build the nRF52 Arduino core upon an **untagged** branch of the [Arduino Core API](https://github.com/arduino/ArduinoCore-API), namely the [arduino_namespace](https://github.com/arduino/ArduinoCore-API/tree/namespace_arduino) branch. Those sources are needed and not part of the Arduino core based on Mbed OS framework.

j0rd, 2019-10-07:
This have anything to do with anything?
https://github.com/arduino/ArduinoCore-nRF528x-mbedos

manuelbl, 2019-10-07:
Yes, that's one piece. Many more are needed...

luke-shields, 2019-11-13:
I would also love to see these boards supported

valeros, 2019-11-29:
We've added support for this board, but we didn't try it on real hardware, so it would be great if someone could try to compile and upload a basic example.

Thanks!

ivankravets, 2019-11-29:
Thanks, @valeros! Please note that you will need to use development version of platform:
```ini
[env:myenv]
platform = https://github.com/platformio/platform-nordicnrf52.git
framework = arduino
board = nano33ble
```

bertrik, 2019-11-29:
I just happened to play with a NANO33BLE board this evening and ran into this thread. Using the platformio config above, it compiles for me.

Uploading didn't work for me yet.
The first problem is that ~/.platformio/platforms/nordicnrf52/builder/main.py has an unresolved symbol basename (line 48). I fixed that by changing a line from
from os.path import isdir, join
to
from os.path import isdir, join, basename

The second problem is some kind of uploading problem:
<pre>
Building in release mode
Compiling .pio/build/default/src/BatteryMonitor.ino.cpp.o
Checking size .pio/build/default/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [===       ]  25.1% (used 65752 bytes from 262144 bytes)
PROGRAM: [===       ]  29.0% (used 285036 bytes from 983040 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, jlink, nrfjprog, nrfutil, sam-ba
CURRENT: upload_protocol = sam-ba
Looking for upload port...
Auto-detected: /dev/ttyACM0
Uploading .pio/build/default/firmware.bin
No device found on ttyACM0
*** [upload] Error 1
</pre>

vaetas, 2019-11-29:
I do have the same problem as @bertrik. Importing `basename` from `os.path` fixed initial Python Error, but now I am getting *No device found on ...* error. I have manually selected `upload_port` in ini file but the problem remains. Building seems alright.

valeros, 2019-11-29:
Thanks @bertrik @vaetas. I fixed missing basename import and also added some modifications to the upload process. Could you please pull the latest changes and try again? Simple `pio update` cpmmand should be enough if there is no conflicts with your changes.

manuelbl, 2019-11-29:
Thanks for the work on the Arduino Nano 33. I've tested it and it looks quite good already. The project compiles and compilation times are much faster than in Arduino IDE (where it's extremly slow).

There are a number of issues though:

**1. Upload**

Same problem as @bertrik and @vaetas. The path to the USB port is correct. And upload works from the Arduino IDE.

The output from `pio run -v -t upload` is:
```
<lambda>(["upload"], [".pio/build/nano33ble/firmware.bin"])
AVAILABLE: cmsis-dap, jlink, nrfjprog, nrfutil, sam-ba
CURRENT: upload_protocol = sam-ba
BeforeUpload(["upload"], [".pio/build/nano33ble/firmware.bin"])
Auto-detected: /dev/cu.usbmodem144401
bossac --info --debug --port "cu.usbmodem144401" --write --reset -U --reset .pio/build/nano33ble/firmware.bin
No device found on cu.usbmodem144401
Set binary mode
Send auto-baud
Set binary mode
*** [upload] Error 1
```

As a comparison, the Arudino IDE uses the below command:
```
/Users/me/Library/Arduino15/packages/arduino/tools/bossac/1.9.1-arduino1/bossac --port=cu.usbmodem144401 -U -i -e -w /var/folders/z2/t5tl6v7d2k95p70t98gd_wqw0000gn/T/arduino_build_635165/BatteryMonitor.ino.bin -R
```

I tried a similar command using the PlatformIO's bossac, and it worked once and then stopped to work (No device found...).

**But PlatformIO's upload task works if I double click the board's reset button first.** The LED then goes in a fade in / fade out mode.

**2. Intellisense**

In a simple Blink sketch, `#include <Arduino.h>` has squiggly red lines and the error message is
```
#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (/Users/me/Documents/PlatformIO/Projects/Nano-33-BLE/src/main.cpp).
cannot open source file "USBDescriptor.h" (dependency of "Arduino.h")
```

The missing file changed over time. It was `mbed.h` earlier.

In `c_cpp_properties.json`, the following include paths are listed:
```
"/Users/me/Documents/PlatformIO/Projects/Nano-33-BLE/include",
"/Users/me/Documents/PlatformIO/Projects/Nano-33-BLE/src",
"/Users/me/.platformio/packages/framework-arduino-nrf52-mbedos/cores/arduino",
"/Users/me/.platformio/packages/framework-arduino-nrf52-mbedos/cores/arduino/api/deprecated",
"/Users/me/.platformio/packages/framework-arduino-nrf52-mbedos/variants/ARDUINO_NANO33BLE",
"/Users/me/.platformio/packages/framework-arduino-nrf52-mbedos/libraries/PDM/src",
"/Users/me/.platformio/packages/framework-arduino-nrf52-mbedos/libraries/SPI",
"/Users/me/.platformio/packages/framework-arduino-nrf52-mbedos/libraries/USBHID",
"/Users/me/.platformio/packages/framework-arduino-nrf52-mbedos/libraries/USBMSD",
"/Users/me/.platformio/packages/framework-arduino-nrf52-mbedos/libraries/Wire",
 "/Users/me/.platformio/packages/tool-unity",
 ""
```
Possibly, directories are missing such as `cores/arduino/mbed` and `cores/arduino/mebed/drivers/internal`.

**3. Create a new project**

Creating a new project with the `nano33ble` board from PIO Home fails. The project is successfully created. But instead of opening it, an error message is displayed:
```
Could not initialize project

PIO Core Call Error: "The current working directory /Users/me/Documents/PlatformIO/Projects/Nano-33-Test2 will be used for the project.\n\nThe next files/directories have been created in /Users/me/Documents/PlatformIO/Projects/Nano-33-Test2\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Unknown board ID 'nano33ble'"
```

More or is likely to come... I will continue to test.

vaetas, 2019-11-29:
Thanks for your help @valeros. I have pulled your changed but the error persisted. 

```
<lambda>(["upload"], [".pio/build/nano33ble/firmware.bin"])
AVAILABLE: cmsis-dap, jlink, nrfjprog, nrfutil, sam-ba
CURRENT: upload_protocol = sam-ba
BeforeUpload(["upload"], [".pio/build/nano33ble/firmware.bin"])
Use manually specified: /dev/ttyACM0
Forcing reset using 1200bps open/close on port /dev/ttyACM0
bossac --info --debug --port "ttyACM0" --write --erase -U --reset .pio/build/nano33ble/firmware.bin
No device found on ttyACM0
*** [upload] Error 1
```

bertrik, 2019-11-29:
I have the upload working now. I uploaded the BatteryMonitor example from the ArduinoBLE library and I can see a "BatteryMonitor" device now in a BLE scanner on my Android phone.

vaetas, 2019-11-29:
What have you changed @bertrik. The Arduino gets into bootloader mode when I press Upload (led starts slowly blinking) but immediately fails.

vaetas, 2019-11-29:
OK, I am not sure what happened but I tried to upload it again and it worked!

valeros, 2019-11-29:
@bertrik @vaetas Thanks!

Thanks @manuelbl, good points! 
1. The uploading should be fixed in the latest version.
2. I think that's fixable, this Arduino core has quite unusual form of include flags and that requires some modifications to VSCode project template.
3. That's strange, I cannot reproduce it, maybe you have two `nordicnrf52` platforms and the stable one is used instead of the latest one from the repository?

vaetas, 2019-11-30:
Thanks a lot for making this work @valeros!

However, I have one weird problem. When I try to upload binary, Nano will switch to bootloader mode (not sure the correct name) and upload from PIO will then fail with `No device found on ttyACM0` error. But when I press Upload again, it will successfully upload. Third upload will again fail and so on.

I have tested this on two different Nano 33 BLEs and the bug appears on both of them. It seems like a problem with Arduino because it apparently closes connection when switching modes before upload. Does this happen to anybody else or is it only my problem? I am currently on Arch Linux.

bertrik, 2019-11-30:
On my system (debian bullseye/sid) I noticed that I had modemmanager installed. I don't really trust it in the sense that I'm really sure it's not messing with anything that looks like a serial port (/dev/ttyUSBx or /dev/ttyACMx), so I made sure it was not running and uninstalled it. Perhaps this is something you can try too and see if it has an effect on upload success.

manuelbl, 2019-11-30:
Thanks @valeros for the quick response. Regarding my issues:

1. Uploading: It's better now but in only works every other time (same as for @vaetas). I think I know why... see below.
2. Intellisense: I'm looking forward to a fix.
3. Opening project: Yes, I had two platforms (the productive one and the one declared in `platformio.ini` according to Ivan's instructions). So it won't be an issue anymore once it's officially released?

In the mean time I have also tested it on Windows. It behaves the same (same things working, same open issues). I've also tested J-Link upload and debugging and it works.

**Upload**

In the Arduino IDE, it works. So what's the difference? It's not the command line arguments for *bossac*. It seems that the Arduino IDE first **resets the board** before it starts *bossac*.

The clue is if the serial port is busy. Then you don't get an error message from *bossac* but from the Arduino IDE:

```
processing.app.debug.RunnerException
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:152)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2075)
	at java.lang.Thread.run(Thread.java:748)
Caused by: processing.app.SerialException: Error touching serial port '/dev/cu.usbmodem144301'.
	at processing.app.Serial.touchForCDCReset(Serial.java:107)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:136)
	... 5 more
Caused by: jssc.SerialPortException: Port name - /dev/cu.usbmodem144301; Method name - openPort(); Exception type - Port busy.
	at jssc.SerialPort.openPort(SerialPort.java:164)
	at processing.app.Serial.touchForCDCReset(Serial.java:101)
	... 6 more
```

The exception is thrown on this line: https://github.com/arduino/Arduino/blob/master/arduino-core/src/processing/app/Serial.java#L107

It's the code for the board reset. It is run if the Arduino board definition contains:
```
upload.use_1200bps_touch=true
```

The setting `upload.wait_for_upload_port=true` is also relevant. It waits for the virtual serial port to reappear after the reset was triggered.

Could that be the missing piece?

ivankravets, 2019-12-01:
@manuelbl Thank you so much for the hint!

@valeros please use reset code from Atmel SAM dev/platform. See “BeforeUpload” pre-action.

valeros, 2019-12-16:
Sorry for the late reply, could someone retest uploading with the latest platform version?

bertrik, 2019-12-16:
Sure, I'll test it this evening (if I don't forget)!

manuelbl, 2019-12-16:
I will test it but won't have time before tomorrow evening...

vaetas, 2019-12-16:
@valeros With your latest commit uploading works all the time! I have tested this with the latest development version for nordicnrf52 platform. Brilliant! Thanks.

bertrik, 2019-12-16:
For me it also work very reliable now, thanks.

fgaetani, 2019-12-17:
I updated PlatformIO and the nRF52 package, but I still don't see Arduino Nano BLE Nano among the available boards. What am I doing wrong?

luke-shields, 2019-12-17:
Hi there, could anyone explain the benefits of using PlatformIO rather than the Microsoft Arduino extension? Thanks and great job getting this working!

valeros, 2019-12-17:
@fgaetani  This feature is not released yet, so you need to use the upstream version of the platform:
```ini
[env:myenv]
platform = https://github.com/platformio/platform-nordicnrf52.git
framework = arduino
board = nano33ble
```
@luke-shields So in a nutshell, PlatformIO is more advanced and customizable ecosystem in comparison with a simple IDE extension like Microsoft Arduino. More detailed info you can find [here](https://docs.platformio.org/en/latest/what-is-platformio.html#what-is-platformio).

fgaetani, 2019-12-17:
@valeros thank you very much. I solved, however I am having problems using the libraries integrated in mbed: "FlashIAPBlockDevice.h" and "FATFileSystem.h". The project is compiled but the board is unable to access the file system and fails when I read the files.

valeros, 2019-12-17:
@fgaetani Does it work in Arduino IDE?

fgaetani, 2019-12-17:
@valeros Sorry I solved it, it was a problem of importing libraries. However FATFileSystem seems to be very heavy, are there better solutions in PIO to save data on the flash memory of nRF52840?

bertrik, 2019-12-17:
> @valeros Sorry I solved it, it was a problem of importing libraries. However FATFileSystem seems to be very heavy, are there better solutions in PIO to save data on the flash memory of nRF52840?

It might be better to open a separate issue for that, this thread is about the platformio build system, not about specific libraries.

fgaetani, 2019-12-17:
@bertrik sorry you're right, now I open a separate issue. However I can compile and run everything with Arduino Nano 33 BLE, but I have problems with error reporting in visual studio code, all include items are reported as errors:
"#Include errors detected. Update includePath. Revision marks are disabled for this conversion unit" and "impossible to open error code" Arduino.h "of the source file"
There is a solution?

manuelbl, 2019-12-17:
I can confirm the findings so far:

- Upload now works without any problems
- Intellisense is unable to locate many header files and shows red squiggly lines

The first file Intellisense complained about is `mbed.h`. After adding the path `/Users/me/.platformio/packages/framework-arduino-nrf52-mbedos/cores/arduino/mbed` to `.vscode/c_cpp_properties.json`, it complained about `mbed_rtx_storage.h`. So most likely all 368 directories from `variants/ARDUINO_NANO33BLE/includes.txt` might need to be added...

valeros, 2019-12-17:
Thanks, should be fixed in the latest `platformio-core`. Please upgrade your `platformio` in your terminal window via the following command; `pio upgrade --dev`

manuelbl, 2019-12-17:
It's an improvement. The squiggly lines on `#include <Arduino.h>` are gone.

However, if I include mbed files directly in `main.cpp`, Intellisense is still not working properly.

I've added:

```
#include "mbed.h"
#include "rtos.h"
```

and:

```
void setup() {
  while (!Serial)
    ThisThread::sleep_for(1);
  ...
}
```

I then get squiggly lines on both include directives (*cannot open source file mbed.h* and similar for *rtos.h*) and code completion for `ThisThread` does not work. Yet it compiles without an issue.

valeros, 2019-12-17:
Hm, strange, but I cannot reproduce your issue:
[image]
Could you please try to rebuild IntelliSense index?

manuelbl, 2019-12-17:
I'm afraid it's worse now. But mainly it's unreliable. 

After rebuilding the index, `#include <Arduino.h>` has squiggly lines again (*cannot open source file "mbed.h"*).

For a short period, it couldn't find `ArduinoBLE.h` either. That now works again.

And no amount of rebuilding, restarting, changing `platformio.ini`, removing `.pio`, building etc. can revert to at least the state where `Arduino.h` was ok.

When I open `main.cpp` immediately after a restart, hovering over `ThisThread` shows decent documentation of the type. About 20 seconds later, the red squiggly lines appears the documentation hover no longer works. Cmd-clicking on it still leads to the correct header file...

fgaetani, 2019-12-18:
@valeros I deleted the .pio and .vscode folders and now I have the error only on arduino.h, the errors are as follows:
"#Include errors were found. Update includePath. Revision marks are disabled for this conversion unit (/Users/user/project/src/main.cpp)." and "could not open the error code" mbed.h "of the source file (dependency of /Users/user/project/src/main.cpp )"

PiotrDabkowski, 2020-01-28:
@valeros, @valeros I am having the same problem -  "could not open the error code" mbed.h " and the code completion feature does not work. The project builds successfully though.  Only nano33ble boards seems to be affected.

It would be really great to have that fixed! nano33ble is a very popular board.

ivankravets, 2020-01-30:
@bobbrow, @sean-mcmanus, @Colengms, @michelleangela could you help us with this issue? How could we debug it?

You can install PlatformIO IDE extension for VSCode and try to open this project https://github.com/platformio/platform-nordicnrf52/tree/develop/examples/arduino-serial-plotter. Please wait a few minutes when PlatformIO installs all dependencies and rebuilds C/C++ configuration.

P.S: It works only on Windows and does not work on other OS.

Thanks in advance!

ivankravets, 2020-01-30:
@PiotrDabkowski how did you fix CLion? What was wrong with the default configuration?

sean-mcmanus, 2020-01-30:
@ivankravets Sure, I got the PlatformIO extension installed and the dependecy installed and the platform-nordicnrf52-develop folder opened -- what do I do next to repro the issue? I don't see any c_cpp_properties.json.

ivankravets, 2020-01-30:
@valeros please help and explain how to reproduce it.

valeros, 2020-01-30:
Hi @sean-mcmanus ! Please keep in mind that this issue only occurs on Linux and Mac.
Steps to reproduce:
1. Install PlatformIO IDE extension
2. Install `nordicnrf52` platform (using menu`PIO Home > Platforms > Embedded` or using CLI command `pio platform install nordicnrf52`)
3. On `PIO Home` page press `Project examples` and select `arduino-serial-plotter` example
4. Wait until PlatformIO generates project files for VSCode

After that steps you should see error messages about missing include files.
Not sure if it might be the reason (it works just fine on Windows), but the include paths are set using @file command. Thanks!

PiotrDabkowski, 2020-01-30:
@ivankravets generated CMakeListsPrivate.txt had some bad add_definitions statements that resulted in `macro names must be identifiers` error, did not debug it in detail, removing these lines fixes the error.

This seems unrelated to VS issue. Possibly just adding few more include folders will fix the problem. After all this is just IDE code inspection, compilation and upload to the board works just fine with `pio build` out of the box.

sean-mcmanus, 2020-01-31:
I see `.../.platformio/packages/framework-arduino-nrf52-mbedos/cores/arduino" in the includePath but the path with mbed appears to be missing: `.../.platformio/packages/framework-arduino-nrf52-mbedos/cores/Arduino/mbed`. Can that folder be added to fix the missing include?

valeros, 2020-01-31:
@sean-mcmanus ! Thanks for reporting, you also need the latest development version of `platformio-core`. Just run `pio upgrade --dev` in the terminal window. After that you should see something like this in `compilerPath` variable:
```
"compilerPath": "\"/home/user/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-gcc\" -mcpu=cortex-m4 -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -mthumb -iprefix/home/user/.platformio/packages/framework-arduino-nrf52-mbedos/cores/arduino @\"/home/user/.platformio/packages/framework-arduino-nrf52-mbedos/variants/ARDUINO_NANO33BLE/includes.txt\""
```
In the file `includes.txt` you can see `-iwithprefixbefore/mbed` flag and with the flag
 `-iprefix/home/user/.platformio/packages/framework-arduino-nrf52-mbedos/cores/arduino`
it should result in a proper path
`/home/user/.platformio/packages/framework-arduino-nrf52-mbedos/cores/arduino/mbed`

sean-mcmanus, 2020-01-31:
The `\"` around the path after the `@` is causing the breakage to our compiler querying -- removing that fixes it.

However, there's another bug in that a bunch of the defines are populated with "-D" in front -- those should be removed.

After that, all the squiggles should be fixed. This seems like a bug with PlatformIO? If you think our C/C++ extension is not handling compiler arguments correctly you could file a bug on our extension: https://github.com/Microsoft/vscode-cpptools/issues.

FYI, we also added a new compilerArgs property as of 0.25.0 if that is more convenient to use than appending args to the end of the compilerPath.

ivankravets, 2020-01-31:
@sean-mcmanus is it possible to pass different configuration for C and C++ compilers? We have different flags for C and C++.

sean-mcmanus, 2020-01-31:
@ivankravets Not currently via c_cpp_properties.json, because there is no separate cppCompilerPath and cCompilerPath setting (you could file a feature request for that...no one else has yet requested that). However, via the custom configuration provider interface (https://github.com/Microsoft/vscode-cpptools-api/blob/master/api.ts) a custom configuration provider could provide separate compilerPath/args for .c files versus .cpp files.

sean-mcmanus, 2020-01-31:
@ivankravets Another possibility is to have 2 different configurations, one for .c files and another for .cpp files, and then run the new cpptools.setActiveConfigName command (see https://github.com/microsoft/vscode-cpptools/pull/4893 ) when the active file type changes, but that command hasn't shipped yet.

valeros, 2020-01-31:
> However, there's another bug in that a bunch of the defines are populated with "-D" in front -- those should be removed.

Thanks, this is already fixed, but hasn't been released yet.

> The \" around the path after the @ is causing the breakage to our compiler querying -- removing that fixes it.

Indeed, it works after removing extra quotes, but how we can handle whitespace characters in path to files specified using `@file` syntax?

sean-mcmanus, 2020-01-31:
Our compilerPath property doesn't actually work with `@\"path with spaces\"` so the usage of `\"` doesn't fix the path with spaces scenario (could be a bug in our handling of the compilerPath).

However, using the compilerArgs enables the `@\"path with spaces\"` to parse correctly, e.g.
```
            "compilerPath": "~/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-gcc",
            "compilerArgs": [
                "-iprefix/~/.platformio/packages/framework-arduino-nrf52-mbedos/cores/arduino",
                "@~/.platformio/packages/framework-arduino-nrf52-mbedos/variants/ARDUINO NANO33BLE/includes.txt"
            ]
```

ivankravets, 2020-02-03:
We opened an issue in PIO Core https://github.com/platformio/platformio-core/issues/3364

@valeros will back soon with updates. Thanks!

valeros, 2020-02-04:
@sean-mcmanus Could you please also give us a hint about `-include` flag in `compilerArgs`. So, in a nutshell,  VSCode is not able to see any definition that is specified in a special header file which is passed to the compiler via `-include file` command although the path to this file is specified in `includePath` section. It only works if path to this file is specified in `compilerArgs` using `-I` flag. Is that how it should work by design?

I prepared [a basic example](https://github.com/valeros/vscode-include) that is compilable using CMake, but when you open it in VSCode you'll see an issue with undefined symbol.
Thanks in advance!

sean-mcmanus, 2020-02-04:
`-include` is equivalent to our "forcedInclude" setting, so you should use that instead. Our compilerArgs setting is mostly just used for compiler querying for the system includes and defines, and code for translating the args into settings like forcedInclude that we understand doesn't exist (it exists for compile_commands.json processing though).

valeros, 2020-02-04:
@sean-mcmanus Thanks for the hint, but looks like `forcedInclude` option also ignores paths specified in `includePath`. It works only if I specify the full path to a header file.
Am I missing something? Thanks!

[20 more replies]