platformio/platform-nordicnrf52 #56: build fails with merge firmware error

H0bbes, 2019-08-29:
Hello I have an error when i try to build the example code for BLE found @https://docs.platformio.org/en/latest/tutorials/nordicnrf52/arduino_debugging_unit_testing.html

It fails when  i add the softdevice line:

`build_flags = -DNRF52_S132`

[env:nrf52_dk]
platform = nordicnrf52@3.6.0 
board = nrf52_dk 
framework = arduino 
debug_tool = jlink 
build_flags = -DNRF52_S132 
lib_deps = 
  BLEPeripheral
 
```
Building .pio/build/nrf52_dk/userfirmware.hex
Checking size .pio/build/nrf52_dk/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   1.6% (used 1060 bytes from 65536 bytes)
PROGRAM: [          ]   3.0% (used 15488 bytes from 524288 bytes)
Building .pio/build/nrf52_dk/firmware.hex
srec_cat: .pio/build/nrf52_dk/userfirmware.hex: 2: multiple 0x0001C000 values
    (previous = 0x01, this one = 0x00)
*** [.pio/build/nrf52_dk/firmware.hex] Error 1
```

When i try older version (nordicnrf52@3.2.0) it compiles but the resulted firmware does not work.
Any idea?
Thanks

H0bbes, 2019-08-29:
I will add that i had to put the s132_nrf52_3.1.0_softdevice.hex in /home/user/.platformio/packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/softdevice/s132/hex/s132_nrf52_3.1.0_softdevice.hex
As I had a warning saying :
> HARDWARE: NRF52832 64MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) On-board (cmsis-dap, jlink) External (blackmagic, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinonordicnrf5 1.600.0 (6.0), tool-sreccat 1.164.0 (1.64)
**Warning! Cannot find an appropriate softdevice binary!**

valeros, 2019-08-30:
Hi @H0bbes ! Thanks a lot for the info! Indeed the softdevice binaries were missing, so I've updated the framework package. But I couldn't reproduce other issues your mentioned: firmware was merged correctly and I can successfully connect to the board. Please run `pio update` and try to upload your example again. Does it work?

H0bbes, 2019-08-30:
It now works!
I can see that the softdevice version you add is 2.0.1. Previously I tried to manually add the 3.1.0 and it seems it does not work with it.
Anyway thank you

closed 2019-08-30 by H0bbes