platformio/platform-nordicnrf52 #59: nordicnrf52 Mbed framework build error on windows

syk543277780, 2019-10-04:
PC: Windows 10 64 bit
board: nrf52840_dk
IDE: VS Code

Here is my platformio.ini
```
[env:nrf52840_dk]
platform = nordicnrf52
framework = mbed
board = nrf52840_dk
build_flags = -DPIO_FRAMEWORK_MBED_RTOS_PRESENT 
```

I get build Error on windows 64bit. also I try this on several computers and it works well on Linux and Mac OS. I tried on 3 Windows 10 PC. None of them build success.

here is the building log

```
Executing task: C:\Users\yongkang.song.platformio\penv\Scripts\platformio.exe run <
Processing nrf52840_dk (platform: nordicnrf52; framework: mbed; board: nrf52840_dk)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via -v, --verbose option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/nrf52840_dk.html
PLATFORM: Nordic nRF52 3.7.0 > Nordic nRF52840-DK
HARDWARE: NRF52840 64MHz, 256KB RAM, 1MB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap, jlink) External (blackmagic, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-mbed 5.51304.190826 (5.13.4), tool-sreccat 1.164.0 (1.64)
Collecting mbed sources…
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies…
No dependencies
Compiling .pio\build\nrf52840_dk\FrameworkMbed\cmsis\TARGET_CORTEX_M\mbed_tz_context.o
arm-none-eabi-gcc: error: CreateProcess: No such file or directory
Compiling .pio\build\nrf52840_dk\FrameworkMbed\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAtmel.o
arm-none-eabi-g++: error: CreateProcess: No such file or directory
Compiling .pio\build\nrf52840_dk\FrameworkMbed\components\802.15.4_RF\atmel-rf-driver\source\at24mac.o
arm-none-eabi-g++: error: CreateProcess: No such file or directory
*** [.pio\build\nrf52840_dk\FrameworkMbed\cmsis\TARGET_CORTEX_M\mbed_tz_context.o] Error 1
*** [.pio\build\nrf52840_dk\FrameworkMbed\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAtmel.o] Error 1
*** [.pio\build\nrf52840_dk\FrameworkMbed\components\802.15.4_RF\atmel-rf-driver\source\at24mac.o] Error 1
Compiling .pio\build\nrf52840_dk\FrameworkMbed\components\802.15.4_RF\mcr20a-rf-driver\source\MCR20Drv.o
arm-none-eabi-gcc: error: CreateProcess: No such file or directory
*** [.pio\build\nrf52840_dk\FrameworkMbed\components\802.15.4_RF\mcr20a-rf-driver\source\MCR20Drv.o] Error 1
```

I do have gccarmnoneeabi on my computer
```
C:\Users\yongkang.song>C:\Users\yongkang.song.platformio\packages\toolchain-gccarmnoneeabi\bin\arm-none-eabi-gcc.exe --version
arm-none-eabi-gcc.exe (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
Copyright © 2017 Free Software Foundation, Inc.
```

valeros, 2019-10-04:
Hi @syk543277780 ! Could you please try the following `platformio.ini`
```
[platformio]
packages_dir = C:\\pio

[env:nrf52840_dk]
platform = nordicnrf52
framework = mbed
board = nrf52840_dk
build_flags = -DPIO_FRAMEWORK_MBED_RTOS_PRESENT 
```

Sundypha, 2019-10-04:
Hi

I have the exact same problem.

I tried the following:
```ini
[platformio]
packages_dir = D:\\CODE\\pio-packages

[env:nrf52840_dk]
platform = nordicnrf52@3.6.0
board = nrf52840_dk
framework = mbed
build_flags = -DPIO_FRAMEWORK_MBED_RTOS_PRESENT
```

Neither `platform = nordicnrf52@3.6.0` nor `platform = nordicnrf52` worked. Both result in the same output as syk543277780  posted.

syk543277780, 2019-10-04:
@valeros  Thanks. 
```
[platformio]
packages_dir = C:\\pio

[env:nrf52840_dk]
platform = nordicnrf52
framework = mbed
board = nrf52840_dk
build_flags = -DPIO_FRAMEWORK_MBED_RTOS_PRESENT 
```

This config works for me

closed 2019-10-04 by syk543277780