platformio/platform-nordicnrf52 #61: FlashIAPBlockDevice.h: No such file or directory

BrettMcGill, 2019-10-15:
I am using VS-Code with MBed for Nordic nRF52840
Running under Windows 10
Get the above error when including FlashIAPBlockDevice.h

platformio.ini contains the following (this was necessary under Windows):
[platformio]
packages_dir = c:\0_docs\PIO

Pretty much trying to use this file per the mbed example.
https://os.mbed.com/docs/mbed-os/v5.14/apis/flashiapblockdevice.html

* Following the lightbulb takes me to an auto-generated version of c_cpp_properties.json. Editing this doesn't help, nor do the changes persist. Not surprising.
* Manually editing this file has the same result.

Any suggestions???

valeros, 2019-10-15:
Hi @BrettMcGill ! mbed components depend on RTOS feature. Try to add the line
`build_flags = -DPIO_FRAMEWORK_MBED_RTOS_PRESENT` to your `platformio.ini`, e.g.:
```
[env:nrf52840_dk]
platform = nordicnrf52
framework = mbed
board = nrf52840_dk
build_flags = -DPIO_FRAMEWORK_MBED_RTOS_PRESENT
```

BrettMcGill, 2019-10-15:
Thanks @valeros 
I already have that exact configuration in my platformio.ini file.

Some possibly relevant details:
- The file that includes "FlashIAPBlockDevice.h" appears to compile successfully before the compile fails:
...
Compiling .pio\build\nrf52840_dk\src\THE_FILE.o
Compiling .pio\build\nrf52840_dk\src\another_file.o
Compiling .pio\build\nrf52840_dk\src\yet_another_file.o
Compiling .pio\build\nrf52840_dk\src\main.o
Generating LD script .pio\build\nrf52840_dk\NRF52840.ld.link_script.ld
src\core_log.cpp:9:10: fatal error: FlashIAPBlockDevice.h: No such file or directory
...
 #include "FlashIAPBlockDevice.h"
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

- If I Ctrl-click the included file:
#include "FlashIAPBlockDevice.h"
it opens "FlashIAPBlockDevice.h"
So it appears to be findable.

- The file "c_cpp_properties.json" contains the following lines:
            "includePath": [
                "c:/0_docs/PIO/framework-mbed/components/storage/blockdevice/COMPONENT_QSPIF",
...
            "browse": {
                    "c:/0_docs/PIO/framework-mbed/components/storage/blockdevice/COMPONENT_QSPIF",
            "defines": [
                "COMPONENT_QSPIF=1",

What I need is for "c_cpp_properties.json" to say "COMPONENT_FLASHIAP" instead of "COMPONENT_QSPIF".
However this file is auto-generated and can not be edited.

I think what I really need to know is what/where do I edit so that "c_cpp_properties.json" will point to the correct library code.

Thanks again,
Brett

BrettMcGill, 2019-10-16:
@valeros 
Woohoo! Got it!
In:
    \PIO\framework\mbed\targets\targets.json
under:
    "MCU_NRF52840": {
I edited:
        "components_add": ["QSPIF"],
to:
        "components_add": ["FLASHIAP"],

I will now try to find a way to override this locally in my project, so that this library remains untouched.

valeros, 2019-10-16:
@BrettMcGill Usually it's done in [mbed_app.json file](http://docs.platformio.org/en/latest/frameworks/mbed.html#configuration-system) in the root folder of your project:
```
{
    "target_overrides": {
        "*": {
            "target.components_add": ["SPIF", "FLASHIAP"]
        }
    }
}
```

BrettMcGill, 2019-10-17:
Many thanks @valeros 
I switched to that and it works perfectly! :-)

closed 2019-10-17 by BrettMcGill