platformio/platform-nordicnrf52 #63: nRF52840-DK Pinsout does not match the schematics.

chappy1978, 2019-11-19:
The nRF52840-DK framework does work correctly with the dev board. The variant.cpp does not match up with the pins on the board. Also the nrf52840_dk.json makes a call to the nRF52DK variant which is a development kit for the nRF52810 and nRF52832.
  
"build": {
    "core": "nRF5",
    "cpu": "cortex-m4",
    "extra_flags": "-DARDUINO_NRF52_DK",
    "f_cpu": "64000000L",
    "ldscript": "nrf52_xxaa.ld",
    "mcu": "nrf52840",
    "variant": "nRF52DK"

tyeth, 2020-12-13:
I've found it to only match up to the NRF52840-DK (adafruit) board in platformio (which is the problem as I don't have an adafruit product). It also seems to tally with variant.h under pca10056 although my intellisense only got that when I manually opened variant.h rather that in debugging when I needed it.

aiunderstand, 2023-09-23:
See issue #171