platformio/platform-nordicnrf52 #65: adafruit-nrfutil missing in Ubuntu 18.04

bromagosa, 2019-11-29:
Hi there.

PlatformIO is complaining that it's not finding adafruit-nrfutil when I try to compile for the Circuit Playground nRF52.

Sure enough, I looked into the `~/.platformio/packages/framework-arduinoadafruitnrf52/tools/adafruit-nrfutil/` directory and the Linux package is missing:

```
%ls ~/.platformio/packages/framework-arduinoadafruitnrf52/tools/adafruit-nrfutil/
macos  win32
```

Here's what the platformio.ini section looks like:

```
[env:cplay52]
platform = nordicnrf52
board = adafruit_cplaynrf52840
```

Any idea how to fix this? pip3-installing it didn't fix the issue either. I guess PlatformIO is trying to find the tool in its own path, not the global one.

bromagosa, 2019-11-29:
Update: after installing the `adafruit-nrftool` package globally by running `sudo -H pip3 install adafruit-nrfutil`, I get the following traceback:

```
%pio run -e cplay52 -t upload
Processing cplay52 (platform: nordicnrf52; board: adafruit_cplaynrf52840; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/adafruit_cplaynrf52840.html
PLATFORM: Nordic nRF52 3.8.0 > Circuit Playground Bluefruit
HARDWARE: NRF52840 64MHz, 243KB RAM, 796KB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), tool-openocd 2.1000.190707 (10.0), tool-sreccat 1.164.0 (1.64), framework-arduinoadafruitnrf52 1.1405.191023 (14.5)
Converting vm.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <Wire> 1.0
Building in release mode
Compiling .pio/build/cplay52/src/vm.ino.cpp.o
Building .pio/build/cplay52/firmware.zip
Checking size .pio/build/cplay52/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [=         ]  11.9% (used 29584 bytes from 248832 bytes)
PROGRAM: [=         ]  11.5% (used 93436 bytes from 815104 bytes)
Failed to import the site module
Error processing line 1 of /usr/local/lib/python3.6/dist-packages/matplotlib-3.1.1-py3.6-nspkg.pth:

Traceback (most recent call last):
  File "/usr/lib/python3.6/site.py", line 174, in addpackage
    exec(line)
  File "<string>", line 1, in <module>
  File "/usr/lib/python3.6/types.py", line 171, in <module>
    import functools as _functools
  File "/usr/lib/python3.6/functools.py", line 21, in <module>
    from collections import namedtuple
  File "/usr/lib/python3.6/collections/__init__.py", line 32, in <module>
    from reprlib import recursive_repr as _recursive_repr
  File "/home/renard/.local/lib/python2.7/site-packages/reprlib/__init__.py", line 7, in <module>
    raise ImportError('This package should not be accessible on Python 3. '
ImportError: This package should not be accessible on Python 3. Either you are trying to run from the python-future src folder or your installation of python-future is corrupted.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.6/site.py", line 570, in <module>
    main()
  File "/usr/lib/python3.6/site.py", line 557, in main
    known_paths = addsitepackages(known_paths)
  File "/usr/lib/python3.6/site.py", line 345, in addsitepackages
    addsitedir(sitedir, known_paths)
  File "/usr/lib/python3.6/site.py", line 213, in addsitedir
    addpackage(sitedir, name, known_paths)
  File "/usr/lib/python3.6/site.py", line 184, in addpackage
    import traceback
  File "/usr/lib/python3.6/traceback.py", line 3, in <module>
    import collections
  File "/usr/lib/python3.6/collections/__init__.py", line 32, in <module>
    from reprlib import recursive_repr as _recursive_repr
  File "/home/renard/.local/lib/python2.7/site-packages/reprlib/__init__.py", line 7, in <module>
    raise ImportError('This package should not be accessible on Python 3. '
ImportError: This package should not be accessible on Python 3. Either you are trying to run from the python-future src folder or your installation of python-future is corrupted.
*** [.pio/build/cplay52/firmware.zip] Error 1
```

vakker, 2019-11-29:
I think you shouldn't need to install the package as `sudo`. For me a simple `pip install adafruit-nrfutil` worked. Have you tried that?

ivankravets, 2019-11-29:
Hi Bernat! ;)

How did you install PlatformIO? Do you use VSCode?

bromagosa, 2019-11-29:
Hi Ivan!

I don't use VSCode, I just use the command line tool :)

To get PlatformIO I just did `pip install pio`.

bromagosa, 2019-11-29:
> pip install adafruit-nrfutil

Yep, that's what I did. That doesn't work in neither of my two computers.

ivankravets, 2019-11-29:
Could you try now?
```
pip uninstall platformio
```

```
pip3 install platformio
```

Just to instruct PlatformIO to use Python 3

bromagosa, 2019-11-30:
I've uninstalled platformio, and now when running `pip3 install platformio` (both with sudo and without) I find that links to `/usr/bin/pio` and `/usr/bin/platformio` haven't been created. Is that normal?

What's more, there seems to be no `pio` or `platformio` executable anywhere in the system :thinking:

bromagosa, 2019-11-30:
Sorry, scratch that. The executable is in `/usr/local/bin`, which for some reason wasn't in the PATH of my current terminal session. Forget it.

After installing pio with pip3 I still have the same problem, though:

```
Building .pio/build/cplay52/firmware.zip
sh: 1: adafruit-nrfutil: not found
*** [.pio/build/cplay52/firmware.zip] Error 127
```

bromagosa, 2019-11-30:
Okay, after running:

```
sudo -H pip3 install adafruit-nrfutil
```

I can now compile for the cplay52. Should we close this? Or should `pio` take care of all this automatically when you try to compile for this framework?

Thanks, Ivan! :)

ivankravets, 2019-11-30:
I marked this issue as a bug. The ideal solution for this issue is providing binary for `adafruit-nrfutil` package. This package is written in Python 3 and does not support Python 2. PlatformIO supports both Python versions.

@jeremypoulter, where did you get binaries for Windows and macOS?

smandalbme, 2020-05-09:
I am getting this error whike trying to run adafruit-nrfutil. How do i add it to PATH?
smandal@sm-thinkpad:~$ adafruit-nrfutil version
Traceback (most recent call last):
  File "/usr/local/bin/adafruit-nrfutil", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 3258, in <module>
    @_call_aside
  File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 3242, in _call_aside
    f(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 3271, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 584, in _build_master
    ws.require(__requires__)
  File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 901, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 787, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'ecdsa>=0.13' distribution was not found and is required by adafruit-nrfutil

cujomalainey, 2020-05-10:
@smandalbme I recommend you contact adafruit as it is their tool that is breaking, but it seems like you are missing a dependency.

`pip3 install ecdsa`

smandalbme, 2020-05-10:
I can see the package when i do a sudo. However without permission, the package is not executed, thus it doesn't work with the Arduino IDE as well. Probably I am setting the path wrong somehow. How do i set the PATH correctly so that the package is accessible?

smandal@sm-thinkpad:~$ sudo adafruit-nrfutil version
adafruit-nrfutil version 0.5.3.post12

Adafruit in its guide just mentions: Add pip3 installation dir to your PATH if it is not added already. Make sure adafruit-nrfutil can be executed in terminal by running.

smandalbme, 2020-05-10:
[Solved]
Seems like the problem is not with the adafruit-nrfutil  package, but the snap distribution of Arduino which is not allowing write permission in sudo mode. Using the native distribution from Arduino portal solves the problem. Sorry for the trouble folks and thanks for the information.