platformio/platform-nordicnrf52 #66: For uploading `--touch` seems to be missing

vakker, 2019-11-29:
I'm testing a setup as:
```
[env:adafruit_feather_nrf52840]
platform = nordicnrf52
board = adafruit_feather_nrf52840
framework = arduino
```

and I'm getting:
```
Failed to upgrade target. Error is: No data received on serial port. Not able to proceed.
Traceback (most recent call last):
  File "/home/user/workspace/tools/Adafruit_nRF52_nrfutil/nordicsemi/__main__.py", line 307, in serial
    dfu.dfu_send_images()
  File "/home/user/workspace/tools/Adafruit_nRF52_nrfutil/nordicsemi/dfu/dfu.py", line 235, in dfu_send_images
    self._dfu_send_image(HexType.APPLICATION, self.manifest.application)
  File "/home/user/workspace/tools/Adafruit_nRF52_nrfutil/nordicsemi/dfu/dfu.py", line 200, in _dfu_send_image
    application_size)
  File "/home/user/workspace/tools/Adafruit_nRF52_nrfutil/nordicsemi/dfu/dfu_transport_serial.py", line 216, in send_start_dfu
    self.send_packet(packet)
  File "/home/user/workspace/tools/Adafruit_nRF52_nrfutil/nordicsemi/dfu/dfu_transport_serial.py", line 304, in send_packet
    ack = self.get_ack_nr()
  File "/home/user/workspace/tools/Adafruit_nRF52_nrfutil/nordicsemi/dfu/dfu_transport_serial.py", line 353, in get_ack_nr
    "No data received on serial port. Not able to proceed.")
nordicsemi.exceptions.NordicSemiException: No data received on serial port. Not able to proceed.
```

PIO for the upload step runs:
```
/home/user/miniconda3/bin/adafruit-nrfutil dfu serial -p /dev/ttyACM0 -b 115200 --singlebank -pkg .pio/build/adafruit_feather_nrf52840/firmware.zip
```
while examples from Adafruit suggests using `--touch 1200`, and indeed doing that does not fail the upload.'/home/user/miniconda3/bin/adafruit-nrfutil', 'dfu', 'serial', '-p', '/dev/ttyACM0', '-b', '115200', '--singlebank', '-pkg', '.pio/build/adafruit_feather_nrf52840/firmware.zip'

It seems like without `--touch 1200` the board doesn't go into DFU mode so the serial write doesn't achieve the same purpose.

valeros, 2019-12-16:
Hi @vakker ! Sorry for the late reply. Where did you find this option `--touch 1200`? Isn't it only used for bootloader uploading? Anyway, I've added default `use_1200bps_touch` option to `nrfutil` upload protocol. Would be great if you could test it with your board:
```
[env:myenv]
platform = https://github.com/platformio/platform-nordicnrf52.git
framework = arduino
board = adafruit_feather_nrf52840
```
Thanks!

vakker, 2019-12-16:
Yes, the bootloader uses that. For some reason without that for me the upload just doesn't work. With touch it works without any issues.
I'm away from home for some time, so I don't have the hardware to test it. My patch worked for me, but probably your solution is the correct (rather than hardcoding the option), so I guess it should work.

ivankravets, 2019-12-17:
Resolved in https://github.com/platformio/platform-nordicnrf52/commit/355ea10957c26da9586c432b20f2573751ed56a4

closed 2019-12-17 by ivankravets

ivankravets, 2019-12-17:
@vakker please re-test with 
```
[env:myenv]
platform = https://github.com/platformio/platform-nordicnrf52.git
framework = arduino
board = adafruit_feather_nrf52840
```

vakker, 2019-12-17:
Yes, I tried it and it works!
Thanks