platformio/platform-nordicnrf52 #71: nRF52832 - #include <set> breaks compilation on PlatformIO but compiles fine on ArduinoIDE

beegee-tokyo, 2020-02-17:
Weird error, maybe someone can help me. Just including **`#include <set>`** into an empty sketch or main.cpp breaks compilation on PlatformIO, but compiles fine under ArduinoIDE.
Error message on PlatformIO:    
### c:\users\beegee\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:243:56: error: macro "min" passed 3 arguments, but takes just 2

### Settings:
**PlatformIO**
Home V3.1.0
Core 4.2.1a3
Nordic nRF52 platform V4.0.0
**platformio.ini**
```
[env:adafruit_feather_nrf52832]
platform = nordicnrf52
board = adafruit_feather_nrf52832
framework = arduino
```
**ArduinoIDE**
Version 1.8.10
Board Adafruit nRF52 V 0.18.0

**Example sketch:**
```cpp
#include <Arduino.h>
#include <set>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

**Compiler output on ArduinoIDE** (I avoided verbose output, because it is very long and doesn't include any info)
```
Sketch uses 22508 bytes (7%) of program storage space. Maximum is 290816 bytes.
Global variables use 3252 bytes (6%) of dynamic memory, leaving 48972 bytes for local variables. Maximum is 52224 bytes.
```
**Compiler output on PlatformIO**
```
> Executing task: C:\Users\beegee\.platformio\penv\Scripts\platformio.exe run <

Processing adafruit_feather_nrf52832 (platform: nordicnrf52; board: adafruit_feather_nrf52832; framework: ardu
ino)
-------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/adafruit_feather_nrf52832.html
PLATFORM: Nordic nRF52 4.0.0 > Adafruit Bluefruit nRF52832 Feather
HARDWARE: NRF52832 64MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (jlink, stlink)
PACKAGES:
 - framework-arduinoadafruitnrf52 1.1405.191023 (14.5)
 - tool-sreccat 1.164.0 (1.64)
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\adafruit_feather_nrf52832\src\main.cpp.o
Compiling .pio\build\adafruit_feather_nrf52832\FrameworkArduino\Adafruit_TinyUSB_Core\tinyusb\src\class\midi\midi_device.c.o
Compiling .pio\build\adafruit_feather_nrf52832\FrameworkArduino\Adafruit_TinyUSB_Core\tinyusb\src\class\msc\msc_device.c.o
Compiling .pio\build\adafruit_feather_nrf52832\FrameworkArduino\Adafruit_TinyUSB_Core\tinyusb\src\class\vendor\vendor_device.c.o
In file included from c:\users\beegee\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_tree.h:63:0,
                 from c:\users\beegee\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\set:60,
                 from src\main.cpp:2:
c:\users\beegee\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:243:56: error: macro "min" passed 3 arguments, but takes just 2
     min(const _Tp& __a, const _Tp& __b, _Compare __comp)
                                                        ^
c:\users\beegee\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:265:56: error: macro "max" passed 3 arguments, but takes just 2
     max(const _Tp& __a, const _Tp& __b, _Compare __comp)
                                                        ^
Compiling .pio\build\adafruit_feather_nrf52832\FrameworkArduino\Adafruit_TinyUSB_Core\tinyusb\src\common\tusb_fifo.c.o
In file included from src\main.cpp:1:0:
Compiling .pio\build\adafruit_feather_nrf52832\FrameworkArduino\Adafruit_TinyUSB_Core\tinyusb\src\device\usbd.c.o
c:\users\beegee\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:195:5: error: expected unqualified-id before 'const'
     min(const _Tp& __a, const _Tp& __b)
     ^
c:\users\beegee\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:195:5: error: expected ')' before 'const'
c:\users\beegee\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:195:5: error: expected ')' before 'const'
c:\users\beegee\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:195:5: error: expected initializer before 'const'
c:\users\beegee\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:219:5: error: expected unqualified-id before 'const'
     max(const _Tp& __a, const _Tp& __b)
     ^
c:\users\beegee\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:219:5: error: expected ')' before 'const'
c:\users\beegee\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:219:5: error: expected ')' before 'const'
c:\users\beegee\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:219:5: error: expected initializer before 'const'
In file included from c:\users\beegee\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_tree.h:63:0,
                 from c:\users\beegee\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\set:60,
                 from src\main.cpp:2:
c:\users\beegee\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:246:7: error: expected primary-expression before 'if'
       if (__comp(__b, __a))
       ^~
Compiling .pio\build\adafruit_feather_nrf52832\FrameworkArduino\Adafruit_TinyUSB_Core\tinyusb\src\device\usbd_control.c.o
c:\users\beegee\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:246:7: error: expected '}' before 'if'
c:\users\beegee\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:246:7: error: expected ';' before 'if'
c:\users\beegee\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:248:7: error: expected unqualified-id before 'return'
       return __a;
       ^~~~~~
c:\users\beegee\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:268:7: error: expected primary-expression before 'if'
       if (__comp(__a, __b))
       ^~
c:\users\beegee\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:268:7: error: expected '}' before 'if'
c:\users\beegee\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:268:7: error: expected ';' before 'if'
c:\users\beegee\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:270:7: error: expected unqualified-id before 'return'
       return __a;
       ^~~~~~
c:\users\beegee\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:271:5: error: expected declaration before '}' token
     }
     ^
*** [.pio\build\adafruit_feather_nrf52832\src\main.cpp.o] Error 1
======================================== [FAILED] Took 1.28 seconds ========================================
The terminal process terminated with exit code: 1
```

valeros, 2020-03-11:
Hi @beegee-tokyo ! Thanks for the report. I'm not able to reproduce the issue with the latest platform from the development branch. It looks like this problem has been fixed in the latest Adafruit code `v0.18.5`. Could you please retest your project with the upstream version of the platform ?

```
[env:adafruit_feather_nrf52832]
platform = https://github.com/platformio/platform-nordicnrf52.git
board = adafruit_feather_nrf52832
framework = arduino
```
Thanks!

beegee-tokyo, 2020-03-12:
Confirmed. Upstream version of the platform fixes the problem.

To test I defined 2 [env] in platformio.ini, one using    
```platform = https://github.com/platformio/platform-nordicnrf52.git```    
the other one with    
```platform = nordicnrf52```    
The first one compiles, the second one fails.

Thank you.

closed 2020-03-12 by beegee-tokyo