platformio/platform-nordicnrf52 #72: Add Support for Adafruit Feather nrf52840 Sense

ptyork, 2020-03-10:
Not sure what else to say in the description. Can support be added for this board:

https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/variants/feather_nrf52840_sense

Thought it might be similar enough to the Express to work, but no.

Thanks.

closed 2020-03-10 by valeros

ptyork, 2020-03-10:
First, Дуже дякую!

Second, though, I'm not terribly familiar with the PIO build system. All I did to try to make this work was change the platformio.ini file as follows:
```
[env]
; platform = nordicnrf52
platform = https://github.com/platformio/platform-nordicnrf52.git
framework = arduino
monitor_speed = 115200

[env:express]
board = adafruit_feather_nrf52840

[env:sense]
board = adafruit_feather_nrf52840_sense
```
The express target correctly creates a firmware.zip file, bu the sense target creates a firmware.hex file. The hex file fails to upload with a "File is not a zip file" error. Likewise there is a warning in the build:

Warning! Cannot find an appropriate softdevice binary!

Are those related? Any advice?

valeros, 2020-03-11:
> First, Дуже дякую!

You're welcome!  :+1: 

> The express target correctly creates a firmware.zip file, bu the sense target creates a firmware.hex file. The hex file fails to upload with a "File is not a zip file" error. 

Indeed, `adafruit_feather_nrf52840_sense` board doesn't have a bootloader binary in the framework package, but In my opinion, it shouldn't prevent firmware uploading. I've made some changes to the upload process, so please update the upstream version of the platform:
`pio platform update https://github.com/platformio/platform-nordicnrf52.git` and try again. Thanks!

ptyork, 2020-03-11:
That did it! Thanks again for your extremely prompt help!! Got me back up and running with these new boards with almost no delay.

commit 630c7d7e (2020-03-10): Add new board Adafruit Feather nrf52840 Sense // Resolve #72