platformio/platform-nordicnrf52 #78: Add nrf52833_DK

Olfox59, 2020-05-15:
What kind of issue is this?

- [ ] **Question**.
      Is it possible to add Nordic nrf52833_DK Board please :).
It should be quiet similar with the nrf52832 board . I tried but dts is different because zephyr example led doesn't blink.

- [ ] **PlatformIO IDE**.
For VS 

- [ ] **Development Platform or Board**.
      nrf52833_DK

------------------------------------------------------------------

### Configuration

**Operating system**:
Linux Kubuntu 64bit v2020

**PlatformIO Version** (`platformio --version`):
VIsual Studio Code

brandonOpener, 2021-01-26:
+1 for nrf52833.  I'm working with the BL653 dev kit, chip support would be excellent, even if it takes some finagling to get my board up and running.

MarkusBansky, 2021-08-06:
Is there any update on this issue, this board is already becoming popular and setting up the Mesh sample using board preset for nRF52840 for example does not work. 

Does anyone have any example of how to set up the nRF52833 DK project?

commit 004d0123 (2021-08-10): Added nRF52833 DK board (#121)

Resolve #78
closed 2021-08-10 by valeros