platformio/platform-nordicnrf52 #84: Add support for Makerdiary nRF52840 M.2 DevKit

nitz, 2020-06-28:
Hello!

I'd love to see support for [Makerdiary's new nRF52840 based M.2 Module Dev Kit](https://makerdiary.com/collections/frontpage/products/nrf52840-m2-developer-kit). I see that the Makerdiary nRF52840-MDK [is supported wit the](https://docs.platformio.org/en/latest/boards/nordicnrf52/nrf52840_mdk.html#frameworks) `zephyr` framework, and it would be great to get the same treatment for their new M.2 module/kit. It'd be awesome to see the support extended for the `nordicrf42` platform as well.

I'm glad to put in some work testing and helping get this developed, I'm just unsure of where to start. (I tried adding a board.json, but it seems there's more to it than that.) I'd be glad to start taking a crack if someone could point me in the right direction. 

I've posted a sister issue on makerdiary/nrf52840-m2-devkit about this as well, [here](https://github.com/makerdiary/nrf52840-m2-devkit/issues/2)

Cheers

valeros, 2020-06-30:
Hi @nitz ! Unfortunately, there is no official support for this board in Zephyr framework https://docs.zephyrproject.org/latest/boards/index.html

nitz, 2020-06-30:
I don't find that too surprising, given that @makerdiary just launched these recently :)

Is that a requirement for PIO to support boards, they'll need to be supported in the upstream proper? If not, this new board [is capable](https://github.com/makerdiary/nrf52840-m2-devkit/blob/master/west.yml) of running Zephyr, with the `west.yml` in their repo there, so it seems @makerdiary has at least put in the work to get that going. If it's required to be in upstream, then I can start looking at the process to get that going in Zephyr!

I also haven't run the Nordic SDK on it yet (as I've just been playing with the CircuitPython they ship the board with), but with it's nRF52840 as it's SoC, it's certainly [capable of running the nRF5 SDK](https://wiki.makerdiary.com/nrf52840-m2-devkit/nrf5-sdk/), so I could look at getting it supporting that in PIO too. I don't suppose there's a guide/example, or checklist for things I would need to look at to getting a new board in and going?

Apologies for being so obtuse with all of this — It's my first time digging into trying to help expand PIO!

valeros, 2020-06-30:
Thanks for your efforts. It looks like this board can be used with Zephyr only via custom configuration. We tend to support the stable versions of packages (including Zephyr), so the official support for this board will be added when it's released in Zephyr.

nitz, 2020-06-30:
That makes sense. I'll bark up that tree then, and see if I can get any traction in making that happen! Appreciate the quick responses!