platformio/platform-nordicnrf52 #90: error trying to create a new zephyr project with PlatformIO in a Win10 build 18363.959

rdmeneze, 2020-07-31:
Hi!
I'm trying to create a new project using the PlatformIO with a nrf52840-mdk board. 
Always I have been getting the error bellow: 

`PIO Core Call Error: "The current working directory C:\\Users\\Rafael Dias\\Documents\\PlatformIO\\Projects\\teste_nrf52840_mdk will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\Rafael Dias\\Documents\\PlatformIO\\Projects\\teste_nrf52840_mdk\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing nrf52840_mdk (platform: nordicnrf52; board: nrf52840_mdk; framework: zephyr)\r\n--------------------------------------------------------------------------------\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/nrf52840_mdk.html\r\nPLATFORM: Nordic nRF52 4.3.0 > Makerdiary nRF52840-MDK\r\nHARDWARE: NRF52840 64MHz, 256KB RAM, 1MB Flash\r\nDEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink, stlink)\r\nPACKAGES: \r\n - framework-zephyr 2.20300.200623 (2.3.0) \r\n - framework-zephyr-canopennode 0.0.0-alpha+sha.5c6b0566d5 \r\n - framework-zephyr-civetweb 0.0.0-alpha+sha.99129c5efc \r\n - framework-zephyr-cmsis 0.0.0-alpha+sha.542b2296e6 \r\n - framework-zephyr-fatfs 0.0.0-alpha+sha.9ee6b9b951 \r\n - framework-zephyr-hal-nordic 0.0.0-alpha+sha.742d8fb839 \r\n - framework-zephyr-libmetal 0.0.0-alpha+sha.3c3c9ec83b \r\n - framework-zephyr-littlefs 0.0.0-alpha+sha.0aefdda69d \r\n - framework-zephyr-loramac-node 0.0.0-alpha+sha.29e516ec58 \r\n - framework-zephyr-lvgl 0.0.0-alpha+sha.74fc2e753a \r\n - framework-zephyr-mbedtls 0.0.0-alpha+sha.4bf099f125 \r\n - framework-zephyr-mcuboot 0.0.0-alpha+sha.e88113bbeb \r\n - framework-zephyr-mcumgr 0.0.0-alpha+sha.5885efb7ca \r\n - framework-zephyr-mipi-sys-t 0.0.0-alpha+sha.957d46bc3c \r\n - framework-zephyr-open-amp 0.0.0-alpha+sha.724f7e2a45 \r\n - framework-zephyr-openthread 0.0.0-alpha+sha.a83d18cf18 \r\n - framework-zephyr-segger 0.0.0-alpha+sha.6fcf61606d \r\n - framework-zephyr-tinycbor 0.0.0-alpha+sha.40daca97b4 \r\n - framework-zephyr-tinycrypt 0.0.0-alpha+sha.3e9a49d267 \r\n - framework-zephyr-trusted-firmware-m 0.0.0-alpha+sha.7de2daa196 \r\n - tool-cmake 3.16.4 \r\n - tool-dtc 1.4.7 \r\n - tool-ninja 1.9.0 \r\n - tool-sreccat 1.164.0 (1.64) \r\n - toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1)\r\nInstalling Zephyr's Python dependencies\r\n'c:\\users\\rafael' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n*** Error 1\r\nModuleNotFoundError: No module named 'yaml':\r\n File \"C:\\Users\\Rafael Dias\\.platformio\\penv\\Lib\\site-packages\\platformio\\builder\\main.py\", line 168:\r\n env.SConscript(\"$BUILD_SCRIPT\")\r\n File \"C:\\Users\\Rafael Dias\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 597:\r\n return _SConscript(self.fs, *files, **subst_kw)\r\n File \"C:\\Users\\Rafael Dias\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 286:\r\n exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\r\n File \"C:\\Users\\Rafael Dias\\.platformio\\platforms\\nordicnrf52\\builder\\main.py\", line 194:\r\n target_elf = env.BuildProgram()\r\n File \"C:\\Users\\Rafael Dias\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 219:\r\n return self.method(*nargs, **kwargs)\r\n File \"C:\\Users\\Rafael Dias\\.platformio\\packages\\framework-zephyr\\scripts\\platformio\\platformio-build-pre.py\", line 29:\r\n env.ProcessProgramDeps()\r\n File \"C:\\Users\\Rafael Dias\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 219:\r\n return self.method(*nargs, **kwargs)\r\n File \"C:\\Users\\Rafael Dias\\.platformio\\penv\\Lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 118:\r\n env.BuildFrameworks(env.get(\"PIOFRAMEWORK\"))\r\n File \"C:\\Users\\Rafael Dias\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 219:\r\n return self.method(*nargs, **kwargs)\r\n File \"C:\\Users\\Rafael Dias\\.platformio\\penv\\Lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 330:\r\n SConscript(env.GetFrameworkScript(f), exports=\"env\")\r\n File \"C:\\Users\\Rafael Dias\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 660:\r\n return method(*args, **kw)\r\n File \"C:\\Users\\Rafael Dias\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 597:\r\n return _SConscript(self.fs, *files, **subst_kw)\r\n File \"C:\\Users\\Rafael Dias\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 286:\r\n exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\r\n File \"C:\\Users\\Rafael Dias\\.platformio\\platforms\\nordicnrf52\\builder\\frameworks\\zephyr.py\", line 28:\r\n SConscript(\r\n File \"C:\\Users\\Rafael Dias\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 660:\r\n return method(*args, **kw)\r\n File \"C:\\Users\\Rafael Dias\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 597:\r\n return _SConscript(self.fs, *files, **subst_kw)\r\n File \"C:\\Users\\Rafael Dias\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 286:\r\n exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\r\n File \"C:\\Users\\Rafael Dias\\.platformio\\packages\\framework-zephyr\\scripts\\platformio\\platformio-build.py\", line 45:\r\n import yaml\r\n========================== [FAILED] Took 3.82 seconds =========================="
`

valeros, 2020-07-31:
Looks like the issue is caused by a whitespace in your username. You can try to override the default core directory in your `platformio.ini` file, for example

```
[platformio]
core_dir = C:\\.platformio
```

valeros, 2020-11-09:
Closed as stale.

closed 2020-11-09 by valeros

maxgerhardt, 2024-01-21:
There must be a path escape issue somewhere in the installation routine for Zephyr dependencies. This user seems to be hitting them too: https://community.platformio.org/t/failing-to-import-a-zephyr-example-on-a-windows-machine/38010