platformio/platform-nordicnrf52 #91: [nRF52840 DK]Debugger stuck on vPortStartFirstTask()

LaBus3, 2020-08-20:
Hi,
I have noticed that without specifying softdevice even the simple blinky does not work. 
In both with and without softdevice situations the debugger steps in a loop at vPortStartFirstTask() function. The program runs alright on direct uploading. 
Device: nrF52840 DK
OS: Pop!_OS 20.04 LTS
BSP: Adafruit
IDE: VScode platformio

Any reason for this issue? Is there any additional step for enabling debugging? Does platformio not support freertos debugging?
Thank you
NB: Does the device use internal flash or external flash for programming in case of DK?

valeros, 2020-08-20:
Hi @LaBus3 ! Unfortunately, I don't have this board at hand to investigate. Could you please paste here your `platformio.ini` file?

> Does platformio not support freertos debugging?

You should be able to debug any application (bare-metal or RTOS). 

> NB: Does the device use internal flash or external flash for programming in case of DK?

Internal flash is used.

> Also pio upload does not seep to upload the bootloader along with the code, is this the default behavior?

As far as I remember, it depends on upload method. But when you're debugging only user application is uploaded to the target.

LaBus3, 2020-08-20:
> Hi @LaBus3 ! Unfortunately, I don't have this board at hand to investigate. Could you please paste here your `platformio.ini` file?

>> [env:nrf52840_dk_adafruit]
>> platform = nordicnrf52
>> board = nrf52840_dk_adafruit
>> framework = arduino
>> build_flags = -DNRF52_S140 

> As far as I remember, it depends on upload method. But when you're debugging only user application is uploaded to the target.

The code fails to run without the bootloader, without the bootloader I cannot upload the code via USB. Either way bootloader seem to be required.
Is there any upload-bootloader command to be used as part of an "initialisation" step? What is the common workflow for fully wiped nRF devices?

LaBus3, 2020-08-20:
```
void vPortStartFirstTask( void )
{
    __asm volatile(
#if defined(__SES_ARM)
                    " ldr r0, =_vectors \n"     /* Locate the stack using _vectors table. */
#else
                    " ldr r0, =__isr_vector \n" /* Locate the stack using __isr_vector table. */
#endif
                    " ldr r0, [r0]          \n"
                    " msr msp, r0           \n" /* Set the msp back to the start of the stack. */
                    " cpsie i               \n" /* Globally enable interrupts. */
                    " cpsie f               \n"
                    " dsb                   \n"
                    " isb                   \n"
#ifdef SOFTDEVICE_PRESENT
                    /* Block kernel interrupts only (PendSV) before calling SVC */
                    " mov r0, %0            \n"
                    " msr basepri, r0       \n"
#endif
                    " svc 0                 \n" /* System call to start first task. */
                    "                       \n"
                    " .align 2              \n"
#ifdef SOFTDEVICE_PRESENT
                    ::"i"(configKERNEL_INTERRUPT_PRIORITY  << (8 - configPRIO_BITS))
#endif
                );
}
```
```
0x0002d20c: 07 48           	ldr	r0, [pc, #28]	; (0x2d22c)
0x0002d20e: 00 68           	ldr	r0, [r0, #0]
0x0002d210: 80 f3 08 88     	msr	MSP, r0
0x0002d214: 62 b6           	cpsie	i
0x0002d216: 61 b6           	cpsie	f
0x0002d218: bf f3 4f 8f     	dsb	sy
0x0002d21c: bf f3 6f 8f     	isb	sy
0x0002d220: 4f f0 e0 00     	mov.w	r0, #224	; 0xe0
0x0002d224: 80 f3 11 88     	msr	BASEPRI, r0
0x0002d228: 00 df           	svc	0
0x0002d22a: 00 bf           	nop
```

I am able to step through the assembly code, but then the execution breaks within the same function.
xpsr Interrupt Number register : 0x03 (Hard Fault??)

cujomalainey, 2020-08-20:
unless you changed the softdevice you dont need to specify it in the build flags, it is part of the board definition. Also just for the record, i have had some difficulty debugging this board, but I was also playing with O flags

tyeth, 2020-12-13:
Late to the party, and more confused that you lot by the sounds of it... I have the official nordic NRF52840-DK board, no adafruit nothing. Sadly the official boards supported have the wrong pin assignments and only the adafruit one includes the correct pins in header files. It is also the only one to offer uploading a bootloader, along with building hex files and dfu packages (zips) (all called DFU images). I'm seeing the same debug issue, and worse no results from test. Seems to just hang and no board reset helps. I fear it may erase the bootloader and then only upload the application. I thought it had the ability to only erase the application portion.

I had not achieved any useful debugging, but now following your advice, I have clicked the restart debugging and when warned that preLaunchTask "undefined" failed with exit code 1, choosing debug anyway rather than abort or show errors. It works unlike aborting the first debug and then triggering again. Thank you a million times...

josephduchesne, 2021-03-25:
I'm having a similar issue:

Device: Adafruit Feather nrF52840 Express
OS: Windows 10
BSP: Adafruit
IDE: VScode platformio

I hit debug, and get a breakpoint at the start of main, then continue, and get stuck on `vPortStartFirstTask` with interrupt number 0x03. Unlike LaBus3, restarting debug at this point doesn't dodge the issue. I get a permissions error on firmware.elf (since I think GDB holds it open, which may be a windows specific issue unrelated to this problem).

xlla, 2021-06-10:
same issue here. click restart will solve it temporary.