platformio/platform-nordicnrf52 #92: Change Low Frequency Clock Source

OmarAlkassab, 2020-08-27:
### **Problem description**
When I program the Raytac MDBT42Q-512KV2 (NRF52832 based module), the code work correctly until reach the functions related to the BLE Peripheral `blePeripheral.begin()`, at this line the module hangs. I debugged it and I stated that the code go into endless loop in the function SVCALL.

### **Hardware description**
I'm trying to program the module. I integrate it in my custom board without external low frequency oscillator (32.678KHz). What change should I do in order to let the module use the internal RC oscillator instead of external oscillator?

### **Work around**
I used the Arduino IDE to program it using the option of RC internal oscillator, see the following screenshot. I copied the code to the Arduino IDE then the code works perfectly.
[image].

### **_What I need is:_**
How to tell the compiler to use the internal oscillator instead of the default external oscillator option?

OmarAlkassab, 2020-09-04:
### I found the solution:

In the platformio.ini file we should add the following line:

`build_flags = -DUSE_LFRC`
In case we need to use the **Softdevice** with the **BLEPeripheral** library, we need to add also the flag `-DNRF52_S132`. Then the build flags will be:
`build_flags = -DNRF52_S132 -DUSE_LFRC`

closed 2020-09-04 by OmarAlkassab

j45p41, 2023-10-14:
You are a life saver my friend.

So obvious I could not find it anywhere.

dennowiggle, 2024-01-07:
@OmarAlkassab Thank you so much for this solution. 

My very simple program controlling LED's would lockup while calling delay() after a period of time running successfully.  (No BLE anywhere).

The delay() function uses millis() which uses the counter driven by RTC1. I looked everywhere for configuration of the 32kHz clock input. It took several days to find your solution of setting the build flag. Thank you.

I wish this was documented on the platform IO page settings for applicable boards. e.g.
https://docs.platformio.org/en/latest/boards/nordicnrf52/bbcmicrobit_v2.html#board-nordicnrf52-bbcmicrobit-v2