platformio/platform-nordicnrf52 #93: Add support for Pinnacle 100 dev kit board

is-none, 2020-09-03:
Hi, I would like you to add support for Pinnacle 100 dvk board (P/N: 453-00011)

- [X] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

https://github.com/LairdCP/zephyr_boards/blob/master/pinnacle_100_dvk/doc/index.rst

https://github.com/LairdCP/zephyr_boards/tree/master/pinnacle_100_dvk

thx

is-none, 2020-09-08:
Hello, I do not want to put any pressure on you guys but I need to finish a project within the next 3 months.
So, I made a custom board that contains the minimum required as I see it.
But, I'm not sure about the "_connectivity_" part where I added: **cat-m1**, **nb-iot** and **lte**.
I'm guessing it's just to display these connectivity "strings" in Platformio IDE in the connectivity section of the board. Right?
I also added an "f" at the end of the CPU name: "cpu": "cortex-m4f". Do I need to specify the exact model?
Do I need to add anything to connect as SWD with jlink segger?

````json
{
  "build": {
    "cpu": "cortex-m4f",
    "f_cpu": "64000000L",
    "mcu": "nrf52840",
    "zephyr": {
       "variant": "pinnacle_100_dvk"
    }
  },
  "connectivity": [
    "cat-m1",
    "nb-iot",
    "lte",
    "bluetooth"
  ],
  "debug": {
    "default_tools": [
      "jlink"
    ],
    "jlink_device": "nRF52840_xxAA",
    "onboard_tools": [
      "jlink"
    ],
    "svd_path": "nrf52840.svd"
  },
  "frameworks": [
    "zephyr"
  ],
  "name": "Pinnacle 100 Development Kit",
  "upload": {
    "maximum_ram_size": 262144,
    "maximum_size": 1048576,
    "protocol": "jlink"
  },
  "url": "https://github.com/LairdCP/zephyr_boards/blob/master/pinnacle_100_dvk/doc/index.rst",
  "vendor": "Laird Connectivity"
}
````

With this information [Pinnacle 100 Specification Summary](https://www.mouser.ca/pdfDocs/Pinnacle_100_DS.pdf#%5B%7B%22num%22%3A47%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C48%2C670%2C0%5D) I think the minimum requirement for the .json file is there. 

Next, to set Zephyr prj.conf file I add the code below I found here [Laird Pinnacle 100](https://github.com/LairdCP/zephyr_boards/blob/master/pinnacle_100_dvk/pinnacle_100_dvk_defconfig)
````

# Copyright (c) 2020 Laird Connectivity

# SPDX-License-Identifier: Apache-2.0

CONFIG_SOC_SERIES_NRF52X=y
CONFIG_SOC_NRF52840_QIAA=y
CONFIG_BOARD_PINNACLE_100_DVK=y

# Enable MPU
CONFIG_ARM_MPU=y

# Enable RTT
CONFIG_USE_SEGGER_RTT=y

# enable GPIO
CONFIG_GPIO=y

# enable uart driver
CONFIG_SERIAL=y

# enable console
CONFIG_CONSOLE=y
CONFIG_UART_CONSOLE=y

# additional board options
CONFIG_GPIO_AS_PINRESET=y

# Modem driver
CONFIG_MODEM=y
CONFIG_MODEM_RECEIVER=y
CONFIG_MODEM_HL7800=y

# Networking
CONFIG_NETWORKING=y
CONFIG_NET_IPV4=y
CONFIG_NET_CONFIG_NEED_IPV4=y
CONFIG_NET_UDP=y
# disable IPv6 because it is not yet supported for the HL7800 driver
CONFIG_NET_IPV6=n
CONFIG_NET_CONFIG_NEED_IPV6=n
````
Actually, I didn't give a try... I'll see tomorrow.
Thanks in advance for helping me out!

valeros, 2020-09-08:
> I'm guessing it's just to display these connectivity "strings" in Platformio IDE in the connectivity section of the board. Right?

That's correct.

> I also added an "f" at the end of the CPU name: "cpu": "cortex-m4f". Do I need to specify the exact model?

There is no need to add that extra `f`. The CPU model should be `cortex-m4`.

> Do I need to add anything to connect as SWD with jlink segger?

Not sure I understand, but if that board have an embedded JLink probe , it should work out of the box.

The problem you may face is that there is no official support for this board in the Zephyr v2.3.0:
https://docs.zephyrproject.org/2.3.0/boards/index.html
So you will need to add that custom board files in your project and consequently configure your project to use that files.

is-none, 2020-09-09:
I tried it but as you said the Pinnacle 100 folder (the board) was missing in the directory. So I added this folder [https://github.com/LairdCP/zephyr_boards/tree/master/pinnacle_100_dvk](https://github.com/LairdCP/zephyr_boards/tree/master/pinnacle_100_dvk) to this directory `~/.platformio/packages/framework-zephyr/boards/arm` and i got this error : 

> ~/.platformio/packages/framework-zephyr/boards/arm/pinnacle_100_dvk/pinnacle_100_dvk_defconfig:31: warning: attempt to assign the value 'y' to the undefined symbol MODEM_HL7800

Then I searched in

> ~/.platformio/packages/framework-zephyr/boards/arm/pinnacle_100_dvk/pinnacle_100_dvk_defconfig

and commented this `CONFIG_MODEM_HL7800=y`

After that no error, but thing is I need the HL7800 semiconductor. I looked for `HL7800` on Zephyr and if I am not wrong it is on only support on the 2.4 rc1 version of Zephyr. What I don't understand it how Laird has already examples of projects with this board using Zephyr 2.3 and HL7800 [here](https://github.com/LairdCP/Pinnacle_100_Sample_Applications/tree/master/apps).  For each of `prj.conf` file of these examples there is a `CONFIG_MODEM_HL7800_......` call.