platformio/platform-nordicnrf52 #95: Remote upload error Adafruit NRF52832

Swifster, 2020-10-03:
Hi! My remote works fine with other stuff, just not with one of the adafruit chips. Has anyone got experience with this error?

PIO Core CLI is installed on RPI4 and also te target is connected to the RPI4.

[rasp4] Uploading .pio/build/adafruit_feather_nrf52832/firmware.hex
[rasp4] Traceback (most recent call last):
[rasp4]   File "/home/pi/.local/bin/adafruit-nrfutil", line 8, in <module>
    sys.exit(cli())
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/nordicsemi/__main__.py", line 288, in serial
    dfu = Dfu(package, dfu_transport=serial_backend)
  File "/home/pi/.local/lib/python3.7/site-packages/nordicsemi/dfu/dfu.py", line 66, in __init__
    self.manifest = Package.unpack_package(self.zip_file_path, self.unpacked_zip_path)
  File "/home/pi/.local/lib/python3.7/site-packages/nordicsemi/dfu/package.py", line 361, in unpack_package
    with ZipFile(package_path, 'r') as pkg:
  File "/usr/lib/python3.7/zipfile.py", line 1222, in __init__
    self._RealGetContents()
  File "/usr/lib/python3.7/zipfile.py", line 1289, in _RealGetContents
    raise BadZipFile("File is not a zip file")
zipfile.BadZipFile: File is not a zip file
*** [upload] Error 1
[rasp4] ========================== [FAILED] Took 5.39 seconds ==========================
The terminal process "platformio 'remote', 'run', '--target', 'upload'" terminat

cujomalainey, 2020-10-03:
Sounds like one of your packages is corrupted. Might need to delete them manually

Swifster, 2020-10-03:
I did a clean install of OS and PIO and PIO remote and got the same result.  But i can try to delete more stuff and see what happens.

cujomalainey, 2020-10-03:
you cleaned `~/.platformio`?

Swifster, 2020-10-04:
I installed everything new. On newly installed os.

cujomalainey, 2020-10-04:
The package is busted, seems to be a far too common issue since 5.x upgrade

GreSuh, 2021-06-20:
having the same problem. Is there an update on this issue?

Swifster, 2021-06-22:
Hi! Did not find a fix for this. Also if i use segger the upload does not work.