platformio/platform-nordicnrf52 #97: uint8_t and uint32_t get mapped wrong by Intellisense for Arduino Nano 33 BLE

Alex-Evo, 2020-10-16:
If I create a new project for an "Arduino Nano 33 BLE" and create test variables of the "stdint" types, I get the following situation:

[image]

Hovering over the types gives the following results:
**int8_t**: typedef __int8_t int8_t => OK

**uint8_t**:
```
<error-type> uint8_t
@brief Function for reading data from the SPI receiver register.
@param[in] p_reg Pointer to the peripheral registers structure.
@return RX data received.
variable "uint8_t" is not a type name
```

F12 takes me to line 228 "[...]/.platformio/packages/framework-arduino-nrf52-mbedos/cores/arduino/mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_15_0/modules/nrfx/hal/nrf_spi.h"

This does not make any sense, but it looks like Intellisense maps the definition of uint8_t to the return type of the macro defined in that line.

**int16_t, uint16_t, int32_t**: all OK

**uint32_t**:
```
typedef <error-type> uint32_t(<error-type>)
! The RND Generate vector function pointer type definition.
The prototype intendent for External and CRYS internal RND functions
pointers definitions.
Full description can be found in ::CRYS_RND_GenerateVector function API.
```

F12 takes me to line 206 of "[...]/.platformio/packages/framework-arduino-nrf52-mbedos/cores/arduino/mbed/features/cryptocell/FEATURE_CRYPTOCELL310/include/crys_rnd.h"

Also here, it looks like Intellisense maps the definition of uint32_t to the return type of the macro defined in that line.

Any help (can also be hacks or workarounds) would be highly appreciated. From past issues, it looks like @sean-mcmanus has been knowledgeable about these topics in the past...

I'm running the following configuration:
PlatformIO Core: 4.3.4
PlatformIO Home: 3.2.3
ms-vscode.cpptools: 0.28.3 (also tried downgrading to 0.26.0 with same result)
VS Code version info:
Version: 1.46.1
Commit: cd9ea6488829f560dc949a8b2fb789f3cdc05f5d
Date: 2020-06-17T21:17:14.222Z (2 wks ago)
Electron: 7.3.1
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
OS: Darwin x64 19.5.0

kaliberr44, 2021-02-15:
I have the same issue.

The build succeeds, but the IntelliSense is showing problems for `uint8_t` and for the Nano 33 BLE SerialUSB port and ArduinoBLE classes and enums as well 9 axis IMU.
```
#include <Arduino.h>
#include <ArduinoBLE.h>
#include <Arduino_LSM9DS1.h>
```

```
identifier "BLEService" is undefined
identifier "BLECharacteristic" is undefined
identifier "BLENotify" is undefined
identifier "BLERead" is undefined

identifier "SerialUSB" is undefined

identifier "IMU" is undefined
```

Every identifier from list is marked with red underline in code, and hover shows it as undefined, but `Go to definition` is working.

GunZi200, 2021-02-26:
@kaliberr44 kaliberr44 
> I have the same issue.
> 
> The build succeeds, but the IntelliSense is showing problems for `uint8_t` and for the Nano 33 BLE SerialUSB port and ArduinoBLE classes and enums as well 9 axis IMU.
> 
> ```
> #include <Arduino.h>
> #include <ArduinoBLE.h>
> #include <Arduino_LSM9DS1.h>
> ```
> 
> ```
> identifier "BLEService" is undefined
> identifier "BLECharacteristic" is undefined
> identifier "BLENotify" is undefined
> identifier "BLERead" is undefined
> 
> identifier "SerialUSB" is undefined
> 
> identifier "IMU" is undefined
> ```
> 
> Every identifier from list is marked with red underline in code editor, and hover shows it as undefined, but `Go to definition` is working.

Just a guess, but does adding:

```json
"defines": [
                "BOARD_PCA_10059", 
                "NRF52840_XXAA" 
            ],
```

to your `c_cpp_properties.json` help? I'm not sure if its applicaple to PlatformIO but I had a similar problem with `uint32_t` on my nRF52832 (using NRF5 SDK). https://github.com/microsoft/vscode-cpptools/issues/6968 if your code compiles and works fine then this is probably just an intellisense issue.

Regarding the other identifiers... is it possible you haven't told VScode where to look for these header files? Does VScode complain? If you hover over the the `#include` does VScode give you suggestion to add a specific include path .

kaliberr44, 2021-03-15:
I had "NRF52840_XXAA" in defines, added "BOARD_PCA_10059", but the error still occurs.

The code always compile fine. It is the annoying Intellisense issue (the same as mentioned by OP).
No problems when hovering over `#include`. As stated I can even do directly to definition of undefined identifier, but the annoying thing is to have everything underlined in red because of this Intellisense issue.

valeros, 2021-03-15:
Hi guys, could someone put together a simple example to reproduce the issue? I tried a basic code with fixed-length types and it worked just fine with the latest VSCode and ArduinoCore-mbed v1.3.2 on both Windows and Ubuntu:
[image]