platformio/platform-nordicnrf52 #98: Support for nrfutil as upload_protocol for zephyr framework

LeoDJ, 2020-11-20:
Trying to upload any code built with the zephyr framework results in the following error: 
`*** [upload] IndexError : list index out of range`

Example platformio.ini:
```
[env:adafruit_feather_nrf52840]
platform = nordicnrf52
board = adafruit_feather_nrf52840
framework = zephyr
```

After digging around a bit in `nordicnrf52/builder/main.py` it seems to me that the nrfutil uploader is only available if you use the Arduino framework. But I want to be able to upload my existing code built with zephyr via nrfutil too.

https://github.com/platformio/platform-nordicnrf52/blob/a7d7f2d16e1c506013c64b4bad415776fded459d/builder/main.py#L56-L59
https://github.com/platformio/platform-nordicnrf52/blob/a7d7f2d16e1c506013c64b4bad415776fded459d/builder/main.py#L68-L70

### Full upload log
```
> pio run -t upload -v
Processing adafruit_feather_nrf52840 (platform: nordicnrf52; board: adafruit_feather_nrf52840; framework: zephyr)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/adafruit_feather_nrf52840.html
PLATFORM: Nordic nRF52 (5.0.0) > Adafruit Feather nRF52840 Express
HARDWARE: NRF52840 64MHz, 243KB RAM, 796KB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-zephyr 2.20400.201110 (2.4.0)
 - framework-zephyr-canopennode 0.0.0-alpha+sha.468d350028
 - framework-zephyr-civetweb 0.0.0-alpha+sha.99129c5efc 
 - framework-zephyr-cmsis 0.0.0-alpha+sha.542b2296e6
 - framework-zephyr-fatfs 0.0.0-alpha+sha.13697783bf
 - framework-zephyr-hal-nordic 0.0.0-alpha+sha.d8a6ea9695
 - framework-zephyr-hal-st 0.0.0-alpha+sha.5b3ec3e182
 - framework-zephyr-libmetal 0.0.0-alpha+sha.0b23894a04
 - framework-zephyr-littlefs 0.0.0-alpha+sha.9e4498d1c7
 - framework-zephyr-loramac-node 0.0.0-alpha+sha.3f545d76a2
 - framework-zephyr-lvgl 0.0.0-alpha+sha.928b61c7c8
 - framework-zephyr-mbedtls 0.0.0-alpha+sha.aef137b1af
 - framework-zephyr-mcuboot 0.0.0-alpha+sha.a5d79cf8cc
 - framework-zephyr-mcumgr 0.0.0-alpha+sha.5051f9d900
 - framework-zephyr-mipi-sys-t 0.0.0-alpha+sha.957d46bc3c
 - framework-zephyr-open-amp 0.0.0-alpha+sha.724f7e2a45
 - framework-zephyr-openthread 0.0.0-alpha+sha.07f430dac6
 - framework-zephyr-segger 0.0.0-alpha+sha.874d9e9696
 - framework-zephyr-tinycbor 0.0.0-alpha+sha.40daca97b4
 - framework-zephyr-tinycrypt 0.0.0-alpha+sha.3e9a49d267
 - framework-zephyr-trusted-firmware-m 0.0.0-alpha+sha.143df67555
 - tool-bossac-nordicnrf52 1.10901.201022 (1.9.1)
 - tool-cmake 3.16.4
 - tool-dtc 1.4.7
 - tool-ninja 1.9.0
 - tool-openocd 2.1000.190707 (10.0)
 - tool-sreccat 1.164.0 (1.64)
 - toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
*** [upload] IndexError : list index out of range
```

LeoDJ, 2020-11-20:
I'm not well versed in the platformio platform structure, so I can't easily implement that myself.

But I created the following workaround for now:
- Install nrfutil from pip (globally)
- Use the following options in `platformio.ini`:
```
upload_protocol = custom
upload_port = COM4
upload_command = 
    nrfutil pkg generate --hw-version 52 --sd-req=0x00 --application $SOURCE --application-version 1 ${SOURCE}.zip 
    nrfutil dfu usb-serial -pkg ${SOURCE}.zip -p $UPLOAD_PORT
```

PabloAbrahamAvanade, 2021-01-30:
I am experiencing the same issue. Is there any official fix for it?

pinsdorf, 2021-05-21:
I have the same issue when using an Adafruit Bluefruit nRF52832 board with Zephyr framework. The upload log also fails with `*** [upload] IndexError : list index out of range`. Platformio (5.1.1) and all packages are up-to-date. Is there any solution to this?

rajithavk, 2021-06-08:
Simply add the upload_protocol to platformio.ini 
for an example if you are using segger jlink,

`upload_protocol = jlink`

LeoDJ, 2021-06-08:
@rajithavk no, I would still like to use nrfutil and if I put `upload_protocol = nrfutil` it'll behave the same way.