platformio/platform-nordicnrf52 #101: Problem with flashing a BLE code in nRF52832

pvonmoradi, 2021-01-05:
I'm using a custom board designed around nRF52832. Basic blinky works but the following code which used BLE as a serial port, doesn't.  
```
[env:my_env]
platform = nordicnrf52
framework = arduino
board = myboard
upload_protocol = blackmagic
build_flags = -DNRF52_S132
lib_deps =
  sandeepmistry/BLEPeripheral @ ^0.4.0
```
```Cpp
#include "Arduino.h"
#include "BLESerial.h"
#include "variant.h"

BLESerial bleSerial;

void setup() {
    Serial.setPins(PIN_SERIAL_RX, PIN_SERIAL_TX);
    Serial.begin(115200);
    Serial.println("Hi from nRF52832!");

    bleSerial.setLocalName("UART over BLE!");
    bleSerial.begin();
    bleSerial.println("Hi from ble!");
}
```
And here is `myboard.json`:  
```json
{
  "build": {
    "arduino":{
      "ldscript": "nrf52832_xxaa.ld"
    },
    "core": "nRF5",
    "cpu": "cortex-m4",
    "f_cpu": "64000000L",
    "mcu": "nrf52832",
    "variant": "Generic",
    "extra_flags": "-DARDUINO_GENERIC"
  },
  "connectivity": [
    "bluetooth"
  ],
  "frameworks": [
    "arduino"
  ],
  "name": "My Board v1",
  "upload": {
    "maximum_ram_size": 65536,
    "maximum_size": 524288,
    "protocol": "blackmagic",
    "protocols": [
      "nrfjprog",
      "blackmagic"
    ]
  },
  "url": "https://developer.mbed.org/platforms/Nordic-nRF52-DK/",
  "vendor": "Nordic"
}
```
In fact, pio build the new code and flashes the firmware but the **old** firmware gets executed. So I guess there might be something wrong with build script or linker script.  
The parts I'm suspicious about:  
nrf52832_xxaa.ld -> armgcc_s132_nrf52832_xxaa.ld ? (tested doesn't work)
any extra flags in boards.json or project ini config?

valeros, 2021-01-06:
Hi @pvonmoradi !

> nrf52832_xxaa.ld -> armgcc_s132_nrf52832_xxaa.ld ? (tested, pio can't find it)

Could you please explain this line? What exactly can't PIO find? `armgcc_s132_nrf52832_xxaa` is the default linker script if the S132 SoftDevice is used.

Also, does your application work if you compile it in Arduino IDE?

pvonmoradi, 2021-01-06:
@valeros Thanks for your reply.
I'm using a custom board so I created a `myboard.json`. In this section:
```
{
  "build": {
    "arduino":{
      "ldscript": "nrf52_xxaa.ld"
    },
```
if I change `nrf52_xxaa.ld` to `armgcc_s132_nrf52832_xxaa`, building the project, causes this error: 
```
make
platformio -f -c vim run
Processing my_env (platform: nordicnrf52; framework: arduino; board: myboard)
-------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/myboard.html
PLATFORM: Nordic nRF52 (6.0.0) > Biofeedback Central Node v1
HARDWARE: NRF52832 64MHz, 64KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic)
PACKAGES: 
 - framework-arduinonordicnrf5 1.700.201209 (7.0) 
 - tool-sreccat 1.164.0 (1.64) 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Warning! Cannot find an appropriate linker script for the required softdevice!
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 3 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <BLEPeripheral> 0.4.0
|   |-- <SPI> 1.0
Building in release mode
Linking .pio/build/my_env/firmware.elf
/home/pooya/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: cannot open linker script file armgcc_s132_nrf52832_xxaa.ld: No such file or directory
collect2: error: ld returned 1 exit status
*** [.pio/build/my_env/firmware.elf] Error 1
============================ [FAILED] Took 0.95 seconds ============================
Makefile:3: recipe for target 'all' failed
make: *** [all] Error 1
```
Haven't tested with Arduino IDE yet...

valeros, 2021-01-06:
> Warning! Cannot find an appropriate linker script for the required softdevice!

What is the reason that you used `nrf52_xxaa.ld` ? Could you change it to the identical `nrf52_xxaa.ld` in you board manifest?

pvonmoradi, 2021-01-06:
> > Warning! Cannot find an appropriate linker script for the required softdevice!
> 
> What is the reason that you used `nrf52_xxaa.ld` ? Could you change it to the identical `nrf52_xxaa.ld` in you board manifest?

Ah sorry I meant `nrf52_xxaa.ld`. `nrf52832_xxaa.ld` was a typo.

pvonmoradi, 2021-01-06:
Even for simple blinky (with `nrf52_xxaa.ld`), that works fine, `blackmagic -V .pio/build/my_env/firmware.elf` can not verify flashed firmware:  
```
λ blackmagic -V .pio/build/my_env/firmware.elf 
Using 1d50:6018 DFE3B6C6 Black Sphere Technologies Black Magic Probe (SWLINK), (Firmware v1.7.1)
Remote is Black Magic Probe (SWLINK), (Firmware v1.7.1) v1.7.1
Verify failed at flash region 0x00000000
```
So at the moment, ignore this unverified firmware issue.

valeros, 2021-01-06:
Could you please try to clean your project, then compile it in verbose mode and attach here the entire log.

pvonmoradi, 2021-01-06:
> Could you please try to clean your project, then compile it in verbose mode and attach here the entire log.

Blinky code (no BLE) -> Works fine
**boards/myboard.json**  
``` 
{
  "build": {
    "arduino":{
      "ldscript": "nrf52_xxaa.ld"
    },
    "core": "nRF5",
    "cpu": "cortex-m4",
    "f_cpu": "64000000L",
    "mcu": "nrf52832",
    "variant": "Generic",
    "extra_flags": "-DARDUINO_GENERIC"
  },
  "connectivity": [
    "bluetooth"
  ],
  "debug": {
    "default_tools": [
      "blackmagic",
      "jlink"
    ],
    "jlink_device": "nRF52832_xxAA",
    "onboard_tools": [
    ],
    "svd_path": "nrf52.svd"
  },
  "frameworks": [
    "arduino"
  ],
  "name": "Biofeedback Central Node v1",
  "upload": {
    "maximum_ram_size": 65536,
    "maximum_size": 524288,
    "protocol": "blackmagic",
    "protocols": [
      "nrfjprog",
      "blackmagic"
    ]
  },
  "url": "https://developer.mbed.org/platforms/Nordic-nRF52-DK/",
  "vendor": "Nordic"
}
```
**platformio.ini**
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:my_env]
platform = nordicnrf52
framework = arduino
board = myboard
upload_protocol = blackmagic
; build_flags = -DNRF52_S132
lib_deps =
  ; BLEPeripheral
  ; sandeepmistry/BLEPeripheral @ ^0.4.0
```
src/main.cpp
[... 114 lines cut ...]
AVAILABLE: blackmagic, nrfjprog
CURRENT: upload_protocol = blackmagic
Looking for BlackMagic port...

Warning! Please install `99-platformio-udev.rules`. 
More details: https://docs.platformio.org/page/faq.html#platformio-udev-rules

Auto-detected: /dev/ttyACM0
Uploading .pio/build/my_env/firmware.hex
Available Targets:
No. Att Driver
 1      Nordic nRF52 M4
 2      Nordic nRF52 Access Port 
0x000009f0 in yield ()
Loading section .text, size 0xcbc lma 0x0
Loading section .ARM.exidx, size 0x8 lma 0xcbc
Loading section .data, size 0x74 lma 0xcc4
Start address 0x434, load size 3384
Transfer rate: 21 KB/sec, 564 bytes/write.
Section .text, range 0x0 -- 0xcbc: matched.
Section .ARM.exidx, range 0xcbc -- 0xcc4: matched.
Section .data, range 0xcc4 -- 0xd38: matched.
Kill the program being debugged? (y or n) [answered Y; input not from terminal]
============================ [SUCCESS] Took 2.41 seconds ============================
```

pvonmoradi, 2021-01-06:
Blinky + BLE code -> Builds and uploads without error. 
Result -> Still the previous firmware is executed! (note the 2 seconds delay on blinks in this case. Does not happen. Still previous fast blink is seen. Also no BLE)
**boards/myboard.json**
```json
{
  "build": {
    "arduino":{
      "ldscript": "nrf52_xxaa.ld"
    },
    "core": "nRF5",
    "cpu": "cortex-m4",
    "f_cpu": "64000000L",
    "mcu": "nrf52832",
    "variant": "Generic",
    "extra_flags": "-DARDUINO_GENERIC"
  },
  "connectivity": [
    "bluetooth"
  ],
  "debug": {
    "default_tools": [
      "blackmagic",
      "jlink"
    ],
    "jlink_device": "nRF52832_xxAA",
    "onboard_tools": [
    ],
    "svd_path": "nrf52.svd"
  },
  "frameworks": [
    "arduino"
  ],
  "name": "Biofeedback Central Node v1",
  "upload": {
    "maximum_ram_size": 65536,
    "maximum_size": 524288,
    "protocol": "blackmagic",
    "protocols": [
      "nrfjprog",
      "blackmagic"
    ]
  },
  "url": "https://developer.mbed.org/platforms/Nordic-nRF52-DK/",
  "vendor": "Nordic"
}

```

**platformio.ini**
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:my_env]
platform = nordicnrf52
framework = arduino
board = myboard
upload_protocol = blackmagic
build_flags = -DNRF52_S132
lib_deps =
  ; BLEPeripheral
  sandeepmistry/BLEPeripheral @ ^0.4.0
```
[... 186 lines cut ...]
CURRENT: upload_protocol = blackmagic

Looking for BlackMagic port...
Warning! Please install `99-platformio-udev.rules`. 
More details: https://docs.platformio.org/page/faq.html#platformio-udev-rules

Auto-detected: /dev/ttyACM0
Uploading .pio/build/my_env/firmware.hex
Available Targets:
No. Att Driver
 1      Nordic nRF52 M4
 2      Nordic nRF52 Access Port 
0x00000984 in ?? ()
Loading section .text, size 0x3c58 lma 0x1c000
Loading section .ARM.exidx, size 0x8 lma 0x1fc58
Loading section .data, size 0x1ec lma 0x1fc60
Start address 0x1e358, load size 15948
Transfer rate: 37 KB/sec, 886 bytes/write.
Section .text, range 0x1c000 -- 0x1fc58: matched.
Section .ARM.exidx, range 0x1fc58 -- 0x1fc60: matched.
Section .data, range 0x1fc60 -- 0x1fe4c: matched.
Kill the program being debugged? (y or n) [answered Y; input not from terminal]
============================ [SUCCESS] Took 4.76 seconds ============================

```

valeros, 2021-01-06:
It's strange how your first attempt was compiled since you didn't specify the SoftDevice defint. Anyway, could you please upload somewhere a complete example as a ZIP to reproduce the issue?

pvonmoradi, 2021-01-06:
> It's strange how your first attempt was compiled since you didn't specify the SoftDevice defint. Anyway, could you please upload somewhere a complete example as a ZIP to reproduce the issue?

First example was not using BLE code. Should a softdevice always be used with nRF52s?
Here are the zip files: 
https://transfer.sh/Ovekj/nrf52-arduino-pio-blinky.zip
https://transfer.sh/T0TzV/nrf52-arduino-pio-BLE_and_slow_blink.zip
1- First test and upload blinky to a nRF52832
2- Then test and upload BLE and slow_blinky to a nRF52832
I'm using a bluepill as black magic probe.

valeros, 2021-01-06:
> Should a softdevice always be used with nRF52s?

It's fine to not use it, if your app doesn't use the BLE feature.

Strange, but I cannot reproduce the issue. I don't have a BMP, but I tried with nRF52-DK with on-board JLink and it works just fine with both examples. Is there any chance you could try to upload your app using a different debug probe?

pvonmoradi, 2021-01-06:
> It's fine to not use it if your app doesn't use the BLE feature.
>
> Strange, but I cannot reproduce the issue. I don't have a BMP, but I tried with nRF52-DK with on-board JLink and it works just fine [...]

1- JLink is the only alternative for nRF devices. I don't have one currently.  
That's also my suspicion cause it seems the bmp commands used do not clear the flash before upload. This is not the case for my custom makefiles though (using nrf-sdk).
Do you know where in the tree, the upload command using bmp is implemented? 

2- Also, in the BLE_with_slow_blinky, the linker script in the board manifest is not `armgcc_s132_nrf52832_xxaa`. Why is it working in your case? What command overrided the linker defined in the manifest?

valeros, 2021-01-06:
> Do you know where in the tree, the upload command using bmp is implemented?

Here is the command https://github.com/platformio/platform-nordicnrf52/blob/develop/builder/main.py#L287

> Also, in the BLE_with_slow_blinky, the linker script in the board manifest is not armgcc_s132_nrf52832_xxaa. Why is it working in your case? What command overrided the linker defined in the manifest?

The linker script file is selected dynamically, based on the softdevice used in the configuration. The entire procedure is implemented here:

https://github.com/platformio/builder-framework-arduino-nrf5/blob/809d9285bb81decabfce64c080a45a917e647f70/nrf5.py#L157-L165

pvonmoradi, 2021-01-06:
> > Do you know where in the tree, the upload command using bmp is implemented?
> 
> Here is the command https://github.com/platformio/platform-nordicnrf52/blob/develop/builder/main.py#L287

I was able to fix that problem with flashing new code with:
```
	arm-none-eabi-gdb -nx --batch \
        -ex "target extended-remote /dev/ttyGdbBmp" \
        -ex "monitor swdp_scan" \
        -ex "attach 1" \
        -ex "monitor erase_mass" \
        -ex "load" \
        -ex "compare-sections" \
        -ex "kill" \
        .pio/build/my_env/firmware.elf
```
The difference between this and the one in the tree is `-ex "monitor erase_mass" \`. Note that it **seems** this command resets nRF528**40**s in a way that you need a jlink to recover from (sets the SWD voltages to 1v8) so I don't know if it is advisable to patch it.
Still, I think I've isolated my issue to that -DNRF52_S132. When this switch is used, no blinky or any code could be run on my nRF52832. 
So either the firmware (elf file) and addresses and linkerscript for Jlink is not identical to bmp, or there is a problem on my part or bmp.

valeros, 2021-01-06:
It would be great if you could experiment with another debug tool. I can confirm that it works just fine with JLink.

pvonmoradi, 2021-01-07:
> The linker script file is selected dynamically, based on the softdevice used in the configuration. The entire procedure is implemented here:
> 
> https://github.com/platformio/builder-framework-arduino-nrf5/blob/809d9285bb81decabfce64c080a45a917e647f70/nrf5.py#L157-L165

I think the problem lies in `myboard.json` (you tested with https://github.com/platformio/platform-nordicnrf52/blob/develop/boards/nrf52_dk.json).
If the linker is set dynamically, and if I don't set any linker in my board manifest, PIO should choose a proper linker because `build_flags = -DNRF52_S132` is in the project ini file, right?
But it throws this error:  
```
make flash
platformio -f -c vim run --target upload
Processing my_env (platform: nordicnrf52; framework: arduino; board: myboard)
-------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/myboard.html
PLATFORM: Nordic nRF52 (6.0.0) > Biofeedback Central Node v1
HARDWARE: NRF52832 64MHz, 64KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic)
PACKAGES: 
 - framework-arduinonordicnrf5 1.700.201209 (7.0) 
 - tool-bossac-nordicnrf52 1.10901.201022 (1.9.1) 
 - tool-jlink 1.68801.0 (6.88.1) 
 - tool-openocd 2.1000.200630 (10.0) 
 - tool-sreccat 1.164.0 (1.64) 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
IndexError: list index out of range:
  File "/home/pooya/.local/pipx/venvs/platformio/lib/python3.8/site-packages/platformio/builder/main.py", line 170:
    env.SConscript("$BUILD_SCRIPT")
  File "/home/pooya/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/pooya/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/pooya/.platformio/platforms/nordicnrf52/builder/main.py", line 194:
    target_elf = env.BuildProgram()
  File "/home/pooya/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/home/pooya/.local/pipx/venvs/platformio/lib/python3.8/site-packages/platformio/builder/tools/platformio.py", line 61:
    env.ProcessProgramDeps()
  File "/home/pooya/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/home/pooya/.local/pipx/venvs/platformio/lib/python3.8/site-packages/platformio/builder/tools/platformio.py", line 127:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "/home/pooya/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/home/pooya/.local/pipx/venvs/platformio/lib/python3.8/site-packages/platformio/builder/tools/platformio.py", line 342:
    SConscript(env.GetFrameworkScript(f), exports="env")
  File "/home/pooya/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py", line 661:
    return method(*args, **kw)
  File "/home/pooya/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/pooya/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/pooya/.platformio/platforms/nordicnrf52/builder/frameworks/arduino.py", line 31:
    env.SConscript("arduino/nrf5.py")
  File "/home/pooya/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/pooya/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/pooya/.platformio/platforms/nordicnrf52/builder/frameworks/arduino/nrf5.py", line 162:
    mcu_family = board.get("build.arduino.ldscript", "").split("_")[1]
============================ [FAILED] Took 0.33 seconds ============================
Makefile:6: recipe for target 'upload' failed
make: *** [upload] Error 1
```

pvonmoradi, 2021-01-07:
In fact, doing a recursive search in `~/.platformio`, we see that only these linkers exist in the tree:
```
λ fd "nrf5*" -e "ld"
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/softdevice/s110/toolchain/armgcc/armgcc_s110_nrf51822_xxaa.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/softdevice/s110/toolchain/armgcc/armgcc_s110_nrf51822_xxab.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/softdevice/s110/toolchain/armgcc/armgcc_s110_nrf51822_xxac.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/softdevice/s130/toolchain/armgcc/armgcc_s130_nrf51822_xxaa.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/softdevice/s130/toolchain/armgcc/armgcc_s130_nrf51822_xxab.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/softdevice/s130/toolchain/armgcc/armgcc_s130_nrf51822_xxac.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/softdevice/s132/toolchain/armgcc/armgcc_s132_nrf52832_xxaa.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/gcc_nrf51_common.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf51422_xxaa.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf51422_xxab.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf51422_xxac.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf51801_xxab.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf51802_xxaa.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf51822_xxaa.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf51822_xxab.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf51822_xxac.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf51824_xxaa.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf51_common.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf51_xxaa.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf51_xxab.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf51_xxac.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf52805_xxaa.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf52810_xxaa.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf52811_xxaa.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf52820_xxaa.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf52832_xxaa.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf52832_xxab.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf52833_xxaa.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf52840_xxaa.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf52_common.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf52_xxaa.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf5340_xxaa_application.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf5340_xxaa_network.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf5x_common.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf9160_xxaa.ld
packages/framework-arduinonordicnrf5/cores/nRF5/SDK/components/toolchain/gcc/nrf_common.ld
```
This means that some boards like https://github.com/platformio/platform-nordicnrf52/blob/6654461aa15920c08f92373ea0bbd813310c94c6/boards/adafruit_feather_nrf52832.json#L4 would never get compiled as the specified linker does not exist.

Even when  choose 
```
        "arduino": {
            "ldscript": "nrf52832_s132_v6.ld"
        },
``` 
in `myboard.json`, PIO can't find it (but it exists in the tree)
```
Linking .pio/build/my_env/firmware.elf
/home/pooya/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: cannot open linker script file armgcc_s132_nrf52832_xxaa.ld: No such file or directory
collect2: error: ld returned 1 exit status
*** [.pio/build/my_env/firmware.elf] Error 1
```

valeros, 2021-01-08:
> If the linker is set dynamically, and if I don't set any linker in my board manifest, PIO should choose a proper linker because build_flags = -DNRF52_S132 is in the project ini file, right?

The linker script name in a board manifest is used in searching process for a new linker script specific for specified SoftDevice so it's mandatory.

> This means that some boards like platform-nordicnrf52/boards/adafruit_feather_nrf52832.json

That boards uses completely different Arduino core from Adafruit https://github.com/platformio/platform-nordicnrf52/blob/develop/builder/frameworks/arduino.py#L28

pvonmoradi, 2021-01-08:
@valeros There are two Arduino HAL support packages for nRF5s. 
[adafruit/Adafruit_nRF52_Arduino](https://github.com/adafruit/Adafruit_nRF52_Arduino) and [sandeepmistry/arduino-nRF5](https://github.com/sandeepmistry/arduino-nRF5) and it seems both are supported by PIO.  
How can I use the adafruit one since it has a newer softdevice?

valeros, 2021-01-11:
Just use one of the Adafruit board manifests, e.g. https://github.com/platformio/platform-nordicnrf52/blob/develop/boards/adafruit_feather_nrf52832.json

ZenanLin1999, 2024-05-22:
I met the same problems. I can upload the hex file without soft device (no BLE function) using jlink tools successfully.  However, after adding "build_flags =  -DNRF52_S132" command in "platformio.ini", my code is not working and still running the last code without soft device. This setting's build and flash is ok. Hopefully, someone can help me, really strange.