platformio/platform-nordicnrf52 #102: Debugging not started or takes ages

maxgerhardt, 2021-01-08:
Refer https://community.platformio.org/t/debugger-not-starting/18338.

When attempting to debug the `nrf52_dk` board from the platform-nordicnrf52, after using "PIO Debug" in VSCode, the time until the actual debugging server is started either seems to be never or extremely late.

With a `platformio.ini` of

```
[env:rak8211_nb]
platform = nordicnrf52
board = nrf52_dk
framework = zephyr
board_build.mcu = nrf52832

upload_protocol = custom
upload_command = $PROJECT_PACKAGES_DIR/tool-openocd/bin/openocd -f $PROJECT_PACKAGES_DIR/tool-openocd/scripts/board/nordic_nrf52_ftx232.cfg -c init -c "reset init" -c halt -c "program {$SOURCE} verify" -c reset -c exit
```

And starting debugging, which uses the automatic preconfigure J-Link, there is a significant delay in the order of half an hour until JLink output appears and the session attempts to start.

[image]

When using a custom debug server command, which was verified by the user to work beforehand, the debugging seems to nevers start at all. 

Additions to the `platformio.ini`

```
debug_tool = custom
debug_server = 
	$PROJECT_PACKAGES_DIR/tool-openocd/bin/openocd
	-s
	$PROJECT_PACKAGES_DIR/tool-openocd/scripts/
	-f 
	board/nordic_nrf52_ftx232.cfg
debug_port = localhost:3333
```

Since there is no debug output in the console it's impossible to tell what's going on. The task manager just shows the invocation of  `platformio debug -e rak8211_nb --interface gdb --interpreter=mi2 -q`. 

Is this reproducible for the devs? 

Is the long delay / never starting (or giving no failure error message) an issue in the core? 

------

Additional info (latest dev core):

```
>pio system info
--------------------------  --------------------------------------------------------------------------
PlatformIO Core             5.0.5a1
Python                      3.8.6-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\Max\.platformio
PlatformIO Core Executable  C:\Users\Max\AppData\Local\Programs\Python\Python38\Scripts\platformio.exe
Python Executable           c:\users\max\appdata\local\programs\python\python38\python.exe
Global Libraries            0
Development Platforms       14
Tools & Toolchains          97
--------------------------  --------------------------------------------------------------------------
```

Montecri, 2021-04-06:
Hello,

Same here. Can never get past the last line:

[image]

`platformio.ini`

[env:genericSTM32F103CB]
platform = ststm32
board = bluepill_f103c8_128k
framework = arduino
upload_flags = -c set CPUTAPID 0x2ba01477
debug_server = -c "set CPUTAPID 0x2ba01477"

`pio system info`

> pio system info
--------------------------  ------------------------------------------
PlatformIO Core             5.1.1
Python                      3.9.2-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\.platformio
PlatformIO Core Executable  C:\.platformio\penv\Scripts\platformio.exe
Python Executable           c:\.platformio\penv\scripts\python.exe    
Global Libraries            0
Development Platforms       4
Tools & Toolchains          59
--------------------------  ------------------------------------------

Ran out of suggestions already.

Thanks,

- Cristiano

maxgerhardt, 2021-04-06:
But this is a different thing because in 

```
 debug_server = -c "set CPUTAPID 0x2ba01477"
```
you set a directive to set the executable that is supposed to start the debug server to ` -c "set CPUTAPID 0x2ba01477"`. That is invalid, it must be a path to OpenOCD that is capable of starting the GDBserver. Something liike 

```ini
debug_server = C:\.platformio\packages\tool-openocd\bin\openocd -s C:\.platformio\packages\tool-openocd\bin\scripts -f interface/stlink.cfg -c "transport select hla_swd" -c "set CPUTAPID 0x2ba01477" -f target/stm32f1x.cfg 
```

If you look into the "Output" or "Debug Console" tab in VSCode it should also tell you that the executable is not found.

Also are you sure you need `-c set CPUTAPID 0x2ba01477` at all, do you have a fake STM32/CS32 chip on the board?

Montecri, 2021-04-06:
Hi,

Thank you for the prompt reply.

Yes, tried the full path in the past and tried again just now. Does nothing. The entire outcome of my debug console is on the screenshot above. Tried enabling verbose but all I get is a few more lines informing segment sizes, etc, nothing about why it is not progressing.

Yes, got a fake blue pill, for sure.

Thanks!

maxgerhardt, 2021-04-06:
My bad, I should have checked before. When I use that `platformio.ini`, PIO immediately gives me 

[image]

in the Debug Console tab. So if you don't have that, you should make sure that your versions are all up to date. E.g. with a [CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli), you can execute `pio upgrade --dev` and `pio platform update ststm32` to update core and platform. 

The lines need to be indeed newline separated. If I connect an ST-Link V2 to my bluepill and use

```
debug_server = 
    C:\Users\Max\.platformio\packages\tool-openocd\bin\openocd 
    -s 
    C:\Users\Max\.platformio\packages\tool-openocd\bin\scripts 
    -f 
    interface/stlink.cfg
    -c "transport select hla_swd"
    -f
    target/stm32f1x.cfg 
    -c "reset_config none"
```
instead of the old `debug_server`, it works.

[image]

So you should maybe try

```
debug_server = 
    C:\.platformio\packages\tool-openocd\bin\openocd 
    -s 
    C:\.platformio\packages\tool-openocd\bin\scripts 
    -f 
    interface/stlink.cfg
    -c "transport select hla_swd"
    -c "set CPUTAPID 0x2ba01477"
    -f
    target/stm32f1x.cfg 
    -c "reset_config none"
```

if the paths are correct.

Montecri, 2021-04-06:
Hi Max,

Replaced debug_server with the newline-separated you supplied and it indeed works! No further changes needed.

Many thanks!

fnork, 2021-04-26:
Hi,

I'm having what sounds like the same problem with the nRF52833-DK.

```
$ pio system info
--------------------------  -------------------------------------------
PlatformIO Core             5.2.0a6
Python                      3.8.4-final.0
System Type                 darwin_x86_64
Platform                    macOS-10.16
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /Users/erik/.platformio
PlatformIO Core Executable  /Users/erik/.platformio/penv/bin/platformio
Python Executable           /Users/erik/.platformio/penv/bin/python
Global Libraries            0
Development Platforms       3
Tools & Toolchains          50
--------------------------  -------------------------------------------
```

Board file `nrf52833_dk.json`
```
{
  "build": {
    "arduino":{
      "ldscript": "nrf52_xxaa.ld"
    },
    "core": "nRF5",
    "cpu": "cortex-m4",
    "extra_flags": "-DARDUINO_NRF52_DK",
    "f_cpu": "64000000L",
    "mcu": "nrf52833",
    "variant": "nRF52DK",
    "zephyr": {
       "variant": "nrf52833dk_nrf52833"
    }
  },
  "connectivity": [
    "bluetooth"
  ],
  "debug": {
    "default_tools": [
      "jlink"
    ],
    "jlink_device": "nRF52833_xxAA",
    "onboard_tools": [
      "cmsis-dap",
      "jlink"
    ],
    "svd_path": "nrf52833.svd"
  },
  "frameworks": [
    "arduino",
    "mbed",
    "zephyr"
  ],
  "name": "Nordic nRF52833-DK",
  "upload": {
    "maximum_ram_size": 131072,
    "maximum_size": 524288,
    "protocol": "jlink",
    "protocols": [
      "jlink",
      "nrfjprog",
      "stlink",
      "blackmagic",
      "cmsis-dap",
      "mbed"
    ]
  },
  "url": "https://www.nordicsemi.com/Software-and-tools/Development-Kits/nRF52833-DK",
  "vendor": "Nordic"
}
```

`platformio.ini`:
```
[env:nrf52833_dk]
platform = nordicnrf52
board = nrf52833_dk
framework = zephyr
monitor_speed = 115200

upload_protocol= jlink

debug_tool = jlink
debug_port = localhost:2331
debug_server =
    /Applications/SEGGER/JLink_V700a/JLinkGDBServerCLExe
    -singlerun
    -if
    SWD
    -select
    USB
    -port
    2331
    -device
    nRF52833_xxAA
```

I have tried with the default settings for `debug_{tool,port,server}` but encounter the same problem.

When running debug I get the same output:
```
inking .pio/build/nrf52833_dk/firmware-pre.elf
Generating ISR list .pio/build/nrf52833_dk/zephyr/isrList.bin
Generating ISR table .pio/build/nrf52833_dk/zephyr/isr_tables.c
Compiling .pio/build/nrf52833_dk/zephyr_final/zephyr/isr_tables.c.o
Linking .pio/build/nrf52833_dk/firmware.elf
Checking size .pio/build/nrf52833_dk/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  17.6% (used 23022 bytes from 131072 bytes)
Flash: [===       ]  28.3% (used 148280 bytes from 524288 bytes)
Building .pio/build/nrf52833_dk/firmware.hex
======================================================================================== [SUCCESS] Took 2.33 seconds ========================================================================================

Terminal will be reused by tasks, press any key to close it.
```

When running `JLinkGDBServerCLExe` in a shell with the same flags as the `platformio.ini` file I can attach with GDB without any problems using `(gdb) target remote localhost:2331`.

Let me know if there's something obvious I have missed or if there's any additional information I can supply.

Thank you.

maxgerhardt, 2021-04-28:
Well first of all for it to respect the `debug_server` you must set `debug_tool = custom` but you set `debug_tool = jlink`. Try and change that. However, also the built in JLink tool should be sufficient.. 

Then, what happens when you [open a CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and run 

```
pio upgrade --dev
pio debug --interface=gdb -x .pioinit
```
with the changes above?

What does the last command output if you remove all `debug_*` options from the `platformio.ini`?

fnork, 2021-04-28:
Thank you for your reply.

Tried both `debug_tool = custom` and `debug_tool = jlink`. Happened to copy the file while it was on the `jlink` option, sorry about that, I should have clarified.

Running `$ pio upgrade --dev` resulted in:
```
Please wait while upgrading PlatformIO ...
PlatformIO has been successfully upgraded to 5.2.0a6
Release notes: https://docs.platformio.org/en/latest/history.html
```
Since I was already on `5.2.0a6` (ran `pio upgrade --dev` just before posting) I assume that's a NOOP?

I cleared all `debug_*` options in `platfomio.ini` and ran `pio debug --interface=gdb -x .pioinit`. It built, started the GDB server and attached to it:
```
SEGGER J-Link GDB Server V6.88b Command Line Version

JLinkARM.dll V6.88b (DLL compiled Nov 27 2020 15:44:17)

Command line: -singlerun -if SWD -select USB -device nRF52833_xxAA -port 2331
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     2331
SWO raw output listening port: 2332
Terminal I/O port:             2333
Accept remote connection:      yes
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               on
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 nRF52833_xxAA
Target interface:              SWD
Target interface speed:        4000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link OB-SAM3U128-V2-NordicSemi compiled Feb  2 2021 16:47:20
Hardware: V1.00
S/N: 685690177
Feature(s): RDI, FlashBP, FlashDL, JFlash, GDB
Checking target voltage...
Target voltage: 3.30 V
Listening on TCP/IP port 2331
Connecting to target...
Connected to target
Waiting for GDB connection...Reading symbols from /Users/erik/code/tmp/nrf52833-dk-test/.pio/build/nrf52833_dk/firmware.elf...
PlatformIO Unified Debugger -> http://bit.ly/pio-debug
PlatformIO: debug_tool = jlink
PlatformIO: Initializing remote target...
Connected to 127.0.0.1
Reading all registers
Read 4 bytes @ address 0x00004C5A (Data = 0xF3BFB662)
Read 2 bytes @ address 0x00004C5A (Data = 0xB662)
arch_cpu_idle ()
    at /Users/erik/.platformio/packages/framework-zephyr/arch/arm/core/aarch32/cpu_idle.S:107
107             cpsie   i
Reading 64 bytes @ address 0x00004C40
Received monitor command: clrbp
Received monitor command: speed auto
Select auto target interface speed (1000 kHz)
Select auto target interface speed (1000 kHz)
Received monitor command: reset
Resetting target
[... 136 lines cut ...]
.ramfunc                            0   536870912
datas                            1767   536870912
devices                           132   536872680
k_mem_slab_area                    84   536872812
k_mutex_area                       20   536872896
k_sem_area                         96   536872916
k_queue_area                       48   536873012
_net_buf_pool_area                240   536873060
bss                              9668   536873304
noinit                          11435   536882976
.comment                          117           0
.debug_aranges                  16032           0
.debug_info                   1098787           0
.debug_abbrev                  134982           0
.debug_line                    461597           0
.debug_frame                    46724           0
.debug_str                     114232           0
.debug_loc                     487215           0
.debug_ranges                   97656           0
.ARM.attributes                    49           0
Total                         2629729
======================================================================================== [SUCCESS] Took 2.04 seconds ========================================================================================

Terminal will be reused by tasks, press any key to close it.
```

maxgerhardt, 2021-04-29:
>Trying to start debugging from inside VS Code (with or without skip pre-debug) still builds and then nothing happens.

All instances of it working via the CLI but not in VSCode have been due to having incompatible extensions installed in VSCode. Does this happen too if only C/C++ by Microsoft and PlatformIO are installed?

fnork, 2021-04-29:
Thank you! 

I disabled all extensions except PlatfomIO and the MS C/C++ one. And now it works as expected. I'll add the other extensions one by one to see which one/ones are causing the problem and report my findings in this thread.

fnork, 2021-04-29:
The extention that caused the issue was [Cloud Code by Google Cloud](https://marketplace.visualstudio.com/items?itemName=GoogleCloudTools.cloudcode), in this case v1.11.0.

Not the culprit I expected tbh.

Verified that a valid workaround is disabling that extension for the the workspace. So anyone that needs both can still do so.

maxgerhardt, 2021-04-29:
I see, then it's the same issue as already noted by another user and tracked in https://github.com/platformio/platformio-vscode-ide/issues/2481.