platformio/platform-nordicnrf52 #105: Add nRF5 SDK support

maxgerhardt, 2021-02-14:
Equivalent to https://github.com/platformio/platform-nordicnrf51/issues/24 from the nRF51 platform. 

Having the native vendor nRF5 SDK framework supported would be great, many users have asked for that over and over again of the last time.

pingdan32, 2021-06-28:
Any plans?

TauqirH, 2021-12-20:
No more NRF SDK , society has progressed past the need for NRF SDK.

tomaszduda23, 2024-02-01:
Some project might be build that way.
```
[env:nrf52_dk]
platform = platformio/nordicnrf52@10.3.0
framework = zephyr
board = nrf52_dk
platform_packages =
    platformio/framework-zephyr@https://github.com/tomaszduda23/framework-sdk-nrf
    platformio/toolchain-gccarmnoneeabi@https://github.com/tomaszduda23/toolchain-sdk-ng

```