platformio/platform-nordicnrf52 #112: Add support for ublox EVK-BMD-350 Evaluation Board

anthonytw, 2021-03-20:
https://www.u-blox.com/en/product/evk-bmd-3035#tab-documentation-resources

valeros, 2021-03-22:
Hi @anthonytw ! What board do you plan to use with this board?

cujomalainey, 2021-03-22:
@valeros I think you mean "framework"?

valeros, 2021-03-22:
Yeap, sorry, I meant framework.

anthonytw, 2021-03-22:
I actually realized this is pretty similar to the ublox NINA board and made a copy of that. I'll either use mbed or zephyr, so the modification I made was to add zephyr to the supported list along with the zephyr variant specifier I found in nrf52dk, which may or may not have been the right thing to do. :-)

That said, I think this eval board should technically be compatible with Arduino (similar to nano 33 ble), mbed, and zephyr. I haven't tried with Arduino though.