platformio/platform-nordicnrf52 #114: Add board ebyte E73-TBA

rikki78, 2021-04-23:
Could the development board ebyte E73-TBA be added to the available boards?
When selecting the generic board it works, though the uart is on the wrong pins.
https://www.ebyte.com/en/downpdf.aspx?id=888

valeros, 2021-04-23:
Hi @rikki78 ! That's the reason why we don't add boards that are not officially supported. You'd better submit a PR with a proper pin definition to the framework which you want to use and then ping us so we add a new board manifest when it's merged and released.

kaessert, 2024-08-08:
For anyone still looking, here's what you need: https://github.com/nrfconnect/sdk-zephyr/tree/98194577de1a3850de578da2a26c5f355e984090/boards/ebyte/e73_tbb