platformio/platform-nordicnrf52 #116: Adafruit nRF52 Arduino BSP 0.22.1 is out.

ogatatsu, 2021-05-28:
https://github.com/adafruit/Adafruit_nRF52_Arduino/releases/tag/0.22.1

maxgerhardt, 2021-07-11:
+1 to this. @valeros? 

People also running into compilation problems when using this old core with the newest library version, e.g. [here](https://community.platformio.org/t/tinyusb-featherwing-nrf52840-incompatibility/22434).

closed 2021-07-19 by valeros

valeros, 2021-07-19:
Updated to the latest v0.24, please retest with the upstream version of the platform:
```ini
[env:adafruit_cplaynrf52840]
platform = https://github.com/platformio/platform-nordicnrf52.git
framework = arduino
board = adafruit_cplaynrf52840
```

AndreasLamparter, 2021-07-20:
I get some linker errors

```
.pio\build\CycleApp\lib12c\libAdafruit nRFCrypto.a(Adafruit_nRFCrypto.cpp.o): In function `Adafruit_nRFCrypto::begin()':
Adafruit_nRFCrypto.cpp:(.text._ZN18Adafruit_nRFCrypto5beginEv+0x14): undefined reference to `SaSi_LibInit'
.pio\build\CycleApp\lib12c\libAdafruit nRFCrypto.a(nRFCrypto_ECC.cpp.o): In function `nRFCrypto_ECC::genKeyPair(nRFCrypto_ECC_PrivateKey&, nRFCrypto_ECC_PublicKey&)':
nRFCrypto_ECC.cpp:(.text._ZN13nRFCrypto_ECC10genKeyPairER24nRFCrypto_ECC_PrivateKeyR23nRFCrypto_ECC_PublicKey+0x3e): undefined reference to `CRYS_ECPKI_GenKeyPair'
nRFCrypto_ECC.cpp:(.text._ZN13nRFCrypto_ECC10genKeyPairER24nRFCrypto_ECC_PrivateKeyR23nRFCrypto_ECC_PublicKey+0x74): undefined reference to `CRYS_RND_GenerateVector'
.pio\build\CycleApp\lib12c\libAdafruit nRFCrypto.a(nRFCrypto_ECC.cpp.o): In function `nRFCrypto_ECC::SVDP_DH(nRFCrypto_ECC_PrivateKey&, nRFCrypto_ECC_PublicKey&, unsigned char*, unsigned long)':   
nRFCrypto_ECC.cpp:(.text._ZN13nRFCrypto_ECC7SVDP_DHER24nRFCrypto_ECC_PrivateKeyR23nRFCrypto_ECC_PublicKeyPhm+0x2a): undefined reference to `CRYS_ECDH_SVDP_DH'
.pio\build\CycleApp\lib12c\libAdafruit nRFCrypto.a(nRFCrypto_ECC_PrivateKey.cpp.o): In function `nRFCrypto_ECC_PrivateKey::begin(CRYS_ECPKI_DomainID_t)':
nRFCrypto_ECC_PrivateKey.cpp:(.text._ZN24nRFCrypto_ECC_PrivateKey5beginE21CRYS_ECPKI_DomainID_t+0x6): undefined reference to `CRYS_ECPKI_GetEcDomain'
.pio\build\CycleApp\lib12c\libAdafruit nRFCrypto.a(nRFCrypto_ECC_PublicKey.cpp.o): In function `nRFCrypto_ECC_PublicKey::begin(CRYS_ECPKI_DomainID_t)':
nRFCrypto_ECC_PublicKey.cpp:(.text._ZN23nRFCrypto_ECC_PublicKey5beginE21CRYS_ECPKI_DomainID_t+0x6): undefined reference to `CRYS_ECPKI_GetEcDomain'
.pio\build\CycleApp\lib12c\libAdafruit nRFCrypto.a(nRFCrypto_ECC_PublicKey.cpp.o): In function `nRFCrypto_ECC_PublicKey::toRaw(unsigned char*, unsigned long)':
nRFCrypto_ECC_PublicKey.cpp:(.text._ZN23nRFCrypto_ECC_PublicKey5toRawEPhm+0x10): undefined reference to `CRYS_ECPKI_ExportPublKey'
.pio\build\CycleApp\lib12c\libAdafruit nRFCrypto.a(nRFCrypto_ECC_PublicKey.cpp.o): In function `nRFCrypto_ECC_PublicKey::fromRaw(unsigned char*, unsigned long)':
nRFCrypto_ECC_PublicKey.cpp:(.text._ZN23nRFCrypto_ECC_PublicKey7fromRawEPhm+0x2c): undefined reference to `_DX_ECPKI_BuildPublKey'
.pio\build\CycleApp\lib12c\libAdafruit nRFCrypto.a(nRFCrypto_Random.cpp.o): In function `nRFCrypto_Random::begin()':
nRFCrypto_Random.cpp:(.text._ZN16nRFCrypto_Random5beginEv+0x2e): undefined reference to `CRYS_RndInit'
```

My configuration:
```
platform = https://github.com/platformio/platform-nordicnrf52.git
board = adafruit_feather_nrf52840_sense
framework = arduino
```

valeros, 2021-07-20:
HI @AndreasLamparter ! Could you please attach here the entire log?

AndreasLamparter, 2021-07-20:
The full log:

```
Processing CycleApp (platform: https://github.com/platformio/platform-nordicnrf52.git; board: adafruit_feather_nrf52840_sense; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Tool Manager: Installing platformio/toolchain-gccarmnoneeabi @ >=1.60301.0,<1.80000.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: toolchain-gccarmnoneeabi @ 1.70201.0 has been installed!
Tool Manager: Installing platformio/framework-arduinoadafruitnrf52 @ ~1.2400.0
Unpacking  [####################################]  100%
Tool Manager: framework-arduinoadafruitnrf52 @ 1.2400.210719 has been installed!
Tool Manager: Installing platformio/framework-cmsis @ ~2.50700.0
Unpacking  [####################################]  100%
Tool Manager: framework-cmsis @ 2.50700.210515 has been installed!
Tool Manager: Installing platformio/tool-sreccat @ ~1.164.0
Tool Manager: tool-sreccat @ 1.164.0 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40100.2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-scons @ 4.40100.2 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/adafruit_feather_nrf52840_sense.html
PLATFORM: Nordic nRF52 (8.1.0+sha.0afb471) > Adafruit Feather Bluefruit Sense
HARDWARE: NRF52840 64MHz, 243KB RAM, 796KB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoadafruitnrf52 1.2400.210719 (24.0)
 - framework-cmsis 2.50700.210515 (5.7.0)
 - tool-sreccat 1.164.0 (1.64)
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Adafruit BusIO> 1.7.3
|   |-- <Wire> 1.0
|   |   |-- <Adafruit TinyUSB Library> 1.3.0
|   |-- <SPI> 1.0
|   |   |-- <Adafruit TinyUSB Library> 1.3.0
|-- <Adafruit Unified Sensor> 1.1.4
|-- <Adafruit LSM6DS> 4.2.0
|   |-- <Wire> 1.0
|   |   |-- <Adafruit TinyUSB Library> 1.3.0
|   |-- <Adafruit BusIO> 1.7.3
|   |   |-- <Wire> 1.0
|   |   |   |-- <Adafruit TinyUSB Library> 1.3.0
|   |   |-- <SPI> 1.0
|   |   |   |-- <Adafruit TinyUSB Library> 1.3.0
|   |-- <SPI> 1.0
|   |   |-- <Adafruit TinyUSB Library> 1.3.0
|   |-- <Adafruit Unified Sensor> 1.1.4
|-- <HX711> 0.7.4
|-- <Adafruit Bluefruit nRF52 Libraries>
|   |-- <Adafruit nRFCrypto> 0.0.5
|   |   |-- <Adafruit TinyUSB Library> 1.3.0
|   |-- <Adafruit TinyUSB Library> 1.3.0
|   |-- <Adafruit Little File System Libraries> 0.11.0
|   |   |-- <Adafruit TinyUSB Library> 1.3.0
|   |-- <Adafruit Internal File System on Bluefruit nRF52> 0.11.0
|   |   |-- <Adafruit Little File System Libraries> 0.11.0
|   |   |   |-- <Adafruit TinyUSB Library> 1.3.0
Building in release mode
Linking .pio\build\CycleApp\firmware.elf
.pio\build\CycleApp\lib12c\libAdafruit nRFCrypto.a(Adafruit_nRFCrypto.cpp.o): In function `Adafruit_nRFCrypto::begin()':
Adafruit_nRFCrypto.cpp:(.text._ZN18Adafruit_nRFCrypto5beginEv+0x14): undefined reference to `SaSi_LibInit'
.pio\build\CycleApp\lib12c\libAdafruit nRFCrypto.a(nRFCrypto_ECC.cpp.o): In function `nRFCrypto_ECC::genKeyPair(nRFCrypto_ECC_PrivateKey&, nRFCrypto_ECC_PublicKey&)':
nRFCrypto_ECC.cpp:(.text._ZN13nRFCrypto_ECC10genKeyPairER24nRFCrypto_ECC_PrivateKeyR23nRFCrypto_ECC_PublicKey+0x3e): undefined reference to `CRYS_ECPKI_GenKeyPair'
nRFCrypto_ECC.cpp:(.text._ZN13nRFCrypto_ECC10genKeyPairER24nRFCrypto_ECC_PrivateKeyR23nRFCrypto_ECC_PublicKey+0x74): undefined reference to `CRYS_RND_GenerateVector'
.pio\build\CycleApp\lib12c\libAdafruit nRFCrypto.a(nRFCrypto_ECC.cpp.o): In function `nRFCrypto_ECC::SVDP_DH(nRFCrypto_ECC_PrivateKey&, nRFCrypto_ECC_PublicKey&, unsigned char*, unsigned long)':   
nRFCrypto_ECC.cpp:(.text._ZN13nRFCrypto_ECC7SVDP_DHER24nRFCrypto_ECC_PrivateKeyR23nRFCrypto_ECC_PublicKeyPhm+0x2a): undefined reference to `CRYS_ECDH_SVDP_DH'
.pio\build\CycleApp\lib12c\libAdafruit nRFCrypto.a(nRFCrypto_ECC_PrivateKey.cpp.o): In function `nRFCrypto_ECC_PrivateKey::begin(CRYS_ECPKI_DomainID_t)':
nRFCrypto_ECC_PrivateKey.cpp:(.text._ZN24nRFCrypto_ECC_PrivateKey5beginE21CRYS_ECPKI_DomainID_t+0x6): undefined reference to `CRYS_ECPKI_GetEcDomain'
.pio\build\CycleApp\lib12c\libAdafruit nRFCrypto.a(nRFCrypto_ECC_PublicKey.cpp.o): In function `nRFCrypto_ECC_PublicKey::begin(CRYS_ECPKI_DomainID_t)':
nRFCrypto_ECC_PublicKey.cpp:(.text._ZN23nRFCrypto_ECC_PublicKey5beginE21CRYS_ECPKI_DomainID_t+0x6): undefined reference to `CRYS_ECPKI_GetEcDomain'
.pio\build\CycleApp\lib12c\libAdafruit nRFCrypto.a(nRFCrypto_ECC_PublicKey.cpp.o): In function `nRFCrypto_ECC_PublicKey::toRaw(unsigned char*, unsigned long)':
nRFCrypto_ECC_PublicKey.cpp:(.text._ZN23nRFCrypto_ECC_PublicKey5toRawEPhm+0x10): undefined reference to `CRYS_ECPKI_ExportPublKey'
.pio\build\CycleApp\lib12c\libAdafruit nRFCrypto.a(nRFCrypto_ECC_PublicKey.cpp.o): In function `nRFCrypto_ECC_PublicKey::fromRaw(unsigned char*, unsigned long)':
nRFCrypto_ECC_PublicKey.cpp:(.text._ZN23nRFCrypto_ECC_PublicKey7fromRawEPhm+0x2c): undefined reference to `_DX_ECPKI_BuildPublKey'
.pio\build\CycleApp\lib12c\libAdafruit nRFCrypto.a(nRFCrypto_Random.cpp.o): In function `nRFCrypto_Random::begin()':
nRFCrypto_Random.cpp:(.text._ZN16nRFCrypto_Random5beginEv+0x2e): undefined reference to `CRYS_RndInit'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\CycleApp\firmware.elf] Error 1
```

valeros, 2021-07-20:
Strange, but I cannot reproduce the issue. Could you please also try to run your project in verbose mode, so we can examine the linker command more carefully.

AndreasLamparter, 2021-07-20:
It compiles and links now without errors!

First time, i cleaned my project with PlatformIO clean. This does not work correctly.

Now i cleaned the project by hand -> it links wihtout error.

Sorry ...

commit 57aa33e8 (2021-07-19): Update Arduino Adafruit core to v0.24 // Resolve #116