platformio/platform-nordicnrf52 #119: Add notes for USB Serial

maxgerhardt, 2021-08-08:
Since updating to the new Adafruit core version 0.24, users experience that a standard sketch of

```cpp
#include <Arduino.h>

void setup() {
   Serial.begin(115200);
}
void loop() {
}
```
fails to compile

```
.pio\build\adafruit_feather_nrf52840_sense\src\main.cpp.o: In function `setup':
main.cpp:(.text.setup+0xc): undefined reference to `Serial'
main.cpp:(.text.setup+0x6): undefined reference to `Adafruit_USBD_CDC::begin(unsigned long)'
```

Which is due to the core being refactored. The Adafruit TinyUSB header now has to be explicitly included to get the TinyUSB library to compile and provide the `Serial` object.

```cpp
#include <Arduino.h>
#include "Adafruit_TinyUSB.h"

void setup() {
   Serial.begin(115200);
}
void loop() {
}
```

However, the project fails to execute correctly -- no USB serial is opened. This is due to the TinyUSB not having a `library.json` where it specifies `"libArchive": false`. The library implements USB interrupt vectors and will not link correctly in PlatformIO as an archive (but magically in the Arduino IDE does). So adding

```ini
lib_archive = no
```

to the `platformio.ini` is further needed as a workaround to get things working.

See [community topic](https://community.platformio.org/t/adafruit-feather-nrf52840-sense-upload-issue/22845/).

The TinyUSB library building should be fixed (either by adapting the Arduino builder code to automatically link *not* link as archive but object files) and/or a note should be added in the documentation regarding this new behavior / needed flags.

valeros, 2021-08-10:
Hi @maxgerhardt ! Thanks, I've updated the `framework-arduinoadafruitnrf52` package so now the `TinyUSB` should be linked in form of object files. As for the undefined references, have you tried it in Arduino IDE? Does it work as expected there?

maxgerhardt, 2021-08-10:
The undefined reference appears in the Arduino IDE as well with the same 0.24 core version.

[image]

and the error goes away in the same way, by `#include "Adafruit_TinyUSB.h"`

And most interestingly the Ardui IDE links TinyUSB by object files to.. Although I don't see how the `library.properties` file directed this o_O

>"C:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\arm-none-eabi-gcc\\9-2019q4/bin/arm-none-eabi-gcc"  [...] -o "C:\\Users\\Max\\AppData\\Local\\Temp\\arduino_build_268909/sketch_aug10a.ino.elf" "C:\\Users\\Max\\AppData\\Local\\Temp\\arduino_build_268909\\libraries\\Adafruit_TinyUSB_Arduino\\tusb.c.o" "C:\\Users\\Max\\AppData\\Local\\Temp\\arduino_build_268909\\libraries\\Adafruit_TinyUSB_Arduino\\arduino\\Adafruit_TinyUSB_API.cpp.o"  [..]

So I'm taking it back that the Arduino IDE links as archive, it also links as object files.

valeros, 2021-08-10:
So I guess that's rather a framework issue, isn't it? Does it make sense to file an issue in their repo?

maxgerhardt, 2021-08-10:
They know about this at https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/653 and the breakage is intentional.

Other frameworks using Adafruit_TinyUSB have implemented workarounds by including the header file in some important core files so that library inclusion is auto-triggered, like https://github.com/earlephilhower/arduino-pico/issues/167

valeros, 2021-08-10:
The latest package contains `library.json` with `"libArchive": false`, so believe this issue is resolved.

closed 2021-08-10 by valeros

Rob192, 2022-07-28:
Hello ! FYI the fix did not work on my side. I still had to do the fix and add `lib_archive = no` to the config file to be able to successfully upload my code. I have installed platformio two days ago and the Arduino IDE is able to upload code as well with no workaround on my setup.

diplodocuslongus, 2022-09-07:
Worked for me with PlatformIO Core version 6.1.4, no need for `lib_archive = no`, only needed to include `#include "Adafruit_TinyUSB.h"`.