platformio/platform-nordicnrf52 #120: Stand alone Hex file - Adafruit NRF52 832

wb8wka, 2021-08-09:
Hello

I'd like to generate a one-piece HEX file I can burn in one pass.  That would consist of the Adafruit bootloader/softdevice and what I presume is the DFU hex file I see I can generate. 

Any thoughts?

valeros, 2021-08-09:
Hi @wb8wka ! I'm not sure what tools you're going to use to merge that hex files, but you can attach any additional steps you want using the [`Post Actions`](https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#before-pre-and-after-post-actions) feature. Alternatively, you can achieve similar results if you handcraft a custom [upload command](https://docs.platformio.org/en/latest/projectconf/section_env_upload.html#upload-command) that will burn several hex files in one pass.

wb8wka, 2021-08-10:
Thanks.  Let me give a bit more clarity.

I'm looking for a standalone HEX file to hand a vendor to program blank chips.  So I need to combine the bootloader file with my program file.   They won't be using platformIO, however I used it to develop the code.

I was able to "copy" my board and that worked, but there has to be a more elegant way

valeros, 2021-08-10:
I guess the best way is to attach a post action in an extra script. Again, I don't know what merge tool you prefer, but if you have the Nordic CLI tools installed in your system, you can set up something like this:

```ini
[env:adafruit_feather_nrf52832]
platform = nordicnrf52
framework = arduino
board = adafruit_feather_nrf52832
extra_scripts = 
    generate_production_image.py
```
`generate_production_image.py`:
```python
import os

Import("env")

def generate_production_binary(source, target, env):
    firmware_hex = target[0]
    bootloader_hex = env.get("DFUBOOTHEX", "")
    production_hex = os.path.join("$BUILD_DIR", "production.hex")
    merge_tool = "c:/Program Files (x86)/Nordic Semiconductor/nrf-command-line-tools/bin/mergehex"
    
    if not os.path.isfile(env.subst(production_hex)):
        assert os.path.isfile(bootloader_hex), "Missing bootloader image!"
        env.Execute(
            env.VerboseAction(
                f'"{merge_tool}" --merge "{bootloader_hex}" "{firmware_hex}" --output "{production_hex}"',
                f"Generating production image {production_hex}"
            )
        )
    
env.AddPostAction("$BUILD_DIR/${PROGNAME}.hex", generate_production_binary)

```

wb8wka, 2021-08-12:
Thank you @valeros that worked well, however it's staying in the bootloader.  I seem to recall it needed a signature or checksum to pass the bootloader test.

wb8wka, 2021-08-12:
OK, success if I go in with nrfprog --memwr 0x7F000 --val 0x01 after I burn production.hex    That tricks the bootloader.  Still trying to figure it out with one file, but here is that discussion

https://forums.adafruit.com/viewtopic.php?f=57&t=167184&p=819209&hilit=0x7f000#p819209

wb8wka, 2021-08-12:
It looks like firmware_signature.bin , which appears to be 32 bytes, should be merged in at 0x7F000.  It begins with 0x01, at least the one that generated on my machine. 

A bit more discussion

https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/133

valeros, 2021-08-12:
It seems that the `pynrfbintool` doesn't work with hex files, so it might be a good idea to use more advanced tools for merging, for example SRecord. I don't have much time to implement your case, but here another workaround that uses `srec_cat`:
```python
import os

Import("env")

def generate_production_binary(source, target, env):
    signature_bin = target[0]
    firmware_bin = "$BUILD_DIR/${PROGNAME}.bin"
    bootloader_hex = env.get("DFUBOOTHEX", "")
    production_image = os.path.join("$BUILD_DIR", "production.hex")
    merge_tool = os.path.join(
        env.PioPlatform().get_package_dir("tool-sreccat") or "", "srec_cat"
    )

    if not os.path.isfile(env.subst(production_image)):
        assert os.path.isfile(bootloader_hex), "Missing bootloader image!"
        env.Execute(
            env.VerboseAction(
                f'"{merge_tool}" "{bootloader_hex}" -intel "{signature_bin}" -Binary -offset 0x7F000 "{firmware_bin}" -Binary -offset 0x26000 -o "{production_image}" -intel',
                f"Generating production image {production_image}",
            )
        )

env.AddPostAction("$BUILD_DIR/${PROGNAME}_signature.bin", generate_production_binary)
```
I'm not sure about offsets, so please double the if that are correct values.

Other useful links:
- Documentation for [Advanced Scripting](https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html) in PlatformIO
- The [Sign Builder](https://github.com/platformio/platform-nordicnrf52/blob/develop/builder/main.py#L148-L172) used in this platform to sign individual binaries
- Documentation for [SRecord](http://srecord.sourceforge.net/man/man1/srec_cat.html) 

I'm closing this issue because this repository is not the right place for such discussions, you can ask general questions in our community https://community.platformio.org/

closed 2021-08-12 by valeros

wb8wka, 2021-08-13:
Understood but thank you for the help.   FYI, your code above worked perfectly.