platformio/platform-nordicnrf52 #125: Could not find the package with 'platformio/tool-dtc @ ~1.4.7' requirements for your system 'linux_armv7l'

lexologe, 2021-10-01:
Hello,

I am trying to run pio test for a Zephyr based project on a Raspberry Pi. The package tool-dtc cannot be downloaded. 

```
$ pio test -v
Collected 20 items

Processing array_utils in nrf52_dk environment
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Building...
Tool Manager: Installing platformio/tool-dtc @ ~1.4.7
Error: Could not find the package with 'platformio/tool-dtc @ ~1.4.7' requirements for your system 'linux_armv7l'
```

It seems that the package is not available for ARM based system. If possible I would like to compile and assemble the tool-dtc package by myself. I tried to find out how it is built but I can’t find a repo where the sources are located. Can someone point me to some help about how to do it?

Thanks,
Alex

maxgerhardt, 2021-10-01:
Per https://api.registry.platformio.org/v3/packages/platformio/tool/tool-dtc one can download the versions for the other OSes as a reference, e.g., Linux x86_64. 

[image]

The `package.json` also reveals

```
  "repository": {
    "type": "git",
    "url": "https://git.kernel.org/pub/scm/utils/dtc/dtc.git"
  }
```

maxgerhardt, 2021-10-01:
I was able to get it to compile using a Raspberry Pi, but it also needed to specify a newer compiler (the exact needed version wasn't available) and `tool-gperf` package besides the `tool-dtc` package. I compiled that from https://ftp.gnu.org/pub/gnu/gperf/gperf-3.0.4.tar.gz.

[tool-dtc-linux_armv6l-1.4.7.tar.gz](https://github.com/platformio/platform-nordicnrf52/files/7267888/tool-dtc-linux_armv6l-1.4.7.tar.gz)
[tool-gperf-linux_armv6l-3.0.4.tar.gz](https://github.com/platformio/platform-nordicnrf52/files/7267889/tool-gperf-linux_armv6l-3.0.4.tar.gz)

With those I finally could compile zephyr-blink for the nrf52_dk. 

`platformio.ini` (adapt paths to archives as needed)

```ini
[env:nrf52_dk]
platform = nordicnrf52
board = nrf52_dk
framework = zephyr
platform_packages = 
   toolchain-gccarmnoneeabi@~1.90301.0
   tool-dtc@file:///home/pi/tool-dtc-linux_armv6l-1.4.7.tar.gz
   tool-gperf@file:///home/pi/tool-gperf-linux_armv6l-3.0.4.tar.gz
```

gives

```
Reading CMake configuration...
Parsing system calls...
Generating syscall files
Generating KObject files...
Validating driver...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/nrf52_dk/offsets/zephyr/arch/arm/core/offsets/offsets.c.o
..
Linking .pio/build/nrf52_dk/firmware.elf
Checking size .pio/build/nrf52_dk/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   8.2% (used 5364 bytes from 65536 bytes)
Flash: [          ]   2.4% (used 12584 bytes from 524288 bytes)
Building .pio/build/nrf52_dk/firmware.hex
======================== [SUCCESS] Took 652.85 seconds ========================
...

```

maxgerhardt, 2021-10-01:
Okay I can build a firmware, but `pio test` which uses Unity fails to compile with the selected 9.3.1 GCC because

```
In file included from /home/pi/.platformio/packages/tool-unity/unity.h:21,
                 from /home/pi/.platformio/packages/tool-unity/unity.c:7:
/home/pi/.platformio/packages/tool-unity/unity_internals.h:15:10: fatal error: setjmp.h: No such file or directory
```

seems like you really need the originally wanted `toolchain-gccarmnoneeabi@~1.80201.0` (8.2.1) toolchain. But compiling GCC on slow Pi is painfull..

lexologe, 2021-10-01:
I can't use the originally wanted toolchain because

```
Tool Manager: Installing platformio/toolchain-gccarmnoneeabi @ ~1.80201.0
Error: Could not find the package with 'platformio/toolchain-gccarmnoneeabi @ ~1.80201.0' requirements for your system 'linux_armv7l'
```

You seem to use a linux_armv6l based Pi?

maxgerhardt, 2021-10-01:
Nono, check the updated comment above, the toolchain wasn't the problem.

lexologe, 2021-10-01:
ok, missed the edit. Now I can run `pio test` and missing packages can be installed. Thank you for compiling them to support the Pi.

lexologe, 2021-10-01:
I am closing this issue because I am able to recompile the packages if there are new versions required using the package.json contained in the tar files above. Thank you.

maxgerhardt, 2021-10-01:
Mmmm, I think for the sake of PlatformIO developers seeing this, this should remain open until it's fixed in the official package repositories.

lexologe, 2021-10-01:
Ok that would be nice. No need to compile these packages in the future. If I can help please let me know.

valeros, 2021-10-04:
@maxgerhardt thanks for the packages, uploaded to the official registry.

closed 2021-10-04 by valeros