platformio/platform-nordicnrf52 #128: Unespaced path problem in main.py

maxgerhardt, 2021-10-12:
Per https://community.platformio.org/t/c-users-xxxxx-is-not-recognized-as-an-internal-or-external-command-operable-program-or-batch-file/23934?u=maxgerhardt.

Users with a space in their username experience a failure open building with Adafruit boards

```
c:\users\ricky gai\.platformio\penv\scripts\python.exe “C:\Users\RICKY GAI.platformio\packages\tool-adafruit-nrfutil\adafruit-nrfutil.py” dfu genpkg --dev-type 0x0052 --sd-req 0x00B7 --application .pio\build\adafruit_feather_nrf52832\firmware.hex .pio\build\adafruit_feather_nrf52832\firmware.zip
MethodWrapper([“checkprogsize”], [".pio\build\adafruit_feather_nrf52832\firmware.elf"])
‘c:\users\ricky’ is not recognized as an internal or external command,
operable program or batch file.
*** [.pio\build\adafruit_feather_nrf52832\firmware.zip] Error 1
```
The path to `python` has not been properly escaped. This is because

https://github.com/platformio/platform-nordicnrf52/blob/1483d3ba84af20430a793fd676086fcd7303ed6f/builder/main.py#L131-L136

lacks double-quote escaping.

maxgerhardt, 2021-10-12:
Same issue exists in

https://github.com/platformio/platform-nordicnrf52/blob/1483d3ba84af20430a793fd676086fcd7303ed6f/builder/main.py#L148-L152

valeros, 2021-10-13:
Fixed, many thanks for reporting!

commit d29b3cc4 (2021-10-13): Properly handle paths to internal build tools

Resolve #128