platformio/platform-nordicnrf52 #136: Nicla mBed Hardfault.

gutzchi, 2021-12-21:
Hello.

I have been redicted from an issue I posted on the Arduino mbed repository: https://github.com/arduino/ArduinoCore-mbed/issues/372

I have tried to use upload the example codes of the Nicla Sense ME board with very little success. The simplest example code is the following one: 
```
#include "Nicla_System.h"

void setup() {
  Serial.begin(115200);
  while(!Serial);
  
  nicla::begin();
  nicla::leds.begin();
}

void loop() {
  nicla::leds.setColor(green);
  delay(1000);
  nicla::leds.setColor(off);
  delay(1000);
} 
```
This is the `platformio.ini' file:
```
[env:nicla_sense_me]
platform = nordicnrf52
board = nicla_sense_me
framework = arduino
monitor_speed=115200
debug_tool = cmsis-dap
upload_protocol = cmsis-dap
```
And this is the compiler log:
[build_log_nicla.txt](https://github.com/platformio/platform-nordicnrf52/files/7757325/build_log_nicla.txt)

If I debug it, it gets stuck when calling the `nicla::begin()` function, when creating and starting the thread (line 31):
https://github.com/arduino/ArduinoCore-mbed/blob/838c70238aaee11c7a8a32cf570de4f51ce240f7/libraries/Nicla_System/src/Nicla_System.cpp#L25-L35

This seems to be a problem of Platformio as it runs perfectly on the Arduino IDE. They use a different toolchain, by the way.

Moreover, if the `Arduino_BHY2.h` library is included, I get the following message form the OS:

```
++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R   0: 00000000
R   1: 2000345C
R   2: 00000001
R   3: 20000BF5
R   4: 00002000
R   5: 00000000
R   6: 20000BEC
R   7: 00000000
R   8: 00000000
R   9: 00000000
R  10: 00000000
R  11: 00000000
R  12: 00016E79
SP   : 20003408
LR   : 00010753
PC   : 00010764
xPSR : 41000000
PSP  : 200033E8
MSP  : 2000FFC0
CPUID: 410FC241
HFSR : 40000000
MMFSR: 00000000
BFSR : 00000000
UFSR : 00000001
DFSR : 00000000
AFSR : 00000000
Mode : Thread
Priv : Privileged
Stack: PSP

-- MbedOS Fault Handler --

++ MbedOS Error Info ++
Error Status: 0x80FF013D Code: 317 Module: 255
Error Message: Fault exception
Location: 0x10764
Error Value: 0x200087B4
Current Thread: main Id: 0x2000345C Entry: 0x17EE3 StackSize: 0xC00 StackMem: 0x20002838 SP: 0x20003408
For more info, visit: https://mbed.com/s/error?error=0x80FF013D&tgt=NICLA
-- MbedOS Error Info --
```
This is the corresponding build log:
[build_log_BHY2.txt](https://github.com/platformio/platform-nordicnrf52/files/7757356/compiler_log.txt)

If I debug it, the program crashes in the same lines (which I believe should not be executed as the macro `NO_NEED_FOR_WATCHDOG_THREAD` is defined:
https://github.com/arduino/ArduinoCore-mbed/blob/838c70238aaee11c7a8a32cf570de4f51ce240f7/libraries/Nicla_System/src/Nicla_System.cpp#L3-L9

Please, could someone help me with this?

valeros, 2022-01-03:
Hi @gutzchi ! Could you please try the following configuration in your platformio.ini:
```ini
[env:nicla_sense_me]
platform = nordicnrf52
board = nicla_sense_me
framework = arduino
monitor_speed=115200
lib_archive = no
```

gutzchi, 2022-01-06:
Hi @valeros. I have tried this configuration in both cases and the result is exactly the same.

epatel, 2022-02-19:
Have the same problem. I have also used the platformio upload command with the Arduino IDE produced binary and that worked fine so I don't think there is a problem with the upload but rather the binary itself.

asuar078, 2022-03-05:
I notice that in ~/.platformio/packages/framework-arduino-mbed/libraries/Nicla_System/src/Nicla_System.cpp the nicla::begin() function does not have a return statements. I fixed my issue by adding 
`return true;`
on line 35.

epatel, 2022-03-12:
@asuar078 That helped me too! Good catch! Thanks!