platformio/platform-nordicnrf52 #137: Erase and Upload on redbear_blenano2 failing [OS X  / Core 5.2.4·Home 3.4.0]

gregoryking, 2021-12-31:
redbear_blenano2 board tested as working on Mac OS X 12.1 (21C52) with the Arduino IDE, but failing with platform.io.

Attempting erase and upload hangs with no errors as follows. Let me know if more info useful to diagnose.

`Erase Flash` command output:

```
> Executing task in folder 211231-162818-arduino-blink: platformio run --target upload --target monitor --environment redbear_blenano2 <

Processing redbear_blenano2 (platform: nordicnrf52; framework: arduino; board: redbear_blenano2)
------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/redbear_blenano2.html
PLATFORM: Nordic nRF52 (9.1.0) > RedBearLab BLE Nano 2
HARDWARE: NRF52832 64MHz, 64KB RAM, 512KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduinonordicnrf5 1.700.201209 (7.0) 
 - tool-bossac-nordicnrf52 1.10901.201022 (1.9.1) 
 - tool-openocd 2.1100.211028 (11.0) 
 - tool-sreccat 1.164.0 (1.64) 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 2 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio/build/redbear_blenano2/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.2% (used 140 bytes from 65536 bytes)
Flash: [          ]   0.4% (used 2176 bytes from 524288 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, nrfjprog, stlink
CURRENT: upload_protocol = cmsis-dap
Uploading .pio/build/redbear_blenano2/firmware.hex
xPack OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev (2021-10-17-00:18)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
```

`Upload` command output:

```> Executing task in folder 211231-162818-arduino-blink: platformio run --target upload --environment redbear_blenano2 <

Processing redbear_blenano2 (platform: nordicnrf52; framework: arduino; board: redbear_blenano2)
------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/redbear_blenano2.html
PLATFORM: Nordic nRF52 (9.1.0) > RedBearLab BLE Nano 2
HARDWARE: NRF52832 64MHz, 64KB RAM, 512KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduinonordicnrf5 1.700.201209 (7.0) 
 - tool-bossac-nordicnrf52 1.10901.201022 (1.9.1) 
 - tool-openocd 2.1100.211028 (11.0) 
 - tool-sreccat 1.164.0 (1.64) 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 2 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio/build/redbear_blenano2/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.2% (used 140 bytes from 65536 bytes)
Flash: [          ]   0.4% (used 2176 bytes from 524288 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, nrfjprog, stlink
CURRENT: upload_protocol = cmsis-dap
Uploading .pio/build/redbear_blenano2/firmware.hex
xPack OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev (2021-10-17-00:18)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1```

valeros, 2022-01-03:
Hi @gregoryking ! The debug output seems truncated, there should be additional info after the `debug_level: 1` line.

gregoryking, 2022-01-04:
No that's pretty much it, that's where it hangs.  Here's an example screen recording of running in verbose mode.  I've re run the command as shown in debug with OpenOCD and it seems to hang just the same

https://user-images.githubusercontent.com/1253994/148137918-b4b86ead-657b-4577-8ac9-bd27a4f24c6a.mov

.

gregoryking, 2022-01-22:
Something must have been quite corrupted further up the chain on my machine as a fresh install of latest OS X remedied the issue.

closed 2022-01-22 by gregoryking