platformio/platform-nordicnrf52 #140: Debug hanging after "Reading symbols"

robsonos, 2022-02-03:
- [ x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Win10/Win11

**PlatformIO Version**: 5.2.4

### Description of problem

The debugger is no longer working in my project for some reason. It compiles but  it hangs after that as shown below:
[image]

This is my platformio.ini:

```
[env:test]
platform = nordicnrf52@9.1.0
board = adafruit_feather_nrf52840
framework = arduino
debug_tool = stlink
```

Things I have tested:
- Clean project files and rebuild project
- Delete PlatformIO files in .vscode and rebuild InteliSense index
- Fresh installation of VSCode
- Fresh installation of PlatformIO
- Reinstalled drivers for the programming/debugging (it is an stlink-v2 clone)
- Replacing the programming/debugging tool
- Programming using the programming/debugging tool instead of bootloader (both work)
- Installing Platformio dev version
- Running a clean Arduino project 
- Tested in different Windows versions ( Win11 and Win10)
- `pio debug` does not show any error
- ` pio debug --interface=gdb --interpreter=mi  -x .pioinit` runs just fine which is weird.

What is even stranger is that I have another computer (Win10) where it works. However, it does not work on this same computer with a different user and a fresh installation of VSCode, PlatformIO and the project.

Both have the same versions of VSCode, PlatformIO, and the tools used in PlatformIO:
```
 - framework-arduinoadafruitnrf52 1.10200.0 (1.2.0) 
 - framework-cmsis 2.50700.210515 (5.7.0) 
 - tool-adafruit-nrfutil 1.503.0 (5.3) 
 - tool-sreccat 1.164.0 (1.64) 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
 - tool-bossac-nordicnrf52 1.10901.201022 (1.9.1) 
 - tool-openocd 2.1100.211028 (11.0) 
```

#### Steps to Reproduce

1. Run PIO Debug form VSCode

### Actual Results

Debug console hangs after `Reading symbols`

### Expected Results

Debug to run normally

valeros, 2022-02-03:
Hi @robsonos ! So it works in CLI mode? Could you please share here a screenshot of the entire editor in the Debug view?

robsonos, 2022-02-03:
Hi @valeros,

Sure. It does seem to work from CLI:
[image]

Here are the screenshots from debug view:
[image]
[image]

valeros, 2022-02-03:
An empty sketch doesn't seem like a great project to test debugging, especially when you set breakpoints at lines with no code. Could you please try at least blink example like this one https://github.com/platformio/platform-nordicnrf52/blob/develop/examples/arduino-blink/src/Blink.cpp ?

robsonos, 2022-02-03:
Hi @valeros,

No problems. 
Just ran the blink example and got the same result for all 3 situations:
[image]
Note that there is a blue line just below PIO Debug on the top left. The debug is still running. I let it run for half an hour earlier today but nothing happened. Another detail that may be relevant is that in order to end the debug session I need to press the stop bottom twice.

valeros, 2022-02-03:
I see, could you also attach here a screenshot of a debug session running in CLI?

robsonos, 2022-02-03:
Sure:
[image]

valeros, 2022-02-03:
TBH, I don't see any obvious reason why it shouldn't work in VSCode as well. Could you please maximize panel size for Debug Console, just to see whether there is additional info on your debug session.

robsonos, 2022-02-03:
This is what I get with the window maximized: 
[image]
Interestingly, if I add `debug_build_flags = -O0 -ggdb3 -g3` to my platformio.ini I get this:
[image]
[image]

ivankravets, 2022-02-04:
@robsonos sorry for the issue, it was caused by a broken Python 3.9.10. See solution at https://github.com/platformio/platformio-vscode-ide/issues/2925#issuecomment-1029914810

closed 2022-02-04 by ivankravets

robsonos, 2022-02-05:
Thanks. I can confirm that this fixes the issue.