platformio/platform-nordicnrf52 #141: Update Zephyr to 3.0.0 - All platforms using Zephyr including Raspberry Pi Pico RP2040

zpm1066, 2022-02-28:
As per https://docs.zephyrproject.org/latest/releases/release-notes-3.0.html, this release now supports Raspberry Pi Pico RP2040.

Please update Zephyr to 3.0.0 for all platforms using Zephyr, including Raspberry Pico. Thank you.

zpm1066, 2022-04-02:
Any updates? Thanks.

Cyril-Roques, 2022-09-20:
Hello,

any progress on this side ?

zpm1066, 2022-09-21:
Unfortunately nothing yet. Pretty disappointing.

zpm1066, 2022-10-04:
Any traction on getting PlatformIO to support Zephyr RTOS v3.x? 

Is this a dead end for all platforms (nRF 52 series, nRF53, nRF9160, STM32, Raspberry PI Pico, and ESP32) that use Zephyr RTOS? If so, it'll be a huge disappointment and folks would move on to using Nordic Semi nRF Connect for VSCode (nRF52, nRF53, nRF9160) and command line Zephyr build process.

nilmonto, 2022-10-07:
Hi all.. I've been trying to use Zephyr 3 on my platformio setup for the past few days and found out - the worst way - that due to major changes done by Zephyr on its CMakeLists.txt, the platformio-build*.py scripts are no longer able to compile the project along with its zephyr binaries.

In order to do that one must modify platformio-build.py (and maybe also platformio-build-pre.py) to consider the new structure introduced by Zephyr 3 to build its artfacts. I've tried to do that and got (surprisingly due to my lack of knowledge about build process) success to build and generate a firmware.hex:

[image]

Unfortunately the work is still in progress and I wont be able to do for the next feel days, the changes I've done on python scripts still doesn't consider the three-phase build process Zephyr now relies on: zephyr-pre0, zephyr-pre1 and zephyr-final (previously it was just zephyr-prebuilt and zephyr-final), and thus, the following warning is generated due to incomplete build instructions:

[image]

To anyone that maybe wanting to upgrade its environment, please see instructions on [git@github.com:nilmonto/platformio-zephyr-3.git](https://github.com/nilmonto/platformio-zephyr-3)

Of course if someone from Platformio have mercy of our soul and translate the CMakeLists.txt into instructions of an updated pair of scripts to properly build Zephyr3 binaries I'm sure we all would be very thankful =)

maxpowel, 2022-11-10:
Would be great to have zephyr 3 for nrf52. I upgraded some parts (like littlefs) in my own but zephyr itself it is too much for me. If someone could bring some light it would be awesome. I spent many hours without success, but I will keep on it.

@nilmonto can you describe/list what is needed appart from translating CMakeLists.txt? I would like to help but I'm a little overwhelmed and don't really know the goals to achieve.

beriberikix, 2022-12-23:
FYI https://github.com/zephyrproject-rtos/zephyr/pull/53303

bradjc, 2023-02-08:
Pretty much a year later and not even a comment. Seems reasonable to assume that platform io is no longer supporting the nrf52 boards, or at least zephyr on the nrf52 boards?

GeoSpark, 2023-02-09:
> Pretty much a year later and not even a comment. Seems reasonable to assume that platform io is no longer supporting the nrf52 boards, or at least zephyr on the nrf52 boards?

Agreed. It's unfortunate. I've just got an Adafruit ItsyBitsy nRF52840, and I was disappointed there's no support for it in the old version of Zephyr that PlatformIO is using. It's a real shame because I love PlatformIO for everything else I have developed on.

zpm1066, 2023-02-09:
Yes, it’s an unfortunate stand taken up by the PlatformIO folks.
Luckily there is a much better solution for nRF52840 and Zephyr. Use the excellent Nordic Semi’s nRF Connect SDK. I’ve been using NCS for the last few years. It has now come a long way. Updates are regular, VSCode extensions are superb!
However, I do miss the ease of switching between nRF, STM32 and Raspberry Pi Pico with Zephyr that PlatformIO provided up to 2.7.x. You never know, common sense may prevail and we may get Zephyr support upgraded to 3.x for PlatformIO.

beriberikix, 2024-07-27:
Zephyr LTS 3.7 was officially [released](https://zephyrproject.org/announcing-zephyr-3-7-new-long-term-support-release-of-zephyr-rtos/).

jonnor, 2025-05-08:
Zephyr 4.x has now been released