platformio/platform-nordicnrf52 #149: precompiled lib has an error

andhieSetyabudi, 2022-04-28:
I tried to implement HRS3300' library from **[atc1441](https://github.com/atc1441)**  `https://github.com/atc1441/HRS3300-Arduino-Library`
but i get an error message _**error: precompiled_lib/hrs3300.c.o uses VFP register arguments, .pio\build\nrf52832_watch\firmware.elf does not**_

i create a dir **precompiled_lib** on my project so I use build_flag -L to add that dir. 

**platformio.ini**

> platform = nordicnrf52
> board =  stct_nrf52_minidev
> framework = arduino
> platform_packages = 
> upload_protocol = stlink
> board_build.mcu = nrf52832
> board_build.f_cpu = 32000000L
> build_flags = 
> 	        -DS132
>                 -DNRF52_S132
>                 -DNRF5
>                 -DUSE_LFXO
>                 -DCONFIG_NFCT_PINS_AS_GPIOS
> 
>                 -DNRF_LOG_USES_RTT=1
>                 -mfloat-abi=hard
>                 -mfpu=fpv4-sp-d16
>                 -L precompiled_lib
>                 -lheart

carlosperate, 2022-06-07:
I haven't really modified build paraments with platform.io before, so having no experience with this specific area take what I'm about to say with a huge pinch of salt, but just in case is useful:

It looks like maybe your precompiled_lib is not using the same `float-abi` and `fpu` flags as your project. I believe the `platformio.ini` provided is for the precompiled_lib, are you 100% your project is using the same flags?

If you have the same configuration for both,  one quick thing that it might be worth trying would be to set your project (or both actually) with `-mfloat-abi=softfp`, which allows linking hard-float and soft-float.
You might still have issues if the `fpu` flag is different (not sure if the compiler/linker would pick it up, but the wrong instruction type would cause an exception in your running programme), but at least that would get you one step closer, and you might be able to confirm if the settings are picked up from the expected files or not.

andhieSetyabudi, 2022-06-08:
> I haven't really modified build paraments with platform.io before, so having no experience with this specific area take what I'm about to say with a huge pinch of salt, but just in case is useful: [...]

Actually i've tried to use `-mfloat-abi=softfp` but i got same result. 

> .platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: error: lib\HRS3300lib\src\cortex-m4\libheart.a(hrs3300_alg_dynamic.o) uses VFP register arguments, .pio\build\hackaBLE_board\firmware.elf does not

I think there is different between gcc-arm version between arduino-package from [atc1441](https://github.com/atc1441/D6-arduino-nRF5) and NRF52-package in platformio..

CMIIW

andhieSetyabudi, 2022-06-16:
finally i can compile it, using 
`lib_archive = no`

closed 2022-06-16 by andhieSetyabudi