platformio/platform-nordicnrf52 #153: This code hangs in PlatformIO, not in Arduino IDE

steph643, 2022-07-04:


cujomalainey, 2022-07-04:
Given you are not initializing the struct, is it possible you have an invalid float that is causing a trap when you do your math?

cujomalainey, 2022-07-04:
If you don't multiply by 3 does it work?

cujomalainey, 2022-07-04:
> If you don't multiply by 3 does it work? 

Also might be worth dumping the pointer to see where it's at in each version to see if something is amuck

steph643, 2022-07-04:
> Also might be worth dumping the pointer

Nothing strange about `pm2` pointers: 536875026 on Arduino, 536873010 on PlatformIO.

> If you don't multiply by 3 does it work?

Yes it does.

cujomalainey, 2022-07-05:
> 
> > Also might be worth dumping the pointer
> 
> Nothing strange about `pm2` pointers: 536875026 on Arduino, 536873010 on PlatformIO.

And what is the location of `s`?

> 
> > If you don't multiply by 3 does it work?
> 
> Yes it does. 

Hmm if the memset doesn't work then the float being invalid is likely not the issue

steph643, 2022-07-05:
> And what is the location of s?

Nothing funny:
Arduino: &s = 536875012, pm2 = 536875026
PlatformIO: &s = 536872996, pm2 = 536873010

> Hmm if the memset doesn't work then the float being invalid is likely not the issue

On top of that, shouldn't an invalid float be considered as NaN and propagated in every float computation?

cujomalainey, 2022-07-05:
> 
> > Hmm if the memset doesn't work then the float being invalid is likely not the issue
> 
> On top of that, shouldn't an invalid float be considered as NaN and propagated in every float computation?
> 

I don't worth floats very often in C, but I wouldn't leave anything to chance here. At this point I would recommend objdumping the assembly code in the .o files and see if there is a difference in the type of instruction used for the multiplication

Richardn2002, 2025-02-06:
Here we are into 2025, and I am bumping into weird issues trying to run Tensorflow Lite on this Arduino Nano 33 Rev 2. Behaviors I am seeing include, a lot of code combination will cause the chip to enter a "bricked" state where it does not show up on any serial port and must be recovered by writing another program in under bootloader mode (entered by pressing reset button twice). I suspect the chip hard faulted, but as I currently don't have a debugger I cannot confirm. My colleague is able to reproduce my observations within PlatformIO with a different setup (I am on Linux and he is on OS X).

I call these behaviors "weird" cause seemingly unrelated code modifications would cause the chip to operate normally instead, for example code addition in a branch that will not be taken, more/less serial prints, etc., which, appear to me, very likely is that there are undefined behaviors in my code and compilers are legally behaving wild.

I choose to reply here cause, neither me or my colleague is able to "brick" the chip with the same application code under Arduino IDE, that would otherwise brick the chip under PlatformIO, which, if all code including libraries are being the same, indicate a difference between the build systems of PlatformIO and Arduino IDE.

As of today, [`framework-arduino-mbed`](https://registry.platformio.org/tools/platformio/framework-arduino-mbed) of PlatformIO, auto installed by

```
platform = nordicnrf52@10.7.0
board = nano33ble
framework = arduino
```

is at version 4.2.1, the same as [ArduinoCore-mbed](https://github.com/arduino/ArduinoCore-mbed) at version 4.2.1. Additionally I put `platform_packages = toolchain-gccarmnoneeabi@1.70201.0` into my `platformio.ini`, so that both PlatformIO and Arduino IDE will use the identical compiler:

```
gcc version 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204] (GNU Tools for Arm Embedded Processors 7-2017-q4-major)
```

but even with this addition I cannot manage to make those "bricking" issues go away. I am currently inspecting differences in build steps and flags to see if I can find anything. The last resort will be analyzing disassembly, but as Tensorflow Lite is really big and I have yet to create a minimal reproduction, I would rather not rush into disassembly analysis now. This issue provides an intriguing minimal case, but sadly I am not able to reproduce it (hanging) with my current project setup. I will try different versions of toolchains and see how I would reproduce it.

I hope someone more familiar with both Arduino IDE and PlatformIO can come up with more efficient ways of locating differences. For now, I am manually checking `compile_commands.json` from both toolchains with my bare eyes.

Related:
- the support for this board is tracked at #53, still open.
- #109 is likely related. Although the reporter describe it as a port discovery issue: I myself was faced with exact same outputs. Turns out, PlatformIO cannot find the port not because port discovery is broken, but because the chip is "bricked" and does not listen on the serial port anymore.
- #175 is also likely related. Although described as an upload issue, again, I myself was faced with the exact same outputs, with the true reason being the chip stopped responding. Interestingly, in the comment section people mentioned that falling back to `platform = nordicnrf52@9.5.0` solves problems for them, and in this specific version the compiler used is indeed, the `gcc version 7.2.1 20170904 ` version, same as the one Arduino IDE uses even till today.

Richardn2002, 2025-02-07:
Yeah based on the initial time of post I `platform = nordicnrf52@9.4.0` but I cannot replicate this. Though it is true that I am on Core 6.1.16 - Home 3.4.4, a newer version.