platformio/platform-nordicnrf52 #156: COM ports inaccessible after building FW in release 6.1.1

singletone, 2022-07-17:


ivankravets, 2022-07-17:
This is not a problem of PlatformIO Core. Please upgrade PlatformIO Core to the latest version via `pio upgrade`.

Also, please share your `platformio.ini`

singletone, 2022-07-17:
Hello, Ivan,
I have typed pio upgrade, got the following:
************************
You're up-to-date!
PlatformIO 6.1.1 is currently the newest version available.
*************************
Then I have re-built the FW and downloaded it to the board again - the same issue, COM ports are 'dead'...
Please note that this does not happen if the code is built in previous releases! Here is the copy of my platformio.ini file, as is (I did not remove comments):
***********************
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:adafruit_feather_nrf52840]
platform = nordicnrf52
board = adafruit_feather_nrf52840
framework = arduino
debug_tool = jlink
monitor_speed = 115200

; debug_tool = jlink

; SWD interface
;upload_protocol = jlink

; JTAG interface
;upload_protocol = jlink-jtag

; SWD interface
;upload_protocol = jlink-swd
;upload_speed = 115200
;debug_init_break = tbreak setup
;debug_speed = auto
; upload_port = COM11
; upload_port = COM[11]
; debug_port  = COM[12]
; monitor_port = COM[12]
**********************************
- be aware that it hasn't been changed for months, and yet caused no problems.
If you help with downgrading to the previous release I would be very obliged, as this is not only the issue of my own but my customer's too: the version gets upgraded automatically, and the process cannot be controlled (as much as I can see).
Thanks in advance for your help!

ivankravets, 2022-07-17:
Could you try `pio upgrade —dev` (double hyphen) and clean/flash again. Does it work now?

ivankravets, 2022-07-18:
Please open VSCode settings and search for "Use PIO Core Development". Enable checkbox. Restart VSCode.

Type `pio --version`. Which version do you see?

ivankravets, 2022-07-18:
You need to run commands in PlatformIO Core CLI (terminal icon on the bottom status bar) or as described in https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli

So, you need ` 6.1.2a3`. Once you have it, please run `pio run -t cleanall`, build the project again and flash it. Does it work now?

singletone, 2022-07-18:
Yeah, it works!!! :) I did exactly what you suggested, now I have PlatformIO Core, version 6.1.2a3, and COM ports are regained. Thanks a lot for your quick response and help, Ivan!
Apart from this, I wish peace and prosperity to your country (I follow the news and my heart is heavy to watch and hear of what happens there these days). Слава Україні!.

closed 2022-07-18 by singletone

ivankravets, 2022-07-18:
Thanks for the testing! The stable 6.1.2 release is planned for today.

P.S: Thanks for the kind words and your support! 🙏  Героям Слава!