platformio/platform-nordicnrf52 #163: LEDs not blinking on nrf52840_dk

pwilkowski, 2022-10-15:
Hello, i have a brand new nrf52840 development kit with jlink on board
When it arrived, LED was blinking (the one near actual NRF chip.

However, when i tried uploading simple code

```c
#include <Arduino.h>

void setup()
{
	// put your setup code here, to run once:
	pinMode(PIN_LED1, OUTPUT);
	pinMode(PIN_LED2, OUTPUT);
	pinMode(PIN_LED3, OUTPUT);
	pinMode(PIN_LED4, OUTPUT);
	pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
	digitalWrite(PIN_LED1, HIGH); // turn the LED on (HIGH is the voltage level)
	digitalWrite(PIN_LED2, HIGH); // turn the LED on (HIGH is the voltage level)
	digitalWrite(PIN_LED3, HIGH); // turn the LED on (HIGH is the voltage level)
	digitalWrite(PIN_LED4, HIGH); // turn the LED on (HIGH is the voltage level)
	digitalWrite(LED_BUILTIN, HIGH); // turn the LED on (HIGH is the voltage level)
	delay(3000);				  // wait for a second
	digitalWrite(PIN_LED1, LOW);  // turn the LED off by making the voltage LOW
	digitalWrite(PIN_LED2, LOW);  // turn the LED on (HIGH is the voltage level)
	digitalWrite(PIN_LED3, LOW);  // turn the LED on (HIGH is the voltage level)
	digitalWrite(PIN_LED4, LOW);  // turn the LED on (HIGH is the voltage level)
	digitalWrite(LED_BUILTIN, LOW); // turn the LED on (HIGH is the voltage level)
	delay(3000);				  // wait for a second
}
```

No diode was blinking anymore. I also did try to upload `arduino-blink` example with same results.

This leads me to believe uploading firmware works however something in program is not correct? Or there is a bug?

my platformio.ini:

```
[env:nrf52840_dk]
platform = nordicnrf52
board = nrf52840_dk
framework = arduino
```

Ocelot7777, 2023-08-17:
Hello, did you solve the problem? Help needed here.

davewyers, 2024-03-11:
Following the Docs here https://docs.platformio.org/en/stable/tutorials/nordicnrf52/arduino_debugging_unit_testing.html 

You need to update your board to 
board = nrf52840_dk_adafruit

I had the same issue, using my ini file 
```
[env:nrf52840_dk_adafruit]
platform = nordicnrf52
board = nrf52840_dk_adafruit
debug_tool = jlink
upload_protocol = jlink
monitor_speed = 115200 
;change microcontroller
board_build.mcu = nrf52840
; change MCU frequency
board_build.f_cpu = 64000000L
; SoftDevice version
build_flags = -DNRF52_S140
```

Using the nRF Programmer App I noticed that the SoftDevice was not getting written to the device on Upload.

When I "Burn Bootloader" first, then "Upload" it worked correctly.

[image]

I am not sure if this step is missing from the docs or this is a bug.

Regards
Dave